morrownr/8821au-20210708 #6: Missing channel (like 157)

frcs6, 2021-10-04:
I'm with Ubuntu 20.04.3 and when my wifi is on channel is 157, I can't connect to 5ghz.
With Windows I don't have any issue.

iwlist chan :

wlx7cc2c61c4a31  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Channel 149 : 5.745 GHz
          Channel 153 : 5.765 GHz
          Current Frequency:5.22 GHz (Channel 44)

I have an TP-Link Mini - AC600 USB 2.0 (Archer T2U Nano)

morrownr, 2021-10-04:
Let me duplicate this issue and see if I can figure out what is going on. Have you connected successfully using any of the channels that do show?

This paint has not dried on this driver code. There are a lot of changes and editions from the last version and this version has shown in my testing to be much better than previous but with big change comes the probability of some bugs.

frcs6, 2021-10-04:
I have the same issue with the previous version.
I can connect to all the channel in the list.
I tried with other distro without success.

morrownr, 2021-10-04:
Hmmm...

Unable to duplicate the problem here so far. Here is what I have done but first are details of my test setup:

Alfa AWUS036ACS usb wifi adapter based on 8811au
Desktop amd64 based system running Linux Mint 20.2 (kernel 5.11)
WiFi router changed to channel 157, 80 MHz channel width

$ iw phy phy0 channels (phy0 is my 8811au adapter)
Band 1:
	* 2412 MHz [1] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2417 MHz [2] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2422 MHz [3] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2427 MHz [4] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2432 MHz [5] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2437 MHz [6] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2442 MHz [7] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2447 MHz [8] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2452 MHz [9] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2457 MHz [10] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2462 MHz [11] 
	  Maximum TX power: 16.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2467 MHz [12] 
	  Maximum TX power: 16.0 dBm
	  No IR
	  Channel widths: 20MHz HT40- HT40+
	* 2472 MHz [13] 
	  Maximum TX power: 16.0 dBm
	  No IR
	  Channel widths: 20MHz HT40- HT40+
	* 2484 MHz [14] (disabled)
Band 2:
	* 5180 MHz [36] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5200 MHz [40] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5220 MHz [44] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5240 MHz [48] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5260 MHz [52] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5280 MHz [56] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5300 MHz [60] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5320 MHz [64] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5500 MHz [100] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5520 MHz [104] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5540 MHz [108] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5560 MHz [112] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5580 MHz [116] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5600 MHz [120] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5620 MHz [124] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5640 MHz [128] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5660 MHz [132] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5680 MHz [136] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5700 MHz [140] 
	  Maximum TX power: 13.0 dBm
	  Radar detection
	  Channel widths: 20MHz HT40- HT40+ VHT80
	  DFS state: usable (for 864 sec)
	  DFS CAC time: 60000 ms
	* 5720 MHz [144] (disabled)
	* 5745 MHz [149] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5765 MHz [153] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5785 MHz [157] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5805 MHz [161] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5825 MHz [165] 
	  Maximum TX power: 13.0 dBm
	  Channel widths: 20MHz HT40- HT40+ VHT80
	* 5845 MHz [169] (disabled)
	* 5865 MHz [173] (disabled)
	* 5885 MHz [177] (disabled)

$ iw dev
phy#1
	Interface wlxe0e1a914dXXX
		ifindex 4
		wdev 0x1
		addr e0:e1:a9:14:db:XX
		type managed
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0
	Interface wlx00c0caac4XXX (this is the 8811au)
		ifindex 3
		wdev 0x100000001
		addr 00:c0:ca:ac:47:XX
		ssid myMT76-5g
		type managed
		channel 157 (5785 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 13.00 dBm

As you can see, it is connected to myMT76-5g. I ran iperf3 and did some surfing. Performance is very good.

Verdict: Unable to duplicate the problem. It works here.

Recommendations:

Don't use "iwlist", use "iw"

Run ```$ iw phy phy0 channels``` and post the results. Change phy0 as needed.

Run ```S iw dev" and post the results.

Run ```$ sudo iw reg get``` and post the results.

morrownr, 2021-10-04:
Oh, something else. I did a lot of work on the driver last night. Recommend you run ```$ sudo ./remove-driver.sh``` followed by deleting the driver directory followed by a reboot and full fresh installation. That will make sure we are on the same code.

frcs6, 2021-10-04:
Oh my bad I so sorry.
I had the old driver from aircrack-ng.

frcs6@YOGA:~$ iw dev
phy#0
	Interface wlx7cc2c61c4a31
		ifindex 2
		wdev 0x1
		addr 7c:c2:c6:1c:4a:31
		ssid FraLi
		type managed
		channel 157 (5785 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 13.00 dBm

Sorry for the issue, I close it and continue to test this driver.

closed 2021-10-04 by frcs6

morrownr, 2021-10-04:
No problem. It happens. I can use all the help I can get testing this driver so please give me reports as you have time.