morrownr/8821au-20210708 #9: Hostapd: Invalid/unknown driver '8821au'

GreenglassT, 2021-11-10:
After installing the driver on my Raspberry Pi 4, attempting to get an AP working I run into this error in journalctl -xe

```
Nov 10 05:53:44 raspberrypi hostapd[943]: Line 2: invalid/unknown driver '8821au'
Nov 10 05:53:44 raspberrypi hostapd[943]: 1 errors found in configuration file '/etc/hostapd/hosta>
Nov 10 05:53:44 raspberrypi hostapd[943]: Failed to set up interface with /etc/hostapd/hostapd.conf
Nov 10 05:53:44 raspberrypi hostapd[943]: Failed to initialize interface
Nov 10 05:53:44 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited, stat>
Subject: Unit process exited
Defined-By: systemd
Support: https://www.debian.org/support

An ExecStart= process belonging to unit hostapd.service has exited.
 
The process' exit code is 'exited' and its exit status is 1.
Nov 10 05:53:44 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Subject: Unit failed
Defined-By: systemd
Support: https://www.debian.org/support

The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
Nov 10 05:53:44 raspberrypi systemd[1]: Failed to start Access point and authentication server for>
░░ Subject: A start job for unit hostapd.service has failed
Defined-By: systemd
Support: https://www.debian.org/support

A start job for unit hostapd.service has finished with a failure.

The job identifier is 6231 and the job result is failed.
```

morrownr, 2021-11-10:
Hi @GreenglassT 

Please post the contents of your hostapd.conf.

Also, there is a file here in the repo called Bridged Wireless Access Point that has example hostapd.conf in it.

GreenglassT, 2021-11-10:
> Please post the contents of your hostapd.conf.

`interface=wlan1
driver=8821au

hw_mode=a
channel=36
country_code=US
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
ht_capab=[HT40][SHORT-GI-20ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][RX>
macaddr_acl=0
ignore_broadcast_ssid=0

auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

ssid="Pi4-Extender"
wpa_passphrase="pass"`
 
> Also, there is a file here in the repo called Bridged Wireless Access Point that has example hostapd.conf in it.

This .conf uses nl80211 for the driver. When I did that though hostapd started without errors, however after reboot the WiFi AP did not show up on my phone.

morrownr, 2021-11-10:
> `interface=wlan1

interface=wlan1
(one little typo can mess things up)

> driver=8821au

driver=nl80211
 
> hw_mode=a
> channel=36

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

> country_code=US
> ieee80211n=1
> ieee80211ac=1
> wmm_enabled=1

> ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][RX> 

ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC1][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]

> macaddr_acl=0
> ignore_broadcast_ssid=0

> auth_algs=1
> wpa=2
> wpa_key_mgmt=WPA-PSK
> rsn_pairwise=CCMP
> 
> ssid="Pi4-Extender"
> wpa_passphrase="pass"`

> AP did not show up on my phone. Also, I am using wlan0 instead of eth0, so I don't think a bridged AP would work?

So, the source of your internet is wlan0 and the AP is wlan1? That can work. That is basically a repeater.

First we need to get the AP where your phone can see the access point and log in. This can happen even with etho and wlan0 are messed up. You just may not be able to connect to the internet. We can fix that later.

Crank up edit-options.sh from your driver directory:

$ sudo ./edit-options.sh

and post what the options line says.

GreenglassT, 2021-11-10:
> crank up edit-options.sh from your driver directory:
> $ sudo ./edit-options.sh
> and post what the options line says.

`options 8821au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=0
`

GreenglassT, 2021-11-10:
> > crank up edit-options.sh from your driver directory:
> > $ sudo ./edit-options.sh
> > and post what the options line says.
> 
> `options 8821au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=0`

Also, should I change the hw_mode=a to vht_oper_chwidth=1? Or do I place it in after hw_mode and channel.

Thank you so much for your help. I purchased a new WiFi adapter that supports 5GHz. I got my old 2.4GHz adapter to work but this is much more complicated because of the driver.

GreenglassT, 2021-11-10:
Here is my new hostapd.conf file:

```
interface=wlan1
driver=nl80211

hw_mode=a
channel=36
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
country_code=US
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC1][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
macaddr_acl=0
ignore_broadcast_ssid=0

auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

ssid="Pi4-Extender"
wpa_passphrase="Tigerg86"
```

Does this look correct?

morrownr, 2021-11-10:
> > `options 8821au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_dfs_region_domain=0`

Change:
rtw_vht_enable=1

To:
rtw_vht_enable=2

> Also, should I change the hw_mode=a to vht_oper_chwidth=1? Or do I place it in after hw_mode and channel.

Add:
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

After:
channel=36

> Thank you so much for your help. I purchased a new WiFi adapter that supports 5GHz. I got my old 2.4GHz adapter to work but this is much more complicated because of the driver.

You are welcome. Going to 5 GHz does involve more settings but you will get used to it quickly. Adapters based on mt7612u and mt7610u have drivers that require fewer settings: https://github.com/morrownr/USB-WiFi

Something else that might be of interest: 

https://openwrt.org/toh/views/toh_fwdownload?dataflt%5B0%5D=supported+current+rel_%3D21.02.1&dataflt%5BBrand*%7E%5D=Raspberry

OpenWRT is pretty cool and it has drivers for the adapters based on the mt7612u and mt7610u chipsets.

morrownr, 2021-11-10:
> Here is my new hostapd.conf file:
>
> ```
> interface=wlan1
> driver=nl80211
>
> hw_mode=a
> channel=36
> vht_oper_chwidth=1
> vht_oper_centr_freq_seg0_idx=42
> country_code=US
> ieee80211n=1
> ieee80211ac=1 [...]

That looks pretty good. Are you able to see the access point and login with your phone? 

Comfirm what OS you are using?

GreenglassT, 2021-11-10:
> That looks pretty good. Are you able to see the access point and login with your phone?
> Comfirm what OS you are using?

I cannot see the access point on my phone. I plan on reflashing Raspbian Lite OS (32-bit) and just going through all the steps again from this article: [https://www.section.io/engineering-education/setting-up-a-wifi-extender-using-a-raspberry-pi/].

I will also change the settings in edit-options.sh to what you recommended.

I have seen OpenWRT and if I cant get these steps to work I plan on playing with OpenWRT. 

I'll update once I have a clean install and the correct steps done.

GreenglassT, 2021-11-11:
I GOT IT TO WORK!!!! I just used a different guide but used the options you helped me with. Internet working great now. THANK YOU SO MUCH FOR YOUR HELP!

closed 2021-11-11 by GreenglassT

morrownr, 2021-11-11:
You are welcome. These wifi adapters and RasPi's are fun to play with and you can learn some networking while you are at it.

Stop by anytime.

Nick

GreenglassT, 2021-11-12:
They are very fun indeed. One last question. My normal router has speeds north of 100 mbps, however my AP is now only shooting out 35 mbps at best. Is this something I can fix on my end or is that just how it goes with using Raspberry Pi's as access points? This USB Wifi adapter is rated for upwards of 433 mbps on the 5GHz band.

morrownr, 2021-11-12:
A RasPi4B is capable of gigabyte speeds. A 8811au wifi adapter is capable of up to a link speed of 433 but given it works half duplex, you can see transfer speeds of up to around 210 Mb/s. For us to tune this baby up for max performance, we need to discuss the details of everything from your ip to your phone. I'm up for it if you are. If you want to start a new issue with an appropriate name, that might be best.