morrownr/8821au-20210708 #10: Tune Raspberry Pi AP to Top Performance

GreenglassT, 2021-11-15:
Alright! I'm ready to get this AP as efficient as possible if you're up for it still.

IP: Spectrum (sadly) 
Devices Connected: iPhone X, M1 MacBook Air, Motorola Z4

What is everything else you need to know?

morrownr, 2021-11-15:
Let's find out what we need to know.

We need a good way to find out how the connections between your devices and the RasPi are performing. I'm going to suggest that you use iperf3. You will need iperf3 running as a client on one or more of your client devices. Here is an article that may help you set iperf3 up on your iPhone. Once that is done, you can probably handle setting it up on your other client devices on your own:

https://blog.fosketts.net/2020/07/09/test-your-wi-fi-with-iperf-for-ios/

iperf3 needs to run as a server also on the RasPi. Please open the file called Bridged Wireless Access Point here in this repo and scroll down to the section called `Install and autostart iperf3`. It is near the bottom. Don't forget to reboot the Pi after setting up the iperf3 server.

Once iperf3 is going, we can start taking some reading which will help us head the right direction.

GreenglassT, 2021-11-17:
Alright, sitting down and going to get iperf3 up and running on my pi. I'll update you when I've got it working.

GreenglassT, 2021-11-17:
iPerf3 is up and running!

morrownr, 2021-11-17:
On the RasPi, did you set iperf3 up to autostart or are you starting it from a terminal?

iperf3 -s

The various client versions can have different interfaces so how you run it depends on the interface. I run my client on my desktop system, Linux Mint 20.2, from the terminal:

iperf3 -c 192.168.1.1 -t  60

-c is the ip address of the system where iperf3 is running in server mode.

-t means for the test to run for 60 seconds.

Post some output so that we can see how the link between your clients and the RasPi AP are doing.

Example:

```
[  5]  54.00-55.00  sec  45.0 MBytes   377 Mbits/sec    0   1.54 MBytes       
[  5]  55.00-56.00  sec  45.0 MBytes   377 Mbits/sec    0   1.54 MBytes       
[  5]  56.00-57.00  sec  45.0 MBytes   377 Mbits/sec    0   1.54 MBytes       
[  5]  57.00-58.00  sec  45.0 MBytes   377 Mbits/sec    0   1.54 MBytes       
[  5]  58.00-59.00  sec  43.8 MBytes   367 Mbits/sec    0   1.54 MBytes       
[  5]  59.00-60.00  sec  46.2 MBytes   388 Mbits/sec    0   1.54 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec  2.63 GBytes   377 Mbits/sec    2             sender
[  5]   0.00-60.01  sec  2.63 GBytes   377 Mbits/sec                  receiver
```

The example above is from a test between my desktop, USB WiFi adapter is a Alfa ACM AC1200, and the AP/iperf3 server is my ZyXEL NBG6817 WiFi router that runs OpenWRT. Both of my RasPis are down this morning for reconfiguration to do testing this afternoon.

GreenglassT, 2021-11-18:
Outputs:

iPhone: 
Average 165 Mbits/sec (5 Streams, 30 seconds)

Macbook: 
`[  5]   0.00-1.00   sec  27.5 MBytes   231 Mbits/sec                  
[  5]   1.00-2.00   sec  29.1 MBytes   244 Mbits/sec                  
[  5]   2.00-3.00   sec  24.9 MBytes   210 Mbits/sec                  
[  5]   3.00-4.00   sec  23.1 MBytes   194 Mbits/sec                  
[  5]   4.00-5.01   sec  21.6 MBytes   181 Mbits/sec                  
[  5]   5.01-6.00   sec  25.1 MBytes   211 Mbits/sec                  
[  5]   6.00-7.01   sec  20.3 MBytes   170 Mbits/sec                  
[  5]   7.01-8.01   sec  13.5 MBytes   113 Mbits/sec                  
[  5]   8.01-9.00   sec  23.2 MBytes   196 Mbits/sec                  
[  5]   9.00-10.00  sec  19.7 MBytes   165 Mbits/sec    `

GreenglassT, 2021-11-18:
Outputs!

iPhone: 

165 Mbits/sec avg (5 streams, 30 seconds0

Macbook:

[  5]   0.00-1.00   sec  27.5 MBytes   231 Mbits/sec                  
[  5]   1.00-2.00   sec  29.1 MBytes   244 Mbits/sec                  
[  5]   2.00-3.00   sec  24.9 MBytes   210 Mbits/sec                  
[  5]   3.00-4.00   sec  23.1 MBytes   194 Mbits/sec                  
[  5]   4.00-5.01   sec  21.6 MBytes   181 Mbits/sec                  
[  5]   5.01-6.00   sec  25.1 MBytes   211 Mbits/sec                  
[  5]   6.00-7.01   sec  20.3 MBytes   170 Mbits/sec                  
[  5]   7.01-8.01   sec  13.5 MBytes   113 Mbits/sec                  
[  5]   8.01-9.00   sec  23.2 MBytes   196 Mbits/sec                  
[  5]   9.00-10.00  sec  19.7 MBytes   165 Mbits/sec                  
^C[  5]  10.00-10.74  sec  8.76 MBytes  99.3 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.74  sec   237 MBytes   185 Mbits/sec                  sender
[  5]   0.00-10.74  sec  0.00 Bytes  0.00 bits/sec                  receiver

morrownr, 2021-11-19:
That is not bad at all. I'm curious what channel you are using? Do your client devices support DFS channels? If so, we could look at finding a totally unused DFS channel so that you have no congestion to deal with.

Maybe we should look at the contents of hostapd.conf and the options line in /etc/modprobe.d/8821au.conf one more time.

Is the setup stable?

GreenglassT, 2021-11-19:
I am currently using channel 36. I'm a complete noob when it comes to networking however I looked up DFS and understand that certain channels have different frequencies (36-48 has 5170-5250 MHz). How do I find an unused DFS channel?

I do not have a file called /etc/modprobe.d/8821au.conf

The setup seems stable, WiFi does not drop in Mbits unless the host network drops Mbits.

morrownr, 2021-11-19:
Let's do a little WiFi 101. The follow page has good info:

https://en.wikipedia.org/wiki/List_of_WLAN_channels

You can go down to the 5 GHz section. I'll use the US as an example but it depends on what country you live in.

Regular 5 GHz channels are 36-48 and 149-161 (and 165 these days). That may seem like a lot but it is not because if 80 MHz channel width is used, and it generally is, then it is like having only 2 channels because 80 MHz covers the entire 36-48 and 149-161 channels. So if you live in an area where there are many wifi access points, 5 GHz can actually be more crowded than 2.4 GHz. I see far less congestion here on channel 11 (2.4 GHz) than on the regular 5 GHz channels because everyone goes for 5 GHz because they think it is better. Since very few people know anything about wifi, they usually just leave their router/access point on whatever settings their internet provider used. 

In my area, the internet providers set 5 GHz up for use on channels 44 and 157 with an 80 MHz channel width. That means everyone is crammed into what basically amount to 2 channels so the regular 5 GHz channels are congested. So, what do I do?

I look for clean, uncongested air. I use two methods here:

1. I have three 5 GHz radios in regular use. Two of them are set to use 80 MHz width channels in DFS space. Nobody uses DFS around here. It is like having the space all to yourself.  I have one radio set to channel 124 and another on channel 60. Both are DFS. Not all routers/access points are capable of DFS support. This driver is but requires a setting in the 8821au.conf file.

2. I also use another adapter on channel 149 but I limit it to 20 MHz width. It is for a couple of clients I have that cannot do DFS. Channel 149 is a regular channel but since all APs in the upper range are using channel 157 which ends up leaving a bit of open air at 149, I use it. Works well

With a little knowledge, one can really improve the wifi experience in most situations. To see what channels and ranges are in heavy use at your location, you can use various apps. I use LinSSID on Linux and WiFi Analyzer on my Android phone. There are several "WiFi Analyzer" apps available so you can probably find one.

If you installed this driver, you should have a file called /etc/modprobe.d/8821au.conf. Run:

$ sudo nano /etc/modprobe.d/8821au.conf

Regards

GreenglassT, 2021-11-19:
I'm not at home right now but I'm close. I'm gonna get LinSSID running on a Linux VM and see what pops up. 

I might have mistyped /etc/modprobe.d/8821au.conf when I SSH'd into my Pi. I'll try again later tonight.

So once I find a channel that is relatively barren, change the channel in my hostapd.conf file to the new one and see if it is relatively better.

I'll report back the differences in speed tonight.

Thank you so much for your help! I am really interested in IT and this has been a great learning experience already.

morrownr, 2021-12-15:
Hey @GreenglassT 

Status?

GreenglassT, 2021-12-23:
All good!! I think the bottleneck is my original speed from the router I'm using to provide WiFi to the Pi. But I am very satisfied! Thank you again so much for your help.

closed 2021-12-23 by GreenglassT