morrownr/8821au-20210708 #11: Driver works as well in openSUSE Tumbleweed

dhdeangelis, 2021-11-20:
Not an issue just to add a line to your list of tested distributions. I have used this driver successfully in openSUSE Tumbleweed since last August (previous version first, this version now).

openSUSE Tumbleweed is a rolling distribution that is continuously upgraded, so a number of kernel versions have come and gone since last August. All of them have worked flawlessly. Current kernel is 5.15.2.

The WiFI adapter is a generic USB dongle based on RTL8811AU (0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter).

Thank you for this effort, much appreciated!

morrownr, 2021-11-20:
Hi @dhdeangelis 

Thanks for the input. Please check the new entry in `Tested Linux Distribution` to see if it meets your approval.

Also, opneSUSE users could benefit from 2 addional entries in the README:

1. See Steps 2 and 3 of Installation Steps. Tell me what works for openSUSE.

2. In the section `Download Locations for Tested Linux Distributions` please tell me what I should put.

For what it is worth, the old version of the driver was pretty solid in good ole client mode but otherwise had a few problems. This new version has really impressed. AP and monitor mode support is now really good. In fact, I did not have to do that much work on this version.

Thanks, Nick

dhdeangelis, 2021-11-21:
Thanks, @morrownr, that is but add please "Tumbleweed" after openSUSE, in that way openSUSE people know which version is being referred, like: 

openSUSE Tumbleweed (rolling) 

Best,
Hernán

morrownr, 2021-11-22:
Hernán,

Changes made as requested. Will be merged soon. Would inexperienced openSUSE users know what to do for Installation steps 2 and 3? I can add the info if you tell me what to add.

dhdeangelis, 2021-11-22:
Nick,

As far as I understand there are a few repositories for openSUSE from where you could easily install this using the `zypper` tool. See for example: [https://build.opensuse.org/search?search_text=8821AU](search 8821AU in openSUSE Open Build Service). But I am not ware that any of this uses your particular implementation.

Regarding the installation recommendation I prefer to just install by unzipping the package and running the scripts as root. I believe it is easy enough for most users. The only caveat is that any kernel upgrade will render the module installation useless and it will have to be removed and reinstalled after that. In a rolling distribution this can happen often, even once a week, but this is something users are aware of when they choose a rolling distro so no surprise here IMO.

morrownr, 2021-11-22:
Hernán,

> Regarding the installation recommendation I prefer to just install by unzipping the package and running the scripts as root.

I think I am communicating my request poorly. A big part of my plan here is to make it as easy as possible for those new to Linux. I also try to keep things such that experienced users, such as yourself, can alter the installation instructions and not have a problem. To make things easy, since most new users do not know what to install to set up the development environment on their distro, Step 3 of the installation instructions provides the command to install the necessary files for development. This can and does vary by distro. From the README:

Step 3: Install the required packages (select the option for the OS you are using)

Option for Raspberry Pi OS
sudo apt install -y raspberrypi-kernel-headers bc build-essential dkms git

Option for Debian and Linux Mint Debian Edition (LMDE)
sudo apt install -y linux-headers-$(uname -r) build-essential dkms git libelf-dev

Option for Ubuntu (all flavors) and Linux Mint
sudo apt install -y dkms git build-essential

Option for Fedora
sudo dnf -y install git dkms kernel-devel kernel-debug-devel

... and more.

I have no idea what this should be for openSUSE. I would be glad to add an option for openSUSE given your guidance. Step 2 could also use some help for openSUSE.

Regards,

Nick

dhdeangelis, 2021-11-22:
I see. I have unfortunately no comments here, as things worked right out of the box. Also I am not proficient in kernel options. But if you are interested in getting more and better information it is perhaps advisable to contact the people maintaining the openSUSE packages listed in the link above. They can surely do deeper than me in this case.

closed 2021-11-29 by morrownr