morrownr/8821au-20210708 #12: How to Enable WPA3 support

morrownr, 2021-11-21:
A guide showing how to enable WPA3 support has been requested.

[update_wpa_supplicant_v2.txt](https://github.com/morrownr/8821au-20210708/files/7575760/update_wpa_supplicant_v2.txt)

The guide is not a general purpose guide. It is very specific and only for the drivers as noted in the guide. Please post your results and suggestions in this issue.

Regards,

Nick

ivanovborislav, 2021-11-27:
Hi all,

You have to enable below settings when build wpa_supplicant. Edit .config, uncomment or add:

CONFIG_TLS=openssl
CONFIG_IEEE80211W=y
CONFIG_SAE=y
CONFIG_LIBNL20=y
CONFIG_LIBNL32=y

Re-compile wpa_supplicant

[wpa_cli_with_wpa_supplicant.pdf](https://github.com/ivanovborislav/document/blob/main/wpa_cli_with_wpa_supplicant.pdf)

morrownr, 2021-11-27:
Hi @ivanovborislav 

I appreciate your test and report. You are correct. With version 2 of the guide, I simply did not see that `CONFIG_TLS=openssl` was commented out in the default configuration file and that `CONFIG_IEEE80211W=y` was missing. My old eyes sometimes let me down. I am preparing a v3 of the guide and I am going to make the time to test it here before posting.

Regards

morrownr, 2021-11-29:
Here is the latest guide to upgrading wpa_supplicant:

[update_wpa_supplicant_v3-WPA3.txt](https://github.com/morrownr/8821au-20210708/files/7620266/update_wpa_supplicant_v3-WPA3.txt)

Hopefully I did not make as many mistakes this time.

I need reports of success or failure for both the installation of wpa_supplicant and if WPA3 is working.

Please include distro and kernel version in reports.

Thanks

Nick