morrownr/8821au-20210708 #15: I have error, please, help me

KirMozor, 2021-12-03:
````
kirill@localhost:~/src> sudo zypper install git dkms kernel-headers
Загрузка данных о репозиториях...
Чтение установленных пакетов...
"git" уже установлен.
Нет кандидатов на обновление "git-2.34.1-1.1.x86_64". Самая последняя версия уже установлена.
"dkms" уже установлен.
Нет кандидатов на обновление "dkms-2.6.1-2.8.noarch". Самая последняя версия уже установлена.
Строка "kernel-headers" не найдена в именах пакетов. Пробуем возможности.
"linux-glibc-devel", предоставляющий "kernel-headers", уже установлен.
Разрешение зависимостей пакетов...
Нечего выполнять.
kirill@localhost:~/src> mkdir -p ~/src
kirill@localhost:~/src> cd ~/src
kirill@localhost:~/src> git clone https://github.com/morrownr/8821au-20210708.git
Клонирование в «8821au-20210708»…
remote: Enumerating objects: 1046, done.
remote: Counting objects: 100% (1046/1046), done.
remote: Compressing objects: 100% (713/713), done.
remote: Total 1046 (delta 475), reused 887 (delta 329), pack-reused 0
Получение объектов: 100% (1046/1046), 3.07 МиБ | 135.00 КиБ/с, готово.
Определение изменений: 100% (475/475), готово.
kirill@localhost:~/src> cd ~/src/8821au-20210708
kirill@localhost:~/src/8821au-20210708> sudo ./install-driver.sh

Running install-driver.sh version 20211002
Starting installation...
Copying source files to: /usr/src/rtl8821au-5.12.5.2
Copying 8821au.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source ->
                 /usr/src/rtl8821au-5.12.5.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.15.3-1-default cannot be found at
/lib/modules/5.15.3-1-default/build or /lib/modules/5.15.3-1-default/source.
An error occurred. dkms build error = 1
Please report this error.
kirill@localhost:~/src/8821au-20210708> 

````

KirMozor, 2021-12-03:
What this error? Where to dig?

KirMozor, 2021-12-03:
My system:

````
kirill@localhost
OS: openSUSE Tumbleweed x86_64
Kernel: 5.15.3-1-default
Uptime: 2 hours, 58 mins
Packages: 2364 (rpm) 
Shell: bash 5.1.8 
Resolution: 1280x1024 
DE: Plasma 5.23.3
WM: KWin
Theme: Breeze [Plasma], Breeze [GTK2/3]
Icons: breeze [Plasma], breeze [GTK2/3]
Terminal: konsole 
CPU: Pentium E5800 (2) @ 3.199GHz 
GPU: NVIDIA GeForce GT 630 
Memory: 2192MiB / 5925MiB

kirill@localhost:~> lsusb
Bus 001 Device 007: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Bus 001 Device 011: ID 0e8d:2005 MediaTek Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 058f:6362 Alcor Micro Corp. Flash Card Reader/Writer
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 0458:0186 KYE Systems Corp. (Mouse Systems) Genius DX-120 Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
kirill@localhost:~> 

````

morrownr, 2021-12-04:
Hi @KirMozor 

The problem is that the development environment is not correct. I realize that I do not have proper instructions for openSUSE users and I would like to correct that if I can get help from an experienced openSUSE user.

I suspect the problem is in this line:

$ sudo zypper install git dkms kernel-headers

I think the specific problem is `kernel-headers`. I think it should be something else but a search of the net did not clarify what is correct so my recommendation is for you to ask about how to properly setup the kernel headers in the openSUSE forums.

Regards,

Nick

KirMozor, 2021-12-04:
Hello, good, I will unsubscribe about this problem on the forums and in telegram chats, if I can solve the problem, I will send a solution to you :)

morrownr, 2021-12-07:
I hope to you back with news about a solution.

Regards

DeadMozay, 2021-12-07:
> ```
> kirill@localhost:~/src> sudo zypper install git dkms kernel-headers
> Загрузка данных о репозиториях...
> Чтение установленных пакетов...
> "git" уже установлен.
> Нет кандидатов на обновление [...]

There is no kernel-headers package in openSUSE, the kernel headers are in the package kernel-default-devel
In addition, there is an already built kernel module without dkms
https://software.opensuse.org//download.html?project=home%3ASauerland&package=rtl8812AU_8821AU
Use Kernel stable repo

morrownr, 2021-12-07:
> There is no `kernel-headers` package in openSUSE, the kernel headers are in the package `kernel-default-devel`

Now we know. Are you willing to guide me on how to update the installation instructions, to include testing?

> In addition, there is an already built kernel module without dkms https://software.opensuse.org//download.html?project=home%3ASauerland&package=rtl8812AU_8821AU Use Kernel stable repo

I see that the title on that link says...

`rtl8812AU_8821AU`

That is an indication to me that the driver in the link is based on old source code. The driver has not been a combined driver supporting multiple chipsets for a while and the driver was not very good back then. The driver here is MUCH better. You are welcome to point the author of the driver in the link to this driver so he can update his code.

DeadMozay, 2021-12-07:
> Now we know. Are you willing to guide me on how to update the installation instructions, to include testing?

There shouldn't be any more differences in the instructions, just the package names, if you like I will send the PR for openSUSE later

> That is an indication to me that the driver in the link is based on old source code. The driver has not been a combined driver supporting multiple chipsets for a while and the driver was not very [...]

the maintainer has to take into account a number of features of the distribution and other necessary changes, late updates are not uncommon, he will have time to update the driver

morrownr, 2021-12-07:
> > Now we know. Are you willing to guide me on how to update the installation instructions, to include testing?
> 
> There shouldn't be any more differences in the instructions, just the package names, if you like I will send the PR for openSUSE later

Yes, please submit a PR as you have time.

Regards

KirMozor, 2021-12-07:
@DeadMozay
Thank you. My wifi is work!
It was enough to establish kernel-default-devel

KirMozor, 2021-12-07:
@morrownr 
For the driver to work, you need to install the dependencies:
`sudo zypper install git dkms kernel-headers kernel-default-devel`
And run the installation script `sudo sh install-driver.sh`
It's work! Finnaly!!!!

DeadMozay, 2021-12-07:
> > > Now we know. Are you willing to guide me on how to update the installation instructions, to include testing?
> > 
> > 
> > There shouldn't be any more differences in the instructions, just the package names, if you like I will send the PR for openSUSE later
> 
> Yes, please submit a PR as you have time.
> 
> Regards

https://github.com/morrownr/8821au-20210708/pull/16

closed 2021-12-12 by KirMozor