morrownr/8821au-20210708 #21: Installation error - cannot find kernel headers

DarumasLegs, 2022-01-08:
Running install-driver.sh version 20211212
Starting installation...
Copying source files to: /usr/src/rtl8821au-5.12.5.2
Copying 8821au.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source ->
                 /usr/src/rtl8821au-5.12.5.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.10.17-v7+ cannot be found at
/lib/modules/5.10.17-v7+/build or /lib/modules/5.10.17-v7+/source.
An error occurred. dkms build error = 1
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

morrownr, 2022-01-08:
Hi @DarumasLegs 

The problem is that the kernel headers are either not installed or are in an unexpected place.

What distro are you using?

Regards

DarumasLegs, 2022-01-08:
5.10.17-v7+

DarumasLegs, 2022-01-08:
I do have several folders under /lib/modules/ - could the headers be in the
wrong folder? How can I tell, and can I just move them to that folder it is
looking for them in?

morrownr, 2022-01-08:
I know the kernel version. I want to the distro... Ubuntu, Debian, Manjaro...

morrownr, 2022-01-08:
> I do have several folders under /lib/modules/ - could the headers be in the wrong folder? How can I tell, and can I just move them to that folder it is looking for them in?

Hold on... don't do that.

DarumasLegs, 2022-01-08:
Oh - sorry - Raspbian.

morrownr, 2022-01-08:
Did you mean to say Raspberry Pi OS?

Is it the regular 32 bit version or the 64 bit beta?

DarumasLegs, 2022-01-08:
I’m running a Raspberry Pi 3B+. I believe it’s the regular 32 bit version.

morrownr, 2022-01-08:
Run:

$ uname -m

 If it says aarch64 then it is 64 bit. If it says armv7l then it is 32 bit.

If it is the 32 bit version run the following:

$ sudo apt update
$ sudo apt full-upgrade
$ sudo apt install -y raspberrypi-kernel-headers bc build-essential dkms git

Tell me what that last line gives us...

DarumasLegs, 2022-01-08:
uname -m returned "armv71", so it's 32 bit.

I ran what you suggested, and here is what the last line gave us:

***@***.****:*~ $* sudo apt install -y raspberrypi-kernel-headers bc
build-essential dkms git

Reading package lists... Done

Building dependency tree

Reading state information... Done

bc is already the newest version (1.07.1-2).

build-essential is already the newest version (12.6).

dkms is already the newest version (2.6.1-4).

git is already the newest version (1:2.20.1-2+deb10u3).

raspberrypi-kernel-headers is already the newest version
(1:1.20211201~buster-1).

0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

***@***.****:*~ $*

DarumasLegs, 2022-01-08:
By the way, what is the best way to tell if the RPi is using the USB WiFi
adapter or the internal WiFi radio? Perhaps running "iw dev" and checking
the Mac address?

morrownr, 2022-01-08:
raspberrypi-kernel-headers is already the newest version
(1:1.20211201~buster-1).

Hmmm... that tells us the kernel headers are installed.

I have a spare Pi right now so give me some time to install and see what is going on.

For what it is worth: The installation script has not been changed in any way that would affect this for at least several months and it is used on 5 different drivers here that are installed thousands of times per week... however I am seeing the exact same report over in the 88x2bu repo so all I can say is please give me some time to duplicate this so I can figure it out.

Regards

DarumasLegs, 2022-01-08:
If there is anything I can do to help - running any specific commands,
etc., please let me know. Thank you!

DarumasLegs, 2022-01-08:
Here's something I find odd. I am running raspberrypiOS 5.10.17-v7+ #1403,
but when I look at /lib/modules/ I don't see a directory that matches.
Check this out:

***@***.****:*/lib/modules $* ls

*5.10.63+*  *5.10.63-v7+*  *5.10.63-v7l+*  *5.10.63-v8+*

***@***.****:*/lib/modules $* uname -a

Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l
GNU/Linux

***@***.****:*/lib/modules $*

Could this be part of my problem?

morrownr, 2022-01-08:
> By the way, what is the best way to tell if the RPi is using the USB WiFi adapter or the internal WiFi radio? Perhaps running "iw dev" and checking the Mac address?

Well, interesting question. I guess i've been doing this long enough that I can just tell. The `iw dev` output will be different for the internal radio and I think the reason is that the driver for the internal radio is now a mac80211 driver that has been up-steamed into the kernel, which is what we want. I can post some samples after I get this new test install going.

morrownr, 2022-01-08:
Just to let you know, I move show stoppers to the top of my priority list and this is a show stopper, however, I have other things that may make tracking this down today not likely to happen. We'll see.

DarumasLegs, 2022-01-08:
I ran across this post:
https://forums.raspberrypi.com/viewtopic.php?t=289999

It describes how to fix a mismatch between the kernel running and the
/lib/modules. I ran the following:

> https://forums.raspberrypi.com/viewtopic.php?t=289999
>

And now the mismatch is gone, so I'm running your install-driver.sh again
now...

DarumasLegs, 2022-01-08:
Your script completed normally - so I guess that was the issue! Now I see
two wlan devices (wlan0 and wlan1)!

morrownr, 2022-01-08:
That is great! To confirm what you say fixed things"

$ sudo apt update; sudo apt install --reinstall raspberrypi-bootloader raspberrypi-kernel; sudo reboot

Is that what fixed things?

I guess we owe the RasPi Foundations thanks for messing things up. It happens. Are you planning on checking to see if this bug has been reported? 

Now, back to business: Run the following and post the results:

$ iw dev

...and I show you the things I look for to tell the difference. 

Regards

DarumasLegs, 2022-01-08:
Honestly, I wouldn't even know where to start to report a bug, and
honestly, I'm not convinced this wasn't something that I did rather than a
bug.

Here is the result of iw dev:

***@***.****:*~ $* iw dev

phy#1

Interface wlan1

ifindex 4

wdev 0x100000001

addr 98:48:27:e8:e2:38

type managed

txpower 42949572.96 dBm

phy#0

Unnamed/non-netdev interface

wdev 0x2

addr ba:27:eb:fb:d1:ac

type P2P-device

txpower 31.00 dBm

Interface wlan0

ifindex 3

wdev 0x1

addr b8:27:eb:fb:d1:ac

type managed

channel 34 (5170 MHz), width: 20 MHz, center1: 5170 MHz

txpower 31.00 dBm

I checked iw dev before plugging in the wifi adapter, and the MAC address
was the same as what's listed as wlan0, so I assume wlan1 is the usb
adapter, right?

I'd like to also be able to tell if the AP that I'm broadcasting from this
pi is going through the USB adapter or the on-board one. It looks like it's
still using the onboard one, right?

Bob

morrownr, 2022-01-08:
>  I assume wlan1 is the usb adapter, right?

That is correct. There are some differences and you will never see your usb adapter with a txpower of 31.

> I'd like to also be able to tell if the AP that I'm broadcasting from this pi is going through the USB adapter or the on-board one. It looks like it's still using the onboard one, right?

Well, since you are settings up at AP, you might want to do what I do when I set up an AP on a Pi. I shut the onboard radio down:

$ sudo nano /boot/config.txt

Add:

```
# turn off onboard WiFi
dtoverlay=disable-wifi
```

Are you using my AP guide that is in the /docs folder here?

DarumasLegs, 2022-01-08:
I'm not - my developer wrote the code for the AP. I'll ask him to check out
your AP guide.

Question about the txpower - is 31 low or high? What range should the USB
adapter txpower be?

morrownr, 2022-01-09:
> Question about the txpower - is 31 low or high?

That is higher than most but it is likely because the Pi's wifi is in a little box inside the Pi.

> What range should the USB adapter txpower be?

Oh, 14 to 23 but likely the default is around 20. Most adapters these days come with txpower fixed in firmware so it can't be changed.

DarumasLegs, 2022-01-09:
And I guess lower numbers actually mean greater power?

morrownr, 2022-01-09:
> And I guess lower numbers actually mean greater power?

Not in this case. It would probably be a good idea to not use the txpower number of any specific adapter or card to judge its range. 

A good place to get information about USB WiFi adapters is:

https://github.com/morrownr/USB-WiFi

There are some pretty knowledgeable people that stop in Discussions/Issues there and there is a lot to read about adapters as well.