morrownr/8821au-20210708 #22: command failed: Network is down (-100)

sdack, 2022-01-09:
Hello,

I cannot get this driver to work. The device I am using the driver with is:
`ID 7392:a812 Edimax Technology Co., Ltd Edimax AC600 USB`

The 8821au driver compiles as a module for 5.15.13 using _clang-13.0.1_, but produces several warning messages. It installs and loads fine, and it shows up as a device with `iw` and also as a network device with `ifconfig`.

However, when trying to run a scan using the command `iw dev wlx... scan` does it produce the error _command failed: Network is down (-100)_.

_hostapd_ in AP mode starts, it initialises and also recognises an incoming connection on the device, but then appears stuck and a full connection never gets established (no DHCP information gets exchanged).

When compiling the kernel and 8821au driver with _gcc-11.2.1_ can I get as far as running a scan on the device and produce a list of the surrounding WiFi networks unlike with clang. _hostapd_ however still fails even when using gcc-11.2.1.

When I use the rtl8812au AirCrack-NG driver (v5.6.4.2) with gcc-11.2.1 does everything work fine. I can see surrounding WiFi APs and I can connect with various devices to my own WiFi AP and access the local network as expected.

I appreciate any suggestions you have. My main goal is to find a driver for this device, which compiles with clang and supports most if not all of it features.

On a minor note, the rtl8812au AirCrack-NG driver reports somewhat different device capabilities than this 8821au driver:

**AirCrack-NG:**
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0x1 RX 0x1
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1972
			HT20/HT40
			Static SM Power Save
			**RX Greenfield**
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
[... 311 lines cut ...]
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	Device supports randomizing MAC-addr in scans.
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	Supported extended features:

morrownr, 2022-01-09:
Hi @sdack 

A lot to unpack here.

Some of things you are talking about will require work over a period of time and testing will need to be accomplished. Are you willing to stick around and see this through?

> when using gcc-11.2.1

It is highly unlikely Realtek has been using or testing with gcc-11. My main dev box is up to gcc 9.3. Is it probable that some issues need to be worked for gcc-11? Yes. Am I willing to work them? Yes. What distro are you using? Depending on the distro, I might install it to one of my test systems and work from it for getting things working with gcc-11.

> using clang-13.0.1

I have no experience with Clang but am willing to give it a try once gcc-11 is working well.

> the rtl8812au AirCrack-NG driver reports somewhat different device capabilities

I am aware of this. The primary differences between this driver and the Aircrack-ng driver have to do with the ages of the drivers' source code. The source code for this driver is only a few months old while the Aircrack-ng driver code base is several years old. This driver will likely be more stable. In fact, I consider this driver to be rock solid. This driver does support DFS channels (maybe not all yet but more can be added), AP mode is rock solid, monitor mode is very good, concurrent mode and more.

To the specific issues you highlighted:

> RX Greenfield

I can add this. It is not clear to me that this is a good thing. Can you make a case for it?

> DSSS/CCK HT40

 I can investigate adding this but most folks that are in band 2 try to run 80 MHz channel width VHT. Is this needed?

> 5075 MHz [15] (30.0 dBm)

This is interesting. Keep in mind that just because the driver is reporting something does not make it so. While that may look cool, I suspect it is just cosmetic but if the driver is capable of trying to get an adapter that high and the adapter accepts the request, it will be breaking laws in many countries. Many adapters will not accept this setting as they would burn out. I think you will be pleased with the power capabilities of the driver here. If there are issues, I can take a look at some point.

If you have access to a system that uses an older gcc, you can test this driver and see.

Regards

sdack, 2022-01-09:
I am cross-compiling on an x86-64 for an armhf 32-bit platform (Allwinner H2+ sunxi) under Debian Testing. When I find the time will I test older compilers and other platforms to see if it makes any difference. I do have however no issues with my setup and I will stick with the Aircrack-NG driver.

I have also tried the RTL8821AU driver by Ivanov Borislav, who uses the same Realtek driver as you do (v5.12.5.2). It shows pretty much identical behaviour to your driver, meaning, I can scan for other APs but cannot create fully working connections to my own AP. The reported capabilities appear to be identical as well. However, there is one difference, which is it does not produce the Network is down (-100) response when compiled with clang-13.0.1, but appears to produce a result identical to gcc. (https://github.com/ivanovborislav/rtl8821au) This might be of interest to you.

The Greenfield feature is meant to improve throughput and the Aircrack-NG driver appears to support it in combination with hostapd. The dBm values of the Aircrack-NG driver are too high in my opinion. The vendor only lists the device with 11-14 dBm. However, I can change the output power and produce a weaker signal with the Aircrack-NG driver. It has been working well so far.

sdack, 2022-01-10:
I was hoping you may have already run into this issue, but apparently you have not. I am going to close it, because this is not a priority for me.

closed 2022-01-10 by sdack