morrownr/8821au-20210708 #23: How to install your driver in Solus Linux?

KirMozor, 2022-01-16:
Hello, I wanted to install your driver on Solus Linux, but there is a problem: DKMS was cut by the creators of the distribution: [https://getsol.us/2018/02/15/exploring-solus-architecture/](https://getsol.us/2018/02/15/exploring-solus-architecture/)
How then to install the driver?

morrownr, 2022-01-16:
Try:

$ make clean
$ sudo make install
$ sudo reboot

Doing it this way will mean that every time an updated kernel comes in, you will have to repeat this process.

KirMozor, 2022-01-16:
```
kirill@kirmozor ~/tools $ git clone https://github.com/morrownr/8821au-20210708.git && cd 8821au-20210708 && make clean && sudo make install
Клонирование в «8821au-20210708»…
remote: Enumerating objects: 1131, done.
remote: Counting objects: 100% (1131/1131), done.
remote: Compressing objects: 100% (770/770), done.
remote: Total 1131 (delta 535), reused 939 (delta 357), pack-reused 0
Получение объектов: 100% (1131/1131), 3.10 МиБ | 108.00 КиБ/с, готово.
Определение изменений: 100% (535/535), готово.
/bin/sh: строка 1: cc: команда не найдена
(standard_in) 1: syntax error
#make -C /lib/modules/5.13.1-187.current/build M=/home/kirill/tools/8821au-20210708 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
/bin/sh: строка 1: cc: команда не найдена
(standard_in) 1: syntax error
install -p -m 644 8821au.ko  /lib/modules/5.13.1-187.current/kernel/drivers/net/wireless/
install: не удалось выполнить stat для '8821au.ko': Нет такого файла или каталога
make: *** [Makefile:2509: install] Ошибка 1
```

sunwire, 2022-01-16:
> /bin/sh: строка 1: cc: команда не найдена

You have to install some programming tools, like compiler linker etc.
PS. Please, temporarily set the language to English, not everyone knows Russian
e.g LANG=C git clone ...

KirMozor, 2022-01-16:
This is after installing gcc `sudo eopkg install gcc`
```
kirill@kirmozor ~/tools $ LANG=ENG
kirill@kirmozor ~/tools $ git clone https://github.com/morrownr/8821au-20210708.git && cd 8821au-20210708 && make clean && sudo make install
Cloning into '8821au-20210708'...
remote: Enumerating objects: 1131, done.
remote: Counting objects: 100% (1131/1131), done.
remote: Compressing objects: 100% (770/770), done.
remote: Total 1131 (delta 535), reused 939 (delta 357), pack-reused 0
Receiving objects: 100% (1131/1131), 3.10 MiB | 242.00 KiB/s, done.
Resolving deltas: 100% (535/535), done.
#make -C /lib/modules/5.14.21-210.current/build M=/home/kirill/tools/8821au-20210708 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
install -p -m 644 8821au.ko  /lib/modules/5.14.21-210.current/kernel/drivers/net/wireless/
install: cannot stat '8821au.ko': No such file or directory
make: *** [Makefile:2509: install] Error 1
```

sunwire, 2022-01-16:
There is one important step between make clean and make install
**make**
`git clone https://github.com/morrownr/8821au-20210708.git && cd 8821au-20210708 && make clean && make && sudo make install`
PS.
LANG=en or LANG=en_US or LANG=en_GB not LANG=ENG

KirMozor, 2022-01-16:
```
kirill@kirmozor ~/tools $ LANG=en_US
kirill@kirmozor ~/tools $ git clone https://github.com/morrownr/8821au-20210708.git && cd 8821au-20210708 && make clean && make && sudo make install
Cloning into '8821au-20210708'...
remote: Enumerating objects: 1131, done.
remote: Counting objects: 100% (1131/1131), done.
remote: Compressing objects: 100% (770/770), done.
remote: Total 1131 (delta 535), reused 939 (delta 357), pack-reused 0
Receiving objects: 100% (1131/1131), 3.10 MiB | 63.00 KiB/s, done.
Resolving deltas: 100% (535/535), done.
#make -C /lib/modules/5.14.21-210.current/build M=/home/kirill/tools/8821au-20210708 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.21-210.current/build M=/home/kirill/tools/8821au-20210708  modules
make[1]: *** /lib/modules/5.14.21-210.current/build: No such file or directory.  Stop.
make: *** [Makefile:2503: modules] Error 2
```

sunwire, 2022-01-16:
> make[1]: *** /lib/modules/5.14.21-210.current/build: No such file or directory.  Stop.

You also need kernel development files

morrownr, 2022-01-16:
I should have mentioned that the installation instructions should all be followed with the exception of the last one, install-driver.sh. Now you need to run the following to clean things up:

sudo ./remove-driver.sh

Now, start the installation from Step 1. I realize many steps make not work on Solus as they are written and I have no knowledge of Solus. A quick look at Distrowatch tells me Solus is an independent distro which means I don't even know where to tell you to start. 

The basic idea is that you are going to have to install gcc and the kernel header files as a minimum so you need to find out how to do that for step 3. You should replace step 10 with:

$ make clean
$ make
$ sudo make install
$ sudo reboot

KirMozor, 2022-01-16:
I installed the following packages: `gcc linux-current-headers make`
Started compilation with commands: `LANG=en_US && git clone https://github.com/morrownr/8821au-20210708.git && cd 8821au-20210708 && make clean && make && sudo make install`
Passed without errors, I'll try to reboot

KirMozor, 2022-01-16:
Driver it works!
Now for the driver, you can add an instruction for Solus Linux
`sudo eopkg install gcc linux-current-headers make`
`git clone https://github.com/morrownr/8821au-20210708.git`
`cd 8821au-20210708`
`make`
`sudo make install`

closed 2022-01-16 by KirMozor

morrownr, 2022-01-16:
Thanks. I updated the information in the README.