morrownr/8821au-20210708 #24: Hi, strange problem, not going to drive

KirMozor, 2022-01-18:
I reinstalled Solus Linux, but when I tried to build the driver, after installing all the dependencies, it didn't build
```
kirill@kirmozor ~/tools/8821au-20210708 $ sudo make
Password: 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.21-210.current/build M=/home/kirill/tools/8821au-20210708  modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.21-210.current'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
  CC [M]  /home/kirill/tools/8821au-20210708/core/rtw_cmd.o
In file included from /home/kirill/tools/8821au-20210708/include/drv_types.h:78,
                 from /home/kirill/tools/8821au-20210708/core/rtw_cmd.c:17:
/home/kirill/tools/8821au-20210708/include/rtw_recv.h: In function 'recvframe_put':
/home/kirill/tools/8821au-20210708/include/rtw_recv.h:755:24: warning: variable 'prev_rx_tail' set but not used [-Wunused-but-set-variable]
  755 |         unsigned char *prev_rx_tail;
      |                        ^~~~~~~~~~~~
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c: In function 'rtw_cmd_thread':
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:529:23: warning: variable 'prspbuf' set but not used [-Wunused-but-set-variable]
  529 |         u8 *pcmdbuf, *prspbuf;
      |                       ^~~~~~~
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c: In function 'rtw_joinbss_cmd':
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:1135:74: warning: taking address of packed member of 'struct _WLAN_BSSID_EX' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1135 |                            pnetwork->network.IELength - 12, &psecnetwork->IELength,
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~

/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:1157:61: warning: taking address of packed member of 'struct _WLAN_BSSID_EX' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1157 |                                 pnetwork->network.IELength, &psecnetwork->IELength);
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:1162:64: warning: taking address of packed member of 'struct _WLAN_BSSID_EX' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1162 |      rtw_append_exented_cap(padapter, &psecnetwork->IEs[0], &psecnetwork->IELength);
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~

/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:993:18: warning: variable 'auth' set but not used [-Wunused-but-set-variable]
  993 |         u8      *auth, res = _SUCCESS;
      |                  ^~~~
gcc: fatal error: cannot execute 'as': execvp: No such file or directory
compilation terminated.
make[2]: *** [scripts/Makefile.build:271: /home/kirill/tools/8821au-20210708/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1858: /home/kirill/tools/8821au-20210708] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.14.21-210.current'
make: *** [Makefile:2503: modules] Error 2
kirill@kirmozor ~/tools/8821au-20210708 $ sudo eopkg install gcc git make linux-current-headers
The following package(s) are already installed and are not going to be installed again:
gcc  git  linux-current-headers  make
No packages to install.
kirill@kirmozor ~/tools/8821au-20210708 $ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.21-210.current/build M=/home/kirill/tools/8821au-20210708  modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.21-210.current'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
  CC [M]  /home/kirill/tools/8821au-20210708/core/rtw_cmd.o
In file included from /home/kirill/tools/8821au-20210708/include/drv_types.h:78,
                 from /home/kirill/tools/8821au-20210708/core/rtw_cmd.c:17:
/home/kirill/tools/8821au-20210708/include/rtw_recv.h: In function 'recvframe_put':
/home/kirill/tools/8821au-20210708/include/rtw_recv.h:755:24: warning: variable 'prev_rx_tail' set but not used [-Wunused-but-set-variable]
  755 |         unsigned char *prev_rx_tail;
      |                        ^~~~~~~~~~~~
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c: In function 'rtw_cmd_thread':
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:529:23: warning: variable 'prspbuf' set but not used [-Wunused-but-set-variable]
  529 |         u8 *pcmdbuf, *prspbuf;
      |                       ^~~~~~~
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c: In function 'rtw_joinbss_cmd':
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:1135:74: warning: taking address of packed member of 'struct _WLAN_BSSID_EX' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1135 |                            pnetwork->network.IELength - 12, &psecnetwork->IELength,
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~

/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:1157:61: warning: taking address of packed member of 'struct _WLAN_BSSID_EX' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1157 |                                 pnetwork->network.IELength, &psecnetwork->IELength);
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~
/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:1162:64: warning: taking address of packed member of 'struct _WLAN_BSSID_EX' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1162 |      rtw_append_exented_cap(padapter, &psecnetwork->IEs[0], &psecnetwork->IELength);
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~

/home/kirill/tools/8821au-20210708/core/rtw_cmd.c:993:18: warning: variable 'auth' set but not used [-Wunused-but-set-variable]
  993 |         u8      *auth, res = _SUCCESS;
      |                  ^~~~
gcc: fatal error: cannot execute 'as': execvp: No such file or directory
compilation terminated.
make[2]: *** [scripts/Makefile.build:271: /home/kirill/tools/8821au-20210708/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1858: /home/kirill/tools/8821au-20210708] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.14.21-210.current'
make: *** [Makefile:2503: modules] Error 2
kirill@kirmozor ~/tools/8821au-20210708 $ 
```
Before installing the driver, I updated the system, could this be the problem?

This system:
```
kirill@kirmozor
--------------- 
OS: Solus x86_64
Kernel: 5.14.21-210.current 
Uptime: 12 mins 
Packages: 716 (eopkg) 
Shell: bash 5.1.16 
Resolution: 1280x1024
DE: MATE 1.26.0 
WM: Metacity (Marco)
Theme: Plata-Noir [GTK2/3] 
Icons: Papirus [GTK2/3]
Terminal: mate-terminal 
Terminal Font: Hack 9
CPU: Intel Core 2 Duo E8500 (2) @ 3.1
GPU: NVIDIA GeForce GT 630 
Memory: 1154MiB / 5942MiB 
```

Also if it helps, I uploaded the list of installed packages to pastebin: [click](https://pastebin.com/VW4gqhw8)

sunwire, 2022-01-18:
Install binutils

morrownr, 2022-01-19:
> arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support

If installing binutils fixes the problem, let me know so that I can update the installation information in the README.

KirMozor, 2022-01-20:
Thank you, I was able to build the driver

closed 2022-01-20 by KirMozor