morrownr/8821au-20210708 #26: Driver doesn't pick up usb wifi adapter

UntouchedWagons, 2022-01-31:
I bought this [USB wifi adapter](https://www.amazon.ca/gp/product/B078NRDTYK/ref=ppx_yo_dt_b_asin_title_o01_s00) to use with my Raspberry Pi 3 and installed your driver and rebooted, but the adapter is still showing up as unclaimed in `lshw`:

```
        *-usb:1 UNCLAIMED
             description: Generic USB device
             product: 802.11ac NIC
             vendor: Realtek
             physical id: 5
             bus info: usb@1:1.5
             version: 2.00
             serial: 123456
             capabilities: usb-2.00
             configuration: maxpower=500mA speed=480Mbit/s
```

Any ideas?

UntouchedWagons, 2022-02-01:
```
# uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux camera-monitor-01 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux
EFI variables are not supported on this system
Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
8812au, 4.2.3, 5.10.63-v7+, armv7l: installed (WARNING! Diff between built and installed module!)
rtl8821au, 5.12.5.2, 5.10.63-v7+, armv7l: installed
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr ba:27:eb:f7:53:16
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr b8:27:eb:f7:53:16
                ssid BRADNET
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 31.00 dBm
```

morrownr, 2022-02-01:
> 8812au, 4.2.3, 5.10.63-v7+, armv7l: installed (WARNING! Diff between built and installed module!)

The above line shows that you had previously installed a driver which needs to be uninstalled. I'm curious as to where you got this driver? It is amazing to me that a driver that old will even compile.

Confirm your Linux distro and hardware? It appears to be the 32 bit version of RasPiOS on a Raspberry Pi xx?

UntouchedWagons, 2022-02-01:
I think it might be [this driver](https://github.com/gnab/rtl8812au), but I don't remember.

I'm running Raspbian (debian 10) on a Raspberry Pi 3B.

I uninstalled the 8812au driver and will reboot.

UntouchedWagons, 2022-02-01:
No change, the 8821au driver still isn't being used

morrownr, 2022-02-01:
Okay. Go into the driver directory and open `Makefile` in a text editor and paste what these three lines say:

```
CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM64_RPI = n
```

Run and post the results of:

```
lsmod
```

UntouchedWagons, 2022-02-01:
```
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM64_RPI = n
```

```
# lsmod
Module                  Size  Used by
cmac                   16384  1
bnep                   20480  2
hci_uart               40960  1
btbcm                  16384  1 hci_uart
bluetooth             393216  24 hci_uart,bnep,btbcm
ecdh_generic           16384  2 bluetooth
ecc                    40960  1 ecdh_generic
bonding               139264  0
joydev                 20480  0
brcmfmac              327680  0
brcmutil               20480  1 brcmfmac
sha256_generic         16384  0
cfg80211              761856  1 brcmfmac
rfkill                 32768  6 bluetooth,cfg80211
raspberrypi_ts         16384  0
raspberrypi_hwmon      16384  0
bcm2835_codec          40960  0
bcm2835_isp            32768  0
bcm2835_v4l2           40960  0
v4l2_mem2mem           36864  1 bcm2835_codec
bcm2835_mmal_vchiq     32768  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         32768  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
snd_bcm2835            24576  2
videobuf2_common       61440  5 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_pcm               110592  2 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    77824  6 snd_timer,snd_bcm2835,snd_pcm
videodev              249856  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
mc                     45056  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
vc_sm_cma              32768  3 bcm2835_isp,bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
rpi_backlight          16384  0
backlight              20480  1 rpi_backlight
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  495616  22
```

I don't know why the rtl8821au driver isn't showing up, `dkms status` lists it as installed but I can't modprobe it.

morrownr, 2022-02-01:
This is turning into a mystery. Let me suggest the following:

go to the driver directory...

```
sudo ./remove-driver.sh
```

```
sudo apt update
sudo apt full-upgrade
```

```
```
sudo ./install-driver.sh
sudo reboot (or do it with the script)
```

```
lsusb
```

morrownr, 2022-02-01:
Hey @UntouchedWagons 

Disregard the above. I found the problem. The online ad is spreading misinformation.

> ID 0bda:c811

That tells us your chipset is a 8811cu. No wonder this driver is not working.

The following is the driver for the 8811cu:

https://github.com/morrownr/8821cu

Be advised that we currently have a new, much better version of the driver for the 8811cu in work and it should be available in the next few days.

Remember to run `sudo ./remove-driver.sh` for this 8821au driver and then delete the driver directory. You can then go over to the 8821cu driver repo and install. Let me know how it...

Nick

UntouchedWagons, 2022-02-01:
Okay I'll give that one a try.

UntouchedWagons, 2022-02-01:
8821cu worked without a hitch.

closed 2022-02-01 by UntouchedWagons