morrownr/8821au-20210708 #28: Unable to set 4addr on

MTHBlox, 2022-02-01:
Hi, 

Using a RTL8811au chipset wifi dongle, I am unable to enable WDS so I can create a real bridge with the RPI 3B ethernet adapter or internal Wifi adapter.

```
root@raspberrypi:~ # iw wlx1cbfce25609c set 4addr on
command failed: Operation not supported (-95)
```

I've seen this issue posted earlier, but not fixed yet. Any idea how this can be solved?

```
phy#1 --> USB wifi adapter, client connection to router (static ip)
        Interface wlx1cbfce25609c
                ifindex 4
                wdev 0x100000001
                addr 1c:bf:ce:25:60:9c
                type managed
                txpower 42949572.96 dBm
phy#0 --> internal wifi chip, already setup as AP.
        Interface wlxb827eba4746c
                ifindex 3
                wdev 0x1
                addr b8:27:eb:a4:74:6c
                ssid MagicMirror
                type AP
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 31.00 dBm

```
Hope you can help me.

Best regards,

Martijn

morrownr, 2022-02-02:
Hi @MTHBlox 

> I've seen this issue posted earlier, but not fixed yet. Any idea how this can be solved?

I'm pretty I know how to fix this. You probably won't like the answer but here goes...

To test to see if your usb wifi adapter or internal card supports 4addr, execute:

$ iw phy phy1 info | grep wds

When I execute it on one of my mt7612u based usb wifi adapters I get the following:

$ iw phy phy1 info | grep wds
		 * set_wds_peer

The second line indicates support.

So, the fix is to go to the following site and shop for an adapter that supports it:

https://github.com/morrownr/USB-WiFi

Please test and research this issue to crosscheck my info. As far as fixing 4addr in this driver goes... well, that is something Realtek needs to fix. Good luck contacting them. It isn't that I would not welcome someone grabbing the code and submitting a PR but I'd say the chance of that happening is near zero.

Regards

MTHBlox, 2022-02-02:
Hi, thanks for your fast response. Unfortunately no response, so no support.
I Googled the chipset and saw some shops advertising the WDS functionality, but that seems not the case then...
Thanks again, will think of other ways to fix this.

closed 2022-02-02 by MTHBlox