morrownr/8821au-20210708 #29: openSUSE Leap 15.3 installation error

Onirae, 2022-03-06:
install-driver.sh returns error
manual installation gives:
~/src/8821au-20210708> LANG=C make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.18-150300.59.49-default/build M=/home/onir/src/8821au-20210708  modules
make[1]: Entering directory '/usr/src/linux-5.3.18-150300.59.49-obj/x86_64/default'
  CC [M]  /home/onir/src/8821au-20210708/os_dep/linux/ioctl_cfg80211.o
/home/onir/src/8821au-20210708/os_dep/linux/ioctl_cfg80211.c:10477:3: error: 'struct cfg80211_ops' has no member named 'mgmt_frame_register'
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/home/onir/src/8821au-20210708/os_dep/linux/ioctl_cfg80211.c:10477:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/onir/src/8821au-20210708/os_dep/linux/ioctl_cfg80211.c:10477:25: note: (near initialization for 'rtw_cfg80211_ops.mgmt_tx_cancel_wait')
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-5.3.18-150300.59.49/scripts/Makefile.build:288: /home/onir/src/8821au-20210708/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** [/usr/src/linux-5.3.18-150300.59.49/Makefile:1675: _module_/home/onir/src/8821au-20210708] Error 2
make[1]: *** [../../../linux-5.3.18-150300.59.49/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-5.3.18-150300.59.49-obj/x86_64/default'
make: *** [Makefile:2510: modules] Error 2
~/src/8821au-20210708> uname -r
5.3.18-150300.59.49-default

morrownr, 2022-03-06:
Hi @Onirae 

I had to do a little research to figure out why you would be installing to a kernel that is not supported by the kernel maintainers and has not been for some time. It appears that openSUSE Leap is to openSUSE kinda like RHEL is to Fedora... more or less. I guess someone involved in the distro is backporting security patches.

The area that is having a problem was patched for kernel 5.8 so I'll see about checking it out early this week.

What compiler are you using?

$ gcc --version

Regards

Onirae, 2022-03-10:
onir@localhost:~> gcc --version
gcc (SUSE Linux) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

onir@localhost:~>

morrownr, 2022-04-12:
Somehow I missed this reply. What is the status?

closed 2022-04-26 by morrownr

Onirae, 2022-05-11:
I ended using old debian release with kernel 4
Couldn't make it work on any distro with kernel 5