morrownr/8821au-20210708 #30: Support for RTL8832AU

blasco51, 2022-03-16:
I've got the usb wi-fi dongle D-Link DWA-X1850 https://wikidevi.wi-cat.ru/D-Link_DWA-X1850_rev_A1
Being optimistic I simply tried to build the driver by adding the following line to os_dep/linux/usb_intf.c under #ifdef CONFIG_RTL8821A

        {USB_DEVICE(0x2001, 0x3321), .driver_info = RTL8821}, /* D-Link - DWA-X1850 */

When I connect the dongle I see these messages in syslog:
Mar 16 16:22:12 condor kernel: [ 3228.702391] rtl8821au 1-2:1.0 wlx00e04cec2869: renamed from wlan0
Mar 16 16:22:12 condor NetworkManager[983]: <info>  [1647444132.2483] device (wlan0): interface index 8 renamed iface from 'wlan0' to 'wlx00e04cec2869'
Mar 16 16:22:12 condor NetworkManager[983]: <info>  [1647444132.2620] device (wlx00e04cec2869): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Mar 16 16:22:12 condor wpa_supplicant[1019]: Could not set interface wlx00e04cec2869 flags (UP): Operation not permitted
Mar 16 16:22:12 condor wpa_supplicant[1019]: nl80211: Could not set interface 'wlx00e04cec2869' UP
Mar 16 16:22:12 condor wpa_supplicant[1019]: nl80211: deinit ifname=wlx00e04cec2869 disabled_11b_rates=0
Mar 16 16:22:12 condor wpa_supplicant[1019]: Could not set interface wlx00e04cec2869 flags (UP): Operation not permitted
Mar 16 16:22:12 condor wpa_supplicant[1019]: WEXT: Could not set interface 'wlx00e04cec2869' UP
Mar 16 16:22:12 condor wpa_supplicant[1019]: wlx00e04cec2869: Failed to initialize driver interface
Mar 16 16:22:12 condor NetworkManager[983]: <error> [1647444132.2827] sup-iface[0x555bb49c6a00,wlx00e04cec2869]: error adding interface: wpa_supplicant couldn't grab this interface.

Can I do something else to get the driver working for my new dongle?

morrownr, 2022-03-16:
Hi @blasco51 

> Can I do something else to get the driver working for my new dongle?

This driver was not designed for the 8832au so the answer is no.

My recommendation: If you can return the adapter and get your money back, do so immediately.

I posted a report on the 8832au chipset and its driver in Discussions in the USB-WiFi repo:

https://github.com/morrownr/USB-WiFi

FYI: The driver for the mt7921u (wifi 6) is going into the Linux kernel right now. That indicates we will likely see WiFi 6 USB WiFi adapters available soon. The benefits of Linux in-kernel drivers are many.

If you need an adapter now, look at README.md in the above link as it has links to over 75 USB WiFi adapters that use in-kernel drivers. If you have questions, post in Issues in the USB-WiFi repo.

Regards

benpicco, 2022-12-14:
https://github.com/lwfinger/rtl8852au is the driver you are looking for

morrownr, 2022-12-14:
Hi @benpicco 

I'm aware of Larry's repo. I also have the same source code and tried, along with some others, to get it into good enough shape to release to the publiclast year but in the end, we decided the driver was not good enough to release so it sits in hibernation.

closed 2022-12-20 by morrownr

morrownr, 2024-11-07:
Hi @Abhishekghosh1998 

Right now I am working on a the driver for the rtl8852/32cu chip and then I have new source for the rtl8852/32bu so I need to upgrade that driver. So, I can't do this immediately but if you stay in touch and help me find new source, I'll see about doing this.

I tried a couple of years ago to bring up a repo for the rtl8852/32au. The Realtek driver was simply terrible so I gave up at some point. Maybe the driver has been improved since then. The problem I have is that I have no source for the driver for the rtl8852/32au. I get my source from a couple of adapter makers who get it directly from Realtek. They can only get driver source for the chips that they buy and none of my sources use the rtl8852/32au chip in their products. It may be possible to find a relatively new source somewhere but that is where you come in. Can you search for a newer version of the source while I finish up work on my current projects?

Do you mind telling me your country? Maybe we can find a source for adapters that will work better for you.

Abhishekghosh1998, 2024-11-07:
@morrownr Am from India

morrownr, 2024-11-07:
> How is one supposed to search for the source? Can you please tell me that?

Search for sellers of adapters that use the rtl8852au or rtl8832au chips that say they support Linux and get the driver. We would then need to check the date of the driver to see if it is recent. My opinion is that the rtl8852/32au chips will like be abandonware. Realtek has a history of that if a chip is shown to have a problem and my work a couple years ago on the then driver source left me with the opinion that there is a problem in silicone. Then Realtek came out soon after with the rtl8852/32au chips that have the same exact capability. Makes you wonder. I have not seen any new products with the rtl8852/32au chips for a while so it is probably deadware.

> Am from India. 

I am from the central US. 

I looked at your Amazon site. Not much there. Do you have any other online sellers that you order from?

You can see what I am looking for by going to the following link at my site:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

The mt7921au section has adapters that are the same capability as your adapter. The adapters in this list are known to work VERY will with Linux.