morrownr/8821au-20210708 #31: Problems with A6100 on RPI4 64 bit

Musiker, 2022-04-10:
Hi,
I am tinkering with a RPI4 and a A6100, but as soon as I plug in the A6100 things go downhill :-| - Basically any iw and uname command as requested for the issue report cause cause some lockup as seen in the syslog dump below. Anyone have experience with this kind of behavior? Is it my kernel version that is unsupported - Any tips for debugging?

I am using wlan0 for connecting to wlan - So I would like to keep that enabled and then use the A6100 for testing.

Info before plugging in:
```
Linux pi4hyper 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
EFI variables are not supported on this system
Bus 002 Device 002: ID 152d:0562 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 3151:3000 YICHIP Wireless Device
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
rtl8821au, 5.12.5.2, 5.15.32-v8+, aarch64: installed
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr de:a6:32:18:5d:4e
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr dc:a6:32:18:5d:4e
		ssid billebib
		type managed
		channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
		txpower 31.00 dBm
```
Log when plugging in the A6100:
```
Bus 002 Device 002: ID 152d:0562 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 3151:3000 YICHIP Wireless Device
Bus 001 Device 004: ID 0846:9052 NetGear, Inc. A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8821au, 5.12.5.2, 5.15.32-v8+, aarch64: installed
```
syslog:
```

Apr 10 21:11:12 pi4hyper PackageKit: daemon start
Apr 10 21:11:12 pi4hyper dbus-daemon[373]: [system] Successfully activated service 'org.freedesktop.PackageKit'
Apr 10 21:11:12 pi4hyper systemd[1]: Started PackageKit Daemon.
Apr 10 21:11:16 pi4hyper PackageKit: refresh-cache transaction /133_bccaabac from uid 1000 finished with success after 3573ms
Apr 10 21:11:17 pi4hyper dhcpcd[484]: wlan0: no IPv6 Routers available
Apr 10 21:11:20 pi4hyper kernel: [   31.710836] cam-dummy-reg: disabling
Apr 10 21:11:22 pi4hyper PackageKit: get-updates transaction /134_dccbaabc from uid 1000 finished with success after 5761ms
Apr 10 21:11:26 pi4hyper systemd[1]: systemd-fsckd.service: Succeeded.
Apr 10 21:11:36 pi4hyper systemd[1]: systemd-hostnamed.service: Succeeded.
Apr 10 21:17:01 pi4hyper CRON[1641]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Apr 10 21:22:39 pi4hyper kernel: [  711.185171] usb 1-1.3: new high-speed USB device number 4 using xhci_hcd
Apr 10 21:22:39 pi4hyper kernel: [  711.289837] usb 1-1.3: New USB device found, idVendor=0846, idProduct=9052, bcdDevice= 2.00
[... 104 lines cut ...]
Apr 10 21:24:55 pi4hyper kernel: [  846.827882]  raw_notifier_call_chain+0x5c/0x80
Apr 10 21:24:55 pi4hyper kernel: [  846.827902]  call_netdevice_notifiers_info+0x60/0xb0
Apr 10 21:24:55 pi4hyper kernel: [  846.827920]  register_netdevice+0x388/0x400
Apr 10 21:24:55 pi4hyper kernel: [  846.827936]  cfg80211_rtw_add_virtual_intf+0x1e4/0x47c [8821au]
Apr 10 21:24:55 pi4hyper kernel: [  846.828269]  nl80211_new_interface+0x15c/0x448 [cfg80211]
Apr 10 21:24:55 pi4hyper kernel: [  846.828449]  genl_family_rcv_msg_doit.isra.15+0x114/0x158
Apr 10 21:24:55 pi4hyper kernel: [  846.828469]  genl_rcv_msg+0xf8/0x1e0
Apr 10 21:24:55 pi4hyper kernel: [  846.828485]  netlink_rcv_skb+0x64/0x128
Apr 10 21:24:55 pi4hyper kernel: [  846.828500]  genl_rcv+0x40/0x58
Apr 10 21:24:55 pi4hyper kernel: [  846.828515]  netlink_unicast+0x1c0/0x280
Apr 10 21:24:55 pi4hyper kernel: [  846.828530]  netlink_sendmsg+0x1d0/0x418
Apr 10 21:24:55 pi4hyper kernel: [  846.828545]  sock_sendmsg+0x54/0x60
Apr 10 21:24:55 pi4hyper kernel: [  846.828561]  ____sys_sendmsg+0x260/0x2a0
Apr 10 21:24:55 pi4hyper kernel: [  846.828576]  ___sys_sendmsg+0x8c/0xd0
Apr 10 21:24:55 pi4hyper kernel: [  846.828591]  __sys_sendmsg+0x74/0xc8
Apr 10 21:24:55 pi4hyper kernel: [  846.828607]  __arm64_sys_sendmsg+0x2c/0x38
Apr 10 21:24:55 pi4hyper kernel: [  846.828622]  invoke_syscall+0x4c/0x110
Apr 10 21:24:55 pi4hyper kernel: [  846.828640]  el0_svc_common.constprop.3+0xfc/0x120
Apr 10 21:24:55 pi4hyper kernel: [  846.828658]  do_el0_svc+0x2c/0x90
Apr 10 21:24:55 pi4hyper kernel: [  846.828675]  el0_svc+0x24/0x60
Apr 10 21:24:55 pi4hyper kernel: [  846.828693]  el0t_64_sync_handler+0x90/0xb8
Apr 10 21:24:55 pi4hyper kernel: [  846.828710]  el0t_64_sync+0x1a0/0x1a4
Apr 10 21:25:40 pi4hyper systemd-udevd[170]: mon1: Worker [2015] processing SEQNUM=1879 killed

```

morrownr, 2022-04-10:
Hi @Musiker 

I just happened to be testing the new RasPiOS 64 bit on a Pi4B. So I followed the Installation Steps here in the README.md and I could find no problem. In fact this message is going through my rtl8811au usb adapter acting as an AP.

So, same hardware and same OS, different result.

Can I get you to post the Installation Steps that you used?

Also, did you reboot after driver installation was completed?

Regards

Musiker, 2022-04-11:
Hi 
Thanks for the swift reply - I followed the steps in the readme to the point:
1) installed all prerequisites.
2) ran the script to change the arch to ARM64 (Also verified it in Makefile)
3) Used the install script - No errors at all.
had the behavior above, so I uninstalled and tried building with a few different options without luck - And yes I rebooted after each installation. If I plugged in the adapter before boot - The OS did not seem to boot up at all (Atleast I did not wait long enough)

Since it seems to work for you I will hunt the issue by looking at my RPI install. I thought my setup is pretty much out of the box (Some i2c control and wlan0 enabled) - I think I'll try to strip it out and see if I can get it to work at all. Maybe I should even try to do a clean install of the RPI OS.

morrownr, 2022-04-11:
Hi @Musiker 

Interested. I rechecked to make sure your device ID is in the driver. Secure boot cannot be an issue because Pi's don't support it. One comment did get my attention:

>  If I plugged in the adapter before boot - The OS did not seem to boot up at all (Atleast I did not wait long enough)

The first thing that pops into mind here is maybe you are trying to pull too much power via the USB subsystem. Tell me everything you have plugged into USB ports and whether you have a power hub in use. The Pi4B USB subsystem can only supply up to 1200 ma of power.

Using another sd card with a fresh installation to test might be a good idea.

Regards

morrownr, 2022-04-11:
Follow on:

> ID 152d:0562 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge

Somehow I skipped over this. Are you running an ssd directly from the Pi4B? If so, this single device could be taking nearly all available power, especially during boot. If you run an sdd, you really need to run it from an external powered usb hub. You need to be careful exactly which power hub you get a most do not run well with Pi's. I can suggest one if you want.

Musiker, 2022-04-12:
Thanks again, 
You are correct I did have a 256GB SSD (M2) plugged in - I did try without it before reporting and just did again to be sure.. Still the same problem in syslog and when I try to execute e.g. ifconfig the command never returrns or show the state of the network devices - Without the A6100 plugged in it works as expected. 
The PI I tested on sofar also have a Thermal Camera and a 4" HyperPixel display attached, maybe one of these is causing some problem on the supply..

I'll I'll try on on another RPI4 - That is completly barebones and with a fresh install of RPIOS64 as well.. Probably not have time to do that until later this week - I'll update my findings as soon as I get time to test..

morrownr, 2022-04-12:
Yes, testing on another Pi that is running from a sd card is a good idea. I have a little in-line power meter that allows me to test how much power each device uses. I also use a powered hub:

Transcend USB 3.0 4-Port Hub TS-HUB3K

I also lean toward using Mediatek based usn wifi adapters instead of Realtek based as the Mediatek based adapters use less power:

https://github.com/morrownr/USB-WiFi

Hopefully the information I have provided helps.

Regards

morrownr, 2022-04-12:
I''m going to mark this issue solved for now as I am pretty sure your issue is related to trying to pull too much power from the usb subsystem. Please let me know how it goes and if I need to unmark this issue as solved, I will do so.

Regards

Musiker, 2022-04-26:
Finally got around testing on a new RPI4 - without the HyperPixel display mounted and a clean install of the 64 Bit image and it worked without problems.
So it could be the total powerdraw or some difference in installed drivers/sotware - Atleast now I have a working system to compare with.
Thanks for the help!

closed 2022-04-26 by Musiker