morrownr/8821au-20210708 #32: Problem installing Driver on Manjaro

RedmanHunter, 2022-05-06:
Hi,
i´ve used your installation guideline, but i got the following error:

Running install-driver.sh version 20220108
Operating System: Manjaro Linux
5.15.32-1-MANJARO
x86_64
Starting installation...
Copying source files to: /usr/src/rtl8821au-5.12.5.2
Copying 8821au.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
**Error! Your kernel headers for kernel 5.15.32-1-MANJARO cannot be found at /usr/lib/modules/5.15.32-1-MANJARO/build or /usr/lib/modules/5.15.32-1-MANJARO/source.
Please install the linux-headers-5.15.32-1-MANJARO package or use the --kernelsourcedir option to tell DKMS where it's located.
An error occurred. dkms build error = 1
Please report this error.**
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

So, i also did the status list:
Bus 002 Device 005: ID 145f:02e7 Trust Trust QHD Webcam
Bus 002 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 003: ID 2357:011f TP-Link 802.11ac WLAN Adapter 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2/50
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8821au/5.12.5.2: added

Honestly i am a newbie to manjaro an Linux, only know a little commands. And with this, i can´t help out myself. Would you be so kind and tell me what to do?

Thank you so far!

morrownr, 2022-05-06:
Hi @RedmanHunter 

I'll try to help. My primary Linux world is Debian based systems. I keep an installation of Manjaro on a test system but something I know is that when Manjaro users install additional kernels things don't always work out.

The problem is that you don't have the headers for the kernel you are currently using. With my limited Manjaro knowledge, my suggestion is to install it via...

$ sudo pacman -S --noconfirm linux515-headers

If that does not get the installation going then you may want to stop by the appropriate Manjaro forum and ask.

Something else that I am aware of is that Manjaro has this thing called AUR (I think) which is a repo of things like drivers that are prepackaged. I think my driver here is the upstream for the one in AUR so that might be another option.

You will need to be careful around here because it won't take long for me to start trying to convince you Debian based distros are far better than Arch based disros.

Regards

RedmanHunter, 2022-05-07:
Hi @morrownr 

thanks for your help! That solved my problem with it!
> 
> The problem is that you don't have the headers for the kernel you are currently using. With my limited Manjaro knowledge, my suggestion is to install it via...
> 
> $ sudo pacman -S --noconfirm linux515-headers
> 

I am now able to use my Archer TU2 USB WIFI Stick, everythink works fine!

Have a great weekend.
Best regards

morrownr, 2022-05-08:
Hi @RedmanHunter 

That is great! Enjoy.

Regards