morrownr/8821au-20210708 #37: Not working on Kernel above 5.13.0.40

rop12770, 2022-05-30:
Hi

As the tittle says, it doesn't work on more recent kernel versions.
Although it installs OK, it doesn't seem to start.

I have to boot with kernel .40 to work.

Thank You

morrownr, 2022-05-30:
Hi

Post the results of:

$ dkms status

rop12770, 2022-05-30:
Hi

Here it is: nvidia, 510.73.05, 5.13.0-40-generic, x86_64: installed
nvidia, 510.73.05, 5.13.0-41-generic, x86_64: installed
nvidia, 510.73.05, 5.13.0-44-generic, x86_64: installed
rtl8812au, 5.6.4.2: added
rtl88x2bu, 5.8.7.1, 5.11.0-27-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.11.0-46-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-25-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-40-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-41-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-44-generic, x86_64: installed

morrownr, 2022-05-30:
Hmmm... well, I don't see any evidence that this driver has  been installed.

Let's step back and confirm what adapter you have. Please post the results of:

$ lsusb

rop12770, 2022-05-31:
I run the same command as I used to do on other versions of kernel.
sudo ./make
make install
Usually it's enough.

lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 18f8:0f97 [Maxxter] Optical Gaming Mouse [Xtrem]
Bus 003 Device 002: ID 1a2c:0021 China Resource Semico Co., Ltd Keyboard
Bus 003 Device 004: ID 7392:a812 Edimax Technology Co., Ltd Edimax AC600 USB
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2022-05-31:
> Bus 003 Device 004: ID 7392:a812 Edimax Technology Co., Ltd Edimax AC600 USB

That confirms that the driver in this repo, 8821au-20210708, is the one you need.

> rtl8812au, 5.6.4.2: added
rtl88x2bu, 5.8.7.1, 5.11.0-27-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.11.0-46-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-25-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-40-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-41-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-44-generic, x86_64: installed

From above, I see other drivers installed via dkms. They are older versions and for adapters based on other chipsets. Unless you need them, recommend you get rid of them per the instructions from where you downloaded them.

> I run the same command as I used to do on other versions of kernel.
sudo ./make
make install
Usually it's enough.

The manual installation should go like this:

$ make clean
$ make
$ sudo make install
$ sudo reboot

I have a script that automates the manual installation for those that want to use it:

$ sudo ./install-driver-no-dkms-sh

If you are doing a manual installation, recommend you run the following first to clean things up:

$ sudo ./remove-driver.no-dkms.sh

Now, all of the above is based on you having the current version of this driver - v5.12.5.2. If that is not the case, you really should delete the old folder and download the current version.

Something else that might be of interesting is my main site here that has a lot of info about using USB WiFi adapters on Linux. Menu items one and two might be of interest if you want an adapter that just works instead of having to fight with Realtek based adapters:

https://github.com/morrownr/USB-WiFi

Regards

closed 2022-08-27 by morrownr