morrownr/8821au-20210708 #38: Stop wifi

Joarell, 2022-06-04:
Hi there!
I hope you are going great!
First of all I would like to congrats all people evolved on this project you are amazing!
Thank you all! 

So, my issue is a sudden stop of the wifi. I'm using a dungle USB TP-Link Archer T2u AC600 dual band. However, I don't know why the internet connection sudden stop on my Xerolinux OS and when I touch the wifi icon the internet returns to work in a few seconds.
I use the same dungle on Windows and I have no issues at all.
My question to you guys is there a setting on the rtl8821au config file that I could do to sort it out or some?
Thank you in advance!
Be fine out there!

morrownr, 2022-06-04:
Hi @Joarell 

Recommend you install `wavemon` so we can take a detailed look  at signal strength, channel and some other details. You can install and run it on Debian based systems like this:

$ sudo apt install wave
$ wavemon

Since you are running an Arch based system, use the appropriate installation command for Arch.

The most common cause I have seen of issues like this is when your WiFi router / AP has both the 2.4 and 5 GHz bands set to the same name. The best way to handle this is to rename the 5 GHz name to something like `currentname + -5`.

Regards

Joarell, 2022-06-05:
Hi @morrownr!
I've installed the waevmon, but my figured the problem out happened when there is no internet internet traffic in my PC.
It's weird because my internet has different name JessCake and JessCake5G.
But, when I have no internet traffic my connections just gone and returns when I click on wifi icon.
Is it a normal behavior?

I've tried to run `ping google.com `, on background during my studying but the issue still there.
When I try to do some search I have no connection it has dropped down I don't know why.
As I mention before when I do my stuffs on windows it never happen. 

Please, don't get me wrong you have done a great job on this drive and firmware. I just would like to solve this annoying issue and keep learning about the Linux environment.
My kernel version is 5.18.1.

Kindest regards from São Paulo!
[image]

Joarell, 2022-06-05:
Hi @morrownr!
I hope you are doing great!
A new report from today about the internet singnal.
[image]

morrownr, 2022-06-05:
> the problem out happened when there is no internet internet traffic in my PC. It's weird because my internet has different name JessCake and JessCake5G. But, when I have no internet traffic my connections just gone and returns when I click on wifi icon. Is it a normal behavior?

No. If your system is set to go to sleep when inactive, it can take a few seconds to reconnect after waking up but what you are describing is different.

> Please, don't get me wrong you have done a great job on this drive and firmware.

Please don't give me too much credit. This driver is made by Realtek. All I and others do is try to fix problems and keep it working on newer kernels. One of the problems with Realtek and its out-of-kernel drivers is that they are not written in accordance with Linux Wireless Standards.

> I just would like to solve this annoying issue and keep learning about the Linux environment. My kernel version is 5.18.1.

I understand.

Let me suggest a couple more things to try:

I see from your wavemon info that your 5 GHz channel is 132. Try setting 52 and or 100. I think those channels are also DFS channels for you.

You could also try turning power saving off. Go to your driver directory and run:

$ sudo ./edit-options.sh

Read the docs in the file that opens and make the appropriate setting then save and reboot.
> 
> Kindest regards from São Paulo!

Same to you. I have been to São Paulo. I enjoyed it even though I was there to work.

Regards

Joarell, 2022-06-11:
Dear @morrownr,
I hope you are doing great!
I'm glad that you had been here even for short time and busy.
Thank you for your tips and all patience on this issue.

I've been trying over many ways to resolve this wifi problem but, I didn't have a success at all.
I have tried `iwconfig`, I've shift my modem frequency to channel 52 as you recommended, I've revised all OS settings about the energy saving and the issue still were there. 
On last Wednesday I've started to search another drive to test if that problem would persist and boooom!
I found another one very stable working pretty well until this moment. I'll let the link from the repository just in case you would like to take a look.

Finally, sorry for my late reply about this situation. I just would like to ensure that I did my best to make thinks work, maybe my limited knowledge didn't get all possibilities. Even so, I wanted to bring this feedback to you and maybe this same problem could happen with others during the installation and run time. Or perhaps it's only happened with me... Rsrsssrsrsr...

Bellow you find the drive git repository that works for me:
https://github.com/aircrack-ng/rtl8812au

Thank you a lot for all assistance and efforts to keep this project working and updated!
Have a great day and be fine!
Warmest regards from São Paulo!

closed 2022-06-11 by Joarell