morrownr/8821au-20210708 #39: about the endian of 802.11 header

ticty, 2022-06-11:
Hi,

I'm not familiar with 802.11 or usb protocol, and I have a question about the endian.

my PC  is debian and driver build config `CONFIG_PLATFORM_I386_PC=y` and this `-DCONFIG_LITTLE_ENDIAN` is enabled.

In `rtw_recv.c`, when get a raw 802.11 frame, driver use `GetFrameType(ptr)` to get frame type, which is:
```
#define GetFrameType(pbuf)	(le16_to_cpu(*(unsigned short *)(pbuf)) & (BIT(3) | BIT(2)))
```
`le16_to_cpu()` means data read from usb is little endain. While when I read other library that parse 802.11 header, they usually use `ntohs()`

This confuses me, so the header endain is depend on the device self or os system type ?

morrownr, 2022-06-13:
Good question. I don't currently have an answer and am busy with other projects.

An individual that has more experience with Realtek driver is Larry Finger. His site:

https://github.com/lwfinger

ticty, 2022-06-14:
> Good question. I don't currently have an answer and am busy with other projects.
> 
> An individual that has more experience with Realtek driver is Larry Finger. His site:
> 
> https://github.com/lwfinger

Thank you for your information.

I have read the realtek official driver code in linux kernel, and the 802.11 header is also operate in host endian.

So I guess if raw 802.11 packet read from local usb device, then it is just like read from local memory and is in host endian, the driver can omit some caculation of endian translation.
While if read from the air from other station, then it is network endian.

closed 2022-06-14 by ticty