morrownr/8821au-20210708 #44: Can I use 2 wifi adaptors on the my laptop on linux?

LowNightSnack, 2022-07-21:
```
$ sudo uname -a
Linux xfce 5.15.0-41-generic #44~20.04.1-Ubuntu SMP Fri Jun 24 13:27:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ mokutil --sb-state
SecureBoot disabled

$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:0c00 Elan Microelectronics Corp. ELAN:ARM-M4
Bus 001 Device 004: ID 30c9:0013 DJKCVA1WSD73IE HP TrueVision HD Camera
Bus 001 Device 007: ID 2357:0120 TP-Link 802.11ac WLAN Adapter 
Bus 001 Device 006: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ rfkill list all
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

$ dkms status
8812au, 5.6.4.2_35491.20191025, 5.15.0-41-generic, x86_64: installed
rtl8821au, 5.12.5.2, 5.15.0-41-generic, x86_64: installed

$ iw dev
phy#0
	Interface wlo1
		ifindex 3
		wdev 0x1
		addr XX:XX:XX:XX:XX:XX
		type managed
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```
Can I install my TP-Link Adapter on wlo2 interface?

LowNightSnack, 2022-07-21:
```
Bus 001 Device 007: ID 2357:0120 TP-Link 802.11ac WLAN Adapter
```
This device is `TP-Link T2U Plus`
I want this to be on wlo2

morrownr, 2022-07-21:
Hi @LowNightSnack 

> Can I use 2 wifi adaptors on the my laptop on linux?

Yes. The one thing to be aware of is if you try to use 2 adapters based on the same Realtek chipset. That won't work as Realtek's drivers are broken in this regard. You can have several wifi adapters working on a computer and it can be a mix of Mediatek and Realtek based adapters or all Mediatek and it will work as long as no two Realtek adapters have the same chipset.

> $ dkms status
8812au, 5.6.4.2_35491.20191025, 5.15.0-41-generic, x86_64: installed

You need to uninstall the above driver as it has code to support the 8811cu/21cu chipset, the same as this driver so there will be a complict and things won't work as expected.

> Can I install my TP-Link Adapter on wlo2 interface?

It is possible to set and lock a wifi interface name but that is not a function of the driver. I can show you how if you wish.

Regards

LowNightSnack, 2022-07-21:
Thanks!
```
$lspci
...
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter
...
```
So apparently I have a Realtek internal wifi adapter. I want to switch this to my external Realtek wifi adapter. Is there a sequence of commands to let me do that?

Preferably, I would like to turn off wlo1 and turn on wlo2 and vice versa.
(with wlo1 interfacing my internal card and wlo2 interfacing my external adapter)

morrownr, 2022-07-21:
> So apparently I have a Realtek internal wifi adapter. 

It appears that you do. Before we work on anything else, we really need to get rid of that old driver you have installed. Run the following:

$ sudo dkms remove -m 8812au -v 5.6.4.2_35491.20191025 --all

Next run and post the output of:

$ dkms status
$ lsmod

Next reboot and then run and post the output of:

$ iw dev

That is enough for now.

morrownr, 2022-07-22:
> $ dkms status
rtl8821au, 5.12.5.2, 5.15.0-41-generic, x86_64: installed

Good. It appears the old driver has been uninstalled.

It appears that your internal wifi card is support by the in-kernel driver RTW88. The interface leads me to think it is working fine. It can be turned off, we just have to figure out what to blacklist but first, let's try to figure out why an interface for your usb adapter is not showing up.

I see that secure boot is off. It is not clear to me why we are not getting a wifi interface for the adapter. Let check again:

$ sudo update
$ sudo upgrade
Make sure the usb wifi adapter is installed in a usb port.
$ sudo reboot
$ iw dev (port the results)

LowNightSnack, 2022-07-23:
```
$ iw dev
phy#0
	Interface wlo1
		ifindex 3
		wdev 0x1
		addr XX:XX:XX:XX:XX:XX
		ssid XXX
		type managed
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
```
The packages that got upgraded were browsers.

morrownr, 2022-07-23:
Something could have went wrong during the installation so maybe we should start over. Let's uninstall the driver from this repo and try again.

Go to the driver directory and then I'll provide the installation steps here:

cd ~/src/8821au-20210708
sudo ./remove-driver.sh
delete the driver directory and reboot
sudo apt install -y build-essential dkms git
mkdir -p ~/src
cd ~/src
git clone https://github.com/morrownr/8821au-20210708.git
cd ~/src/8821au-20210708
sudo ./install-driver.sh
reboot
iw dev (post the results)

LowNightSnack, 2022-07-23:
```
$ dkms status
rtl8821au, 5.12.5.2, 5.15.0-41-generic, x86_64: installed

$ iw dev
phy#0
	Interface wlo1
		ifindex 3
		wdev 0x1
		addr XX:XX:XX:XX:XX:XX
		ssid XXX
		type managed
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
```

morrownr, 2022-07-23:
Let's try a manual installation:

cd ~/src/8821au-20210708
sudo ./remove-driver.sh
make clean
make
sudo make install
sudo reboot

LowNightSnack, 2022-07-24:
`$ dkms status` shows no drivers
```
$ lsmod | grep rt
xt_addrtype            16384  2
rtw88_8723de           16384  0
rtw88_8723d            73728  1 rtw88_8723de
rtw88_pci              32768  1 rtw88_8723de
rtw88_core            253952  2 rtw88_8723d,rtw88_pci
mac80211             1228800  2 rtw88_pci,rtw88_core
cfg80211              954368  2 rtw88_core,mac80211
parport_pc             53248  0
parport                69632  3 parport_pc,lp,ppdev
x_tables               53248  14 ip6table_filter,xt_conntrack,iptable_filter,ip6table_nat,xt_tcpudp,xt_addrtype,xt_CHECKSUM,ip6_tables,ipt_REJECT,ip_tables,iptable_nat,ip6table_mangle,xt_MASQUERADE,iptable_mangle

$ iw dev
phy#0
	Interface wlo1
		ifindex 3
		wdev 0x1
		addr XXX
		ssid XXX
		type managed
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00

```

morrownr, 2022-07-24:
Did you run the following lines?

```
make clean
make
sudo make install
sudo reboot
```

I was hoping to see the output of the second line so as to see if there are compile errors.

LowNightSnack, 2022-07-24:
```
$ make
ake ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-41-generic/build M=/home/user/src/8821au-20210708  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-41-generic'
  CC [M]  /home/user/src/8821au-20210708/core/rtw_cmd.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_security.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_debug.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_io.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_ioctl_query.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_ioctl_set.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_ieee80211.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_mlme.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_mlme_ext.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_mi.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_wlan_util.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_vht.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_pwrctrl.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_rf.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_chplan.o
  CC [M]  /home/user/src/8821au-20210708/core/monitor/rtw_radiotap.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_recv.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_sta_mgt.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_ap.o
  CC [M]  /home/user/src/8821au-20210708/core/wds/rtw_wds.o
  CC [M]  /home/user/src/8821au-20210708/core/mesh/rtw_mesh.o
  CC [M]  /home/user/src/8821au-20210708/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/user/src/8821au-20210708/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_xmit.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_p2p.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_rson.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_tdls.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_br_ext.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_iol.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_sreset.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_btcoex_wifionly.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_btcoex.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_beamforming.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_odm.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_rm.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_rm_fsm.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_ft.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_wnm.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_mbo.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_rm_util.o
  CC [M]  /home/user/src/8821au-20210708/core/efuse/rtw_efuse.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_roch.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/aes-internal.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/aes-internal-enc.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/aes-gcm.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/aes-ccm.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/aes-omac1.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/ccmp.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/gcmp.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/aes-siv.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/aes-ctr.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/sha256-internal.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/sha256.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/sha256-prf.o
  CC [M]  /home/user/src/8821au-20210708/core/crypto/rtw_crypto_wrap.o
  CC [M]  /home/user/src/8821au-20210708/core/rtw_swcrypto.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/osdep_service.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/os_intfs.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/usb_intf.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/ioctl_linux.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/xmit_linux.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/mlme_linux.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/recv_linux.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/user/src/8821au-20210708/os_dep/linux/rtw_cfgvendor.o
[... 72 lines cut ...]
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/halrf.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/halrf_debug.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/halrf_psd.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/rtl8821a/halhwimg8821a_mac.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/rtl8821a/halhwimg8821a_bb.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/rtl8821a/halhwimg8821a_rf.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/rtl8812a/halrf_8812a_ce.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/rtl8821a/halrf_8821a_ce.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /home/user/src/8821au-20210708/hal/phydm/halrf/rtl8821a/halrf_iqk_8821a_ce.o
[... cut ...] /haltxbfjaguar.o
  CC [M]  /home/user/src/8821au-20210708/hal/btc/halbtc8821a1ant.o
  CC [M]  /home/user/src/8821au-20210708/hal/btc/halbtc8821a2ant.o
  CC [M]  /home/user/src/8821au-20210708/platform/platform_ops.o
  LD [M]  /home/user/src/8821au-20210708/8821au.o
  MODPOST /home/user/src/8821au-20210708/Module.symvers
  CC [M]  /home/user/src/8821au-20210708/8821au.mod.o
  LD [M]  /home/user/src/8821au-20210708/8821au.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-41-generic'
```

morrownr, 2022-07-24:
That is a clean compile, exactly as we want. 

I have no idea why we are not seeing a wifi interface for your usb wifi adapter. Maybe we need to clean house again and reinstall and turn on some logging:

cd ~/src/8821au-20210708
sudo ./remove-driver-no-dkms.sh
make clean
sudo ./install-driver.sh
sudo reboot

Do you know how to turn on the logging?

LowNightSnack, 2022-07-25:
No I dont know how to turn on logging.

I was looking into the official drivers provided. They dont have a linux driver on the website but then I thought that the Mac driver might have more information on the type of hardware on the adapter. I havent investigated properly. It maybe of help?

Leaving the link down below
https://www.tp-link.com/in/support/download/archer-t2u-plus/

morrownr, 2022-07-25:
> No I dont know how to turn on logging.

cd ~/src/8821au-20210708
sudo ./edit-options.sh

In the `options` line, change `rtw_drv_log_level=0` to `rtw_drv_log_level=0`
save and reboot

```
You can save a log file that only includes RTW log entries by running
the following in a terminal:

sudo ./save-log.sh

The name of the log file will be ```rtw.log```.
```

Paste the ciontents of `rtw.log` here.

Tell which distro you are using.

Regards

LowNightSnack, 2022-07-27:
```
 rtw_8723de 0000:02:00.0: enabling device (0000 -> 0003)
 rtw_8723de 0000:02:00.0: Firmware version 48.0.0, H2C version 0
 rtw_8723de 0000:02:00.0 wlo1: renamed from wlan0
 rtw_8723de 0000:02:00.0: start vif XX:XX:XX:XX:XX:XX on port 0
 rtw_8723de 0000:02:00.0: sta YY:YY:YY:YY:YY:YY joined with macid 0
```
The `save-log.sh` did not work. I used this command `dmesg | cut -d"]" -f2- | grep "rtw"`

morrownr, 2022-07-28:
> Linux xfce 5.15.0-41-generic #44~20.04.1-Ubuntu SMP Fri Jun 24 13:27:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Well, that is the kernel, not the OS. It might help if I know the OS (distro).

> 8821au: module verification failed: signature and/or required key missing - tainting kernel

That is not necessarily the problem but I can't say it is not at this point.

LowNightSnack, 2022-07-28:
I am using xubuntu 20

LowNightSnack, 2022-07-28:
I will try to install the driver and operate my adapter in a VM. I'll post the results here. Maybe that will help

morrownr, 2022-07-29:
Did you see the fix for #47 ?

> I am using xubuntu 20

This driver should work for xubuntu 20.04 if it was using the original version of the kernel that shipped with it which I think was 5.4 but you are now using the kernel from Ubuntu 22.04 and many parts of the dev system have changed so this is kind of a hybrid installation which makes it very difficult to work on. Is there any possibility of getting you to backup your important files so you can do a clean installation of Xubuntu 22.04?

Regards

LowNightSnack, 2022-07-29:
```
Description:	Ubuntu 22.04 LTS
```
I updated my xubuntu.
Also i have an internal Realtek wifi card. Is that why the adapter cannot run?
```
❯ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f3:0c00 Elan Microelectronics Corp. ELAN:ARM-M4
Bus 001 Device 005: ID 30c9:0013 Luxvisions Innotech Limited HP TrueVision HD Camera
Bus 001 Device 003: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

❯ iw dev
phy#0
	Interface wlo1
		ifindex 3
		wdev 0x1
		addr XXX
		ssid XXX
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

morrownr, 2022-07-29:
> Also i have an internal Realtek wifi card. Is that why the adapter cannot run?

No. Computers can have several wifi adapters and cards running.

If I have time this afternoon, I am going to reread this thread to see if anything comes to mind.

LowNightSnack, 2022-07-30:
```
$ dkms status
rtl8821au/5.12.5.2, 5.15.0-41-generic, x86_64: installed (WARNING! Diff between built and installed module!)
```

morrownr, 2022-07-30:
>  (WARNING! Diff between built and installed module!)

Never seen that before. Your issue really has me stumped. I'll keep an eye open but, for now, I have no idea what to try next.

LowNightSnack, 2022-08-03:
The router is working with my system now. I have no idea how.
So in that sense, this issue seems to be solved.
SUMMARY:
I had the latest version of the linux kernel (`5.15`) but my system was ubuntu 20 something.
I updated my system from ubuntu 20 to ubuntu 22 and installed the driver and that seems to have worked.

LowNightSnack, 2022-08-03:
Thanks @morrownr ! Greatly appreciate your help