morrownr/8821au-20210708 #46: Doesn't compile in Alpine Linux

KirMozor, 2022-07-27:
My dystrohop turned me on to Alpine Linux, but I can't compile a driver for it
Here's what I did:
```
git clone https://github.com/morrownr/8821au-20210708.git
apk add make g++ linux-headers
cd 8821au-20210708
sh install-driver-no-dkms.sh
```
[image]

morrownr, 2022-07-28:
Hi @KirMozor 

Something seems to be wrong with the development environment.

I am not familiar with Alpine. How about we not use an installation script and go manually to see if that tells us anything:

First cleanup:

$ sudo ./remove-driver-no-dkms.sh

Manual compile:

$ make clean
$ make
$ sudo make install
$ sudo reboot

Please post the output of `make`.

Regards

KirMozor, 2022-07-28:
Alpine Linux is a system with OpenRC and they replaced the g++ compiler with musl

KirMozor, 2022-07-28:
```
alpine:~$ cd ~/Tools/8821au-20210708/
alpine:~/Tools/8821au-20210708$ make clean
#make -C /lib/modules/5.15.55-0-lts/build M=/home/kirill/Tools/8821au-20210708 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
alpine:~/Tools/8821au-20210708$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.55-0-lts/build M=/home/kirill/Tools/8821au-20210708  modules
make[1]: *** /lib/modules/5.15.55-0-lts/build: No such file or directory.  Stop.
make: *** [Makefile:2511: modules] Error 2
alpine:~/Tools/8821au-20210708$ 
```

KirMozor, 2022-07-28:
```
alpine:~/Tools/8821au-20210708$ su
Password: 
alpine:/home/kirill/Tools/8821au-20210708# uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw 
dev
Linux alpine 5.15.55-0-lts #1-Alpine SMP Mon, 18 Jul 2022 06:51:44 +0000 x86_64 Linux
ash: mokutil: not found
Bus 005 Device 001: ID 1d6b:0001
Bus 003 Device 001: ID 1d6b:0001
Bus 001 Device 004: ID 2357:011e
Bus 001 Device 001: ID 1d6b:0002
Bus 004 Device 001: ID 1d6b:0001
Bus 004 Device 002: ID 058f:6362
Bus 002 Device 002: ID 0458:0186
Bus 001 Device 006: ID 0e8d:2004
Bus 002 Device 001: ID 1d6b:0001
rfkill: /dev/rfkill: No such file or directory
ash: dkms: not found
ash: iw: not found
alpine:/home/kirill/Tools/8821au-20210708# 
```

morrownr, 2022-07-28:
> ...and they replaced the g++ compiler with musl

This is a problem. As much as I would like to be everything to everybody, I can't. This is an issue for the makers of Alpine. 

Distro hoping is fine. I regularly test distros but I do so with a long history of using Linux as a background (since '94). Many new distro makers have a lot of great ideas... or so they think but it is not a great idea if it breaks some things without providing big gains.

One option you could look at is to go with a USB WiFi adapter that uses in-kernel drivers. The Main Menu for this site follows and menu item 2 is where the adapters are shown:

https://github.com/morrownr/USB-WiFi

Regards

KirMozor, 2022-07-29:
Although g++ was replaced by musl, but g++ can still be installed, because the developers are aware of the problems that the user may encounter

KirMozor, 2022-07-29:
I think the problem is that headers and kernel in the repositories are different versions (I don't know how they let this happen)
I will try to build from source linux-headers of the correct version, maybe help
[image]
[image]

KirMozor, 2022-07-29:
although I think it would be better to compile the kernel from source

morrownr, 2022-07-29:
> Although g++ was replaced by musl, but g++ can still be installed, because the developers are aware of the problems that the user may encounter

There is more to it than g++. Your system is likely missing various supporting packages. Here are some links to information regarding the `build-essentials` package in Debian/Ubuntu. You will have to sort out and install as appropriate:

https://linuxhint.com/install-build-essential-ubuntu/#:~:text=The%20build%2Dessentials%20packages%20are,required%20to%20compile%20a%20program.

https://packages.debian.org/bullseye/build-essential

morrownr, 2022-08-20:
@KirMozor 

Trying to do some house cleaning. Were you able to resolve this?

Nick

KirMozor, 2022-08-20:
> 
Unfortunately, no😢
I gave up on Alpine Linux because programs strongly tied to glibc do not run.

morrownr, 2022-08-20:
Ah, that would explain the Void PR. How is Void working out?

KirMozor, 2022-08-25:
> Ah, that would explain the Void PR. How is Void working out?

Everything is great! True, I do not know what it is, maybe void, maybe the updates, but it happens that the system is permanently frozen when I connect to wifi, the problem is solved only by reinstalling, this problem appears after updating the system

morrownr, 2022-08-25:
This sounds like a new kernel is flowing in that has changes that the driver needs to know about. Are using the dkms version of the install script?

KirMozor, 2022-08-25:
Yes, I use the version with dkms

morrownr, 2022-08-25:
How about we remove the installation using ./remove-driver.sh and then use the manual installation instructions?

This might help us see what the problem is.

KirMozor, 2022-08-25:
In case it hangs again, I will try this option and report back here
Or maybe create a separate issue?

closed 2022-08-26 by morrownr