morrownr/8821au-20210708 #47: Could not install with Kernel 5.15

jonghkim, 2022-07-27:
Hi. I have tried to install the driver on kernel (5.15.0-41-generic), but it doesn't seem to work.

Thank you.

```
sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

Linux jongvis 5.15.0-41-generic #44-Ubuntu SMP Wed Jun 22 14:20:53 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot disabled
Platform is in Setup Mode
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 002 Device 002: ID 1058:25e2 Western Digital Technologies, Inc. My Passport (WD40NMZW)
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 05ac:024f Apple, Inc. Aluminium Keyboard (ANSI)
Bus 001 Device 004: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
Bus 001 Device 002: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
rtl8821au/5.12.5.2, 5.15.0-41-generic, x86_64: installed

```

morrownr, 2022-07-27:
Hi @jonghkim 

This driver is normally a very reliable driver but now I see this issue and am already working on #44 which seems to be similar.

Your title is a little misleading. The driver is installed. it just is not working. Give me a change to dig out my 8811au adapter and get my dev machine on kernel 5.15 and I will see if I can figure something out.

Nick

jonghkim, 2022-07-28:
Thanks for the quick reply. I updated the title.

Best,
Jongho

morrownr, 2022-07-28:
I returned to kernel 5.15 and did a clean installation with my 8811au adapter. The driver appears to be working fine but I will use it for a couple of days to see what I can figure out.

jonghkim, 2022-07-28:
Hi @morrownr 

Great 👍 Thanks a lot for the quick update!

jonghkim, 2022-07-28:
After updating Ubuntu using Software & Update, now the driver starts to work! 

Specifically, I changed my updates option from "Security and recommended updates" to "All updates".

Thanks.

Best,
Jongho

morrownr, 2022-07-28:
That's great. I'll mark this issue as solved.

Since you are a USB WiFi user, here is the link to the Main Menu here. It contains a lot of good information:

https://github.com/morrownr/USB-WiFi