morrownr/8821au-20210708 #48: rtl8821au or RT5370??

akhil3417, 2022-08-01:
sry i know this proably not the right place but , i gotta ask i bought  tp link t2u plus (rtl8821au) , should i return it and get the one with rt5370?

tp link has both 2.4 and 5 .0 hz support while other doesn't, they both cost the same, and other options are too costly in my region?

morrownr, 2022-08-01:
Hi @akhil3417 

> tp link has both 2.4 and 5 .0 hz support while other doesn't, they both cost the same, and other options are too costly in my region?

If you have a good 2.4 Ghz signal and it meets your needs, the answer is easy for me, I would get the adapter with the rt5370. The reason is that the rt5370 has a good, full function in-kernel driver that works well. You never have to mess with compiling an out-of-kernel driver which is problematic.

Have you had a chance to look at the long list of adapters that use in-kernel drivers? See item 2 at the Main Menu:

https://github.com/morrownr/USB-WiFi

akhil3417, 2022-08-02:
thanks for the insight, i do have read your post about the subject and few more , rt5370 seems great option.

closed 2022-08-02 by akhil3417