morrownr/8821au-20210708 #49: Unable to build on Arch with kernel 5.19.2

margual56, 2022-08-18:
```C
rtl8812au/os_dep/linux/ioctl_cfg80211.c: At top level:
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10133:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
10133 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10133:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10219:24: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)’ [-Werror=incompatible-pointer-types]
10219 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10219:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10398:17: error: ‘struct wireless_dev’ has no member named  current_bss’
10398 |         if (wdev->current_bss) {
      |                 ^~
```
OS: Arch Linux x86_64
Kernel: 5.19.2
Device: RTL8821AU

It worked perfectly with kernel 5.19.1

Same issue with [https://github.com/aircrack-ng/rtl8812au](https://github.com/aircrack-ng/rtl8812au) ([this](https://github.com/aircrack-ng/rtl8812au/issues/993) issue)

morrownr, 2022-08-18:
Hi @margual56 

I am aware of the issue. This issue just appeared as kernel 5.19.1 was okay. I am testing a patch right now.

Nick

morrownr, 2022-08-18:
Patch applied. Can I get you to test and report?

margual56, 2022-08-18:
Can confirm it compiles and installs without problems. 

I'm getting 156 Mbits/sec on my LAN, so no regressions. Good job! And thank you very much 😄 

As far as I'm concerned, we can close the issue.

morrownr, 2022-08-18:
Okay. Thanks for the report.

How about we leave it open for now as there will likely be others that have the unpatched version that are going to upgrade to 5.19.2 or 6.0 and if they see this msg they will know to clean the old version old and reinstall.

Nick

margual56, 2022-08-18:
Great idea! Thank you for the fast response Nick :)

karpaz, 2022-08-19:
I'm another guy here who's been helped by your issue, thanks ! :)

buildershed, 2022-08-29:
is the patch likely to be pushed out to this git repo?

morrownr, 2022-08-29:
Hi @buildershed 

I'm not sure if I understand the question. The patch has already been merged here.