morrownr/8821au-20210708 #51: No interface showing on Raspi 2 (Raspbian GNU/Linux 11)

blue112, 2022-08-27:
> sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
```
Linux raspberrypi 5.15.56-v7+ #1575 SMP Fri Jul 22 20:28:11 BST 2022 armv7l GNU/Linux
-bash: mokutil: command not found
Bus 001 Device 009: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl88x2bu, 5.13.1, 5.15.56-v7+, armv7l: installed
```

After installation, no interface shows in ifconfig or iw dev.
No error in the the kernel log.

morrownr, 2022-08-27:
Hello @blue112 

Okay, let me see... latest 32 bit RasPiOS... secure boot is not an issue on RasPi's... device ID checks, it needs the 8821au driver and this repo contains said driver... hmmm...

> rtl88x2bu, 5.13.1, 5.15.56-v7+, armv7l: installed

That line tells me that you installed the 882xbu driver which is for rtl8812bu or 8822bu chipsets. If you have another adapter based on one of those chipsets, it is fine to have both drivers installed but what I am not seeing a line telling me that you installed the driver from this repo... which would cause `iw dev` to show nothing if all you have in the usb port is a 8811au based adapter.

Does this make sense?

You can use the `./remove-driver.sh` script in the 882xbu directory to remove that driver, unless you have one of those adapters in addition to the 8811au based adapter. After the script you would simple delete the /src/88x2bu directory and then install the driver from this repo and things should work.

Let me know,

Nick

blue112, 2022-08-27:
Woops, looks like I cloned the wrong repository :blush: 

It works now, thanks a lot for the support !

morrownr, 2022-08-27:
Glad it is working. I'll mark the issue as solved.

Nick