morrownr/8821au-20210708 #52: Linux mint 21 xfce with Archer T2U nano can't connect to 5ghz of Tplink arhcer ax72 router.

ttx99, 2022-08-30:
I am only able to connect to 2.4ghz, when I connect to 5ghz it just keep on asking for  password. I made a detailed description on Tplink routers forum because I don't know if its the wifi dongle or the router that has the problem. 
https://community.tp-link.com/us/home/forum/topic/578692

morrownr, 2022-08-30:
Hi @ttx99 

Please post the results of the following commands:

$ lsusb
$ dkms status
$ iw reg get

Regards

morrownr, 2022-08-30:
Delete all of your WiFi connections and start fresh:

[image]

ttx99, 2022-08-30:
no luck still cant connect

morrownr, 2022-08-30:
Why is the 2.4G connection listed?

morrownr, 2022-08-30:
Are you certain that the password you put in the router for 5G is the same as the one you are using to connect?

There is a section in the README here that talks about router settings. You might want to take a look.

ttx99, 2022-08-30:
> Why is the 2.4G connection listed?

i dont know whatt you mean by this. I connected to it after failing to connect 5ghz.

ttx99, 2022-08-30:
> Are you certain that the password you put in the router for 5G is the same as the one you are using to connect?
> 
> There is a section in the README here that talks about router settings. You might want to take a look.

yes, password is the same for both. I'm trying to find the readme.

morrownr, 2022-08-30:
>  I'm trying to find the readme.

It is the primary page that you see when you enter the repo.

Consider changing the channel from 48 to 36.

What security is the 5G connection using? Don't use WPA3 or WPA3/WPA2 mixed.

> I connected to it after failing to connect 5ghz.

Understand. Keep in mind that once you have 5G working, the system may try to move to 2G, if you leave a connection, since it will be a stronger signal.

ttx99, 2022-08-30:
Fixed, thanks. It was either this 

Channels for 5 GHz: Not all devices are capable of using DFS channels (I'm looking at you Roku.) It may be necessary to set a fixed channel in the range of 36 to 48 or 149 to 165 in order for all of your devices to work on 5 GHz. (For US, other countries may vary.)

or 

The freaking usb 3.1 port.

closed 2022-08-30 by ttx99

morrownr, 2022-08-30:
Good. The README also contains a section on USB. I have tried to put a lot of good info in the README.

Nick

morrownr, 2022-08-30:
Before I go... the Main Menu for the site is located at:

https://github.com/morrownr/USB-WiFi

It contains LOT of good information for USB WiFi users.