morrownr/8821au-20210708 #53: Netgear ac600 wont use the driver

Azertick, 2022-09-15:
I have installed the driver 3 times on new installs and on 2 different sd cards and I still cannot get it to work with my Netgear ac600 the driver installs normally and after installing the device isn't used for wlan

morrownr, 2022-09-15:
Post the results of:

$ lsusb

Azertick, 2022-09-15:
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver 
Bus 001 Device 005: ID 0846:9052 NetGear, Inc. A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU] 
Bus 001 Device 006: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub 
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2022-09-15:
> ID 0846:9052 NetGear, Inc. A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU]

That confirms this is the correct driver.

What hardware and distro are you running?

Azertick, 2022-09-15:
I'm sorry I forgot to add this earlier
Distro
ARM32 Raspbian OS Version 11
Hardware
Raspberry Pi 3b+

Result of command

5.15.61-v7+ armv7l
bash: mokutil: command not found
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 0846:9052 NetGear, Inc. A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU]
Bus 001 Device 006: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
rtl8812au, 5.13.6, 5.15.61-v7+, armv7l: installed (WARNING! Diff between built and installed module!)
rtl8814au, 5.8.5.1, 5.15.61-v7+, armv7l: installed
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr 1e:6f:a2:12:c3:a2
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:84:aa:d3
		ssid MCPSWIFI
		type managed
		channel 56 (5280 MHz), width: 20 MHz, center1: 5280 MHz
		txpower 31.00 dBm
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, NO-IR
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)

Azertick, 2022-09-15:
I think I found my issue

morrownr, 2022-09-15:
Ah, you are running the RasPiOS, 32 bit. Did you run the 32 bit script in Installation Step 9?

Remember to run the following if you have to reinstall:

$ sudo ./remove-driver.sh

Azertick, 2022-09-16:
Yes, I used the 32-bit command before running make.

morrownr, 2022-09-16:
> rtl8812au, 5.13.6, 5.15.61-v7+, armv7l: installed (WARNING! Diff between built and installed module!)
> rtl8814au, 5.8.5.1, 5.15.61-v7+, armv7l: installed

I was going over the details again and noticed the about. Those are what dkms is showing as installed. Neither are the driver in this repo. Are you using `./install-driver.sh` for installation?

Azertick, 2022-09-16:
I found the issue it was a user error, thank you for your help.

closed 2022-09-16 by Azertick

morrownr, 2022-09-16:
You are welcome.