morrownr/8821au-20210708 #54: Not running on kernel 5.15

Sankalp1Singh, 2022-09-18:
Running install-driver.sh version 20220705
5.15.32-v7l+
armv7l
Starting installation...
Copying source files to: /usr/src/rtl8821au-5.12.5.2
Copying 8821au.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source ->
                 /usr/src/rtl8821au-5.12.5.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.15.32-v7l+ cannot be found at
/lib/modules/5.15.32-v7l+/build or /lib/modules/5.15.32-v7l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
An error occurred. dkms build error = 1
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

morrownr, 2022-10-18:
Hi @Sankalp1Singh 

Sorry for missing this issue. Sometimes I get too busy. What is the status?

yelled1, 2022-10-26:
Having the same problem:
Ok here's what I think is the issue:

https://askubuntu.com/questions/714874/how-to-point-dkms-to-kernel-headers
` ln -s /usr/src/linux-headers-$(uname -r)  /lib/modules/$(uname -r)/build`
Should do the trick

Well here's the problem 
```
$ uname -r
5.15.0-52-generic

$ ls -ld  /usr/src/linux-headers-*
drwxr-xr-x 24 root root 4096 Oct 26 10:24 /usr/src/linux-headers-5.4.0-131
drwxr-xr-x  7 root root 4096 Oct 26 10:25 /usr/src/linux-headers-5.4.0-131-generic
```

Elementary OS 6.1 never downloaded the headers

/usr/lib/modules/5.15.0-52-generic is the likely candidate I am looking for (from `$ find / -iname 5.15.0-52`)

I guess I will just see if I can reinstall the kernel & pray
Let me know if you have suggestions & I will repost/add when I get somewhere.

yelled1, 2022-10-26:
Going back to 5.4 for now, but hit
```
Warning: Please don't use old title `elementary OS 6.1 Jólnir, with Linux 5.4.0-131-generic' for GRUB_DEFAULT, use `Advanced options for elementary OS 6.1 Jólnir>elementary OS 6.1 Jólnir, with Linux 5.4.0-131-generic' (for versions before 2.00) or `gnulinux-advanced-976535d5-bece-4781-936f-9ba7054df58c>gnulinux-5.4.0-131-generic-advanced-976535d5-bece-4781-936f-9ba7054df58c' (for 2.00 or later)
Adding boot menu entry for UEFI Firmware Settings
done
```
updated grub accordingly

morrownr, 2022-10-26:
Hi @yelled1 

> Elementary OS 6.1 never downloaded the headers

What core distro is Elementary based on? Debian, Arch, Fedora?

The thing that pops to mind here is that we probably need to be thinking about kernels and header files as two separate things. Installing a kernel does not install header files. From reading your post, I could see no indication that the 5.15 header files have been installed. 

Once I know what core distro Elementary is based on, then I will know how header files are handled. It seems the leadership of all distros think they have the best idea when it comes to header files. This leads to a question: In the installation instructions here, since instructions for Elementary are not listed, what did you do to install the header files? if anything?

Nick

yelled1, 2022-10-26:
This is based on Ubuntu 20.04 LTS

morrownr, 2022-10-26:
> This is based on Ubuntu 20.04 LTS

Ah, so Debian > Ubuntu > Elementary

Now that I have the genetic background I can get you on the right track.

I did a little checking. Elementary 6.1 is based on Ubuntu 20.04. Ubuntu 20.04.5 flowed the 5.15 kernel into Ubuntu 20.04 so an updated Ubuntu 20.04 should be running 5.15. I don't know if Elementary tracks with and includes the point releases of Ubuntu. How did you upgrade to 5.15?

yelled1, 2022-10-26:
Just did the usual `apt update; apt upgrade` to get the kernels
Those msgs are usually blurr. So, did not really pay attention to it too much

morrownr, 2022-10-26:
Did you run the following from the installation instructions while 5.15 was active?

$ sudo apt install -y build-essential dkms git

The build-essential package should install the headers.

yelled1, 2022-10-26:
I did `build-essential` again to just to make sure & did usual `update & upgrade`
but no dice, so tried below again
```
$ sudo apt-get install --reinstall linux-image-generic linux-image-5.15.0-52-generic
Reading package lists... Done
Building dependency tree       
Reading state information... Done
 ... (snip)
Setting up linux-image-generic (5.4.0.131.131) ...
Setting up linux-image-5.15.0-52-generic (5.15.0-52.58~20.04.1) ...
Processing triggers for linux-image-5.15.0-52-generic (5.15.0-52.58~20.04.1) ...
/etc/kernel/postinst.d/dkms:
 * dkms: running auto installation service for kernel 5.15.0-52-generic
Error! echo
Your kernel headers for kernel 5.15.0-52-generic cannot be found at
/lib/modules/5.15.0-52-generic/build or /lib/modules/5.15.0-52-generic/source.
```
** but the solution is likely **
$ sudo apt install --reinstall linux-headers-5.15.0-52-generic
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  linux-hwe-5.15-headers-5.15.0-52
The following NEW packages will be installed:
  linux-headers-5.15.0-52-generic linux-hwe-5.15-headers-5.15.0-52
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 14.8 MB of archives.
After this operation, 104 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up linux-headers-5.15.0-52-generic (5.15.0-52.58~20.04.1) ...
... (snip)
/etc/kernel/header_postinst.d/dkms:
 * dkms: running auto installation service for kernel 5.15.0-52-generic

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.15.0-52-generic KSRC=/lib/modules/5.15.0-52-generic/build...............
Signing module:
 - /var/lib/dkms/rtl8821au/5.12.5.2/5.15.0-52-generic/x86_64/module/8821au.ko
Nothing to do.
cleaning build area...

DKMS: build completed.

8821au.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-52-generic/extra/

depmod...

DKMS: install completed.
   ...done.
hyoon@Z370XPSLIElem61:~$ ls -ld  /usr/src/linux-headers-*
drwxr-xr-x  7 root root 4096 Oct 26 13:00 /usr/src/linux-headers-5.15.0-52-generic
drwxr-xr-x 24 root root 4096 Oct 26 10:24 /usr/src/linux-headers-5.4.0-131
drwxr-xr-x  7 root root 4096 Oct 26 10:25 /usr/src/linux-headers-5.4.0-131-generic
```
Well report back if this works. Think it will

yelled1, 2022-10-26:
** Eureka **
Above  worked -> `$ sudo apt install --reinstall linux-headers-5.15.0-52-generic`
@morrownr  thank you much for your help!

morrownr, 2022-10-26:
Great! Glad you are going.

I'll mark the issue solved but let's leave it open in case someone else has the problem.

Nick

yelled1, 2022-10-26:
**Sure hope it benefits many**
& here's how I removed the previous kernel's module (for those who might have followed exactly my steps)
```
$ sudo dkms status
[sudo] password for hyoon:           
rtl8821au, 5.12.5.2, 5.15.0-52-generic, x86_64: installed
rtl8821au, 5.12.5.2, 5.4.0-131-generic, x86_64: built

$ sudo dkms uninstall rtl8821au/5.12.5.2 -k 5.4.0-131-generic

$ sudo dkms remove rtl8821au/5.12.5.2 -k 5.4.0-131-generic
-------- Uninstall Beginning --------
Module:  rtl8821au
Version: 5.12.5.2
Kernel:  5.4.0-131-generic (x86_64)
-------------------------------------
Status: This module version was INACTIVE for this kernel.
depmod....
DKMS: uninstall completed.
```