morrownr/8821au-20210708 #55: rtl8821au can't detect any SSID if compiled with Clang/LLVM 15

King-Mucus, 2022-09-27:
I'm using Musl here. It worked fine with Clang/LLVM 14 until at least kernel 5.19.6

Device ID is:
ID 2357:011f TP-Link 802.11ac WLAN Adapter

King-Mucus, 2022-10-15:
```
[   29.003441] RTW: TXPWR: [A][20MHz]ch:1, OFDM_6M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.009439] RTW: TXPWR: [A][20MHz]ch:1, OFDM_9M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.015439] RTW: TXPWR: [A][20MHz]ch:1, OFDM_12M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.021439] RTW: TXPWR: [A][20MHz]ch:1, OFDM_18M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.027332] RTW: TXPWR: [A][20MHz]ch:1, OFDM_24M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.033440] RTW: TXPWR: [A][20MHz]ch:1, OFDM_36M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.039440] RTW: TXPWR: [A][20MHz]ch:1, OFDM_48M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.045441] RTW: TXPWR: [A][20MHz]ch:1, OFDM_54M 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.051441] RTW: TXPWR: [A][20MHz]ch:1, MCS0 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.057442] RTW: TXPWR: [A][20MHz]ch:1, MCS1 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.063442] RTW: TXPWR: [A][20MHz]ch:1, MCS2 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.069443] RTW: TXPWR: [A][20MHz]ch:1, MCS3 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.075443] RTW: TXPWR: [A][20MHz]ch:1, MCS4 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.081443] RTW: TXPWR: [A][20MHz]ch:1, MCS5 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.087443] RTW: TXPWR: [A][20MHz]ch:1, MCS6 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.093443] RTW: TXPWR: [A][20MHz]ch:1, MCS7 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.099443] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS0 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.105444] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS1 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.111444] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS2 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.117445] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS3 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.123446] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS4 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.129445] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS5 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.135446] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS6 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.141447] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS7 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.147447] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS8 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.153447] RTW: TXPWR: [A][20MHz]ch:1, VHT1SMCS9 1T, pwr_idx:48 = 46 + (0=0:63) + (0) + (0)
[   29.165451] RTW: rtw_hal_set_macaddr_port wlan0- hw port(0) mac_addr =c0:06:c3:03:87:5f
[   29.193450] RTW: hw_var_set_opmode()-4338 mode = 2
[   29.211451] RTW: MAC Address = c0:06:c3:03:87:5f
[   29.211454] RTW: rtw_start_drv_threads(wlan0) enter
[   29.211455] RTW: rtw_start_drv_threads(wlan0) start RTW_CMD_THREAD
[   29.211618] RTW: rtw_cfg80211_init_wiphy:rf_type=0
[   29.211621] RTW: [HT] HAL Support LDPC = 0x02
[   29.211622] RTW: [HT] HAL Support STBC = 0x01
[   29.211623] RTW: [HT] HAL Support LDPC = 0x02
[   29.211623] RTW: [HT] HAL Support STBC = 0x01
[   29.211624] RTW: [VHT] Support LDPC = 0x02
[   29.211625] RTW: [VHT] Support STBC = 0x01
[   29.211625] RTW: [VHT] Support Beamformee
[   29.211626] RTW: rtw_build_vht_cap_ie, line818, Set MAX MPDU len = 11454 bytes
[   29.211627] RTW: [VHT] Declare supporting SGI 80MHz
[   29.211628] RTW: [VHT] Declare supporting RX STBC = 1
[   29.211629] RTW: [VHT] Declare supporting SU Bfee
[   29.211633] RTW: -871x_drv - drv_open, bup=1
[   29.211685] RTW: cfg80211_rtw_set_power_mgmt(wlan0) enabled:0, timeout:-1
[   29.211835] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   29.212285] RTW: cfg80211_rtw_change_iface(wlan0) type=2, hw_port:0
[   29.212287] RTW: cfg80211_rtw_change_iface(wlan0) call netdev_open
[   29.212288] RTW: _netdev_open(wlan0) , bup=1
[   29.212289] RTW: -871x_drv - drv_open, bup=1
[   29.212290] RTW: cfg80211_rtw_change_iface(wlan0) old_iftype=2, new_iftype=2
[   29.212306] RTW: rtw_hal_set_macaddr_port wlan0- hw port(0) mac_addr =c0:06:c3:03:87:5f
[   29.239355] RTW: hw_var_set_opmode()-4338 mode = 2
[   29.257401] RTW: cfg80211_rtw_change_iface(wlan0) ret:0
[   29.257405] RTW: cfg80211_rtw_get_txpower
[   29.257431] RTW: cfg80211_rtw_scan(wlan0)
[   29.257434] RTW: rtw_ps_deny(wlan0): [WARNING] Reason 1 had been set before!!
[   29.266357] RTW: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)TXBUF Empty(1) in 9 ms
[   29.272358] RTW: wlan0 sleep m0=0x00000002, ori reg_0x4d4=0x00000000
[   29.335452] RTW: wlan0 wakeup m0=0x00000002, ori reg_0x4d4=0x00000002
[   29.338378] RTW: survey done event(0) band:0 for wlan0
[   29.338385] RTW: rtw_indicate_scan_done(wlan0)
[   29.338554] RTW: cfg80211_rtw_scan(wlan0)
[   29.338558] RTW: rtw_ps_deny(wlan0): [WARNING] Reason 1 had been set before!!
[   29.347385] RTW: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)TXBUF Empty(1) in 9 ms
[   29.353362] RTW: wlan0 sleep m0=0x00000002, ori reg_0x4d4=0x00000000
[   29.416813] RTW: wlan0 wakeup m0=0x00000002, ori reg_0x4d4=0x00000002
[   29.419369] RTW: survey done event(0) band:0 for wlan0
[   29.419377] RTW: rtw_indicate_scan_done(wlan0)
[... 86 lines cut ...]
[  198.652719] RTW: rtw_indicate_scan_done(wlan0)
[  238.652805] RTW: cfg80211_rtw_scan(wlan0)
[  238.652821] RTW: rtw_ps_deny(wlan0): [WARNING] Reason 1 had been set before!!
[  238.661152] RTW: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)TXBUF Empty(1) in 8 ms
[  238.667153] RTW: wlan0 sleep m0=0x00000002, ori reg_0x4d4=0x00000000
[... cut ...]
[  478.810928] RTW: cfg80211_rtw_scan(wlan0)
[  478.810936] RTW: rtw_ps_deny(wlan0): [WARNING] Reason 1 had been set before!!
[  478.819919] RTW: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)TXBUF Empty(1) in 9 ms
[  478.825867] RTW: wlan0 sleep m0=0x00000002, ori reg_0x4d4=0x00000000
[  478.888871] RTW: wlan0 wakeup m0=0x00000002, ori reg_0x4d4=0x00000002
[  478.891872] RTW: survey done event(0) band:0 for wlan0
[  478.891877] RTW: rtw_indicate_scan_done(wlan0)
```

This is the dmesg output I keep getting.

morrownr, 2022-10-15:
Hi @King-Mucus 

I don't use Clanf/LLVM but another user in another repo here had the same problem and this appeared to be the answer:

`do EXPORT LLVM=1 && EXPORT LLVM_IS=1 and it is going to compile`

Let me know if it works as I may need to add this to the FAQ.

Nick

morrownr, 2022-10-18:
Status?

King-Mucus, 2022-10-19:
It made no difference. It was already compiling and loading fine when I asked. I tried exporting/setting those variables as well, but it didn't change the module's behavior.

If you'd like to check the complete build log, please refer to [this link](https://matrix.org/_matrix/media/r0/download/matrix.org/rzTOtZlDCKQNPWHQcxMTiHqu/unpatched.log).

It is worth mentioning that dmesg repeatedly outputs "Reason 1 has already been set!!", so I believe it bears some relation to that specific bit of code.

morrownr, 2022-10-19:
> If you'd like to check the complete build log, please refer to this link.

That build log seems to indicate that the 8812au driver you are building is: 

Package:    net-wireless/rtl8812au_aircrack-ng-9999:0

If that is the case, then you need to seek support from the aircrack-ng folks because that is not the driver I have up here.

morrownr, 2022-10-19:
Would you mind pasting the output of:

$ lsusb

There is something going on here. Not deteching SSIDs can also be attributed to installing the wrong driver. We need to know what chipset you have.

King-Mucus, 2022-10-19:
`Bus 001 Device 003: ID 2357:011f TP-Link 802.11ac WLAN Adapter `

King-Mucus, 2022-10-19:
I understand. I posted that because the build log for both drivers seem pretty similar, and both present the same issue. However, I will be able to rebuild your driver for the 6.0.1 kernel in a few hours, and I will be sure to upload the proper log. Thank you very much for pointing this out.

The device ID is specified in https://github.com/morrownr/8821au-20210708/blob/5a92bf29fdb4fb6dc01ac16396843bee96d21660/supported-device-IDs. As a matter of fact, I'm using it right now on a 5.19.6 kernel, but the kernel and the driver have been compiled with Clang/LLVM 14. The issue was also present on 5.19.9. As the update from 5.19.6 to 5.19.9 seems pretty minor, I believe the real cause lies in the different compiler.

```
$ lsmod
Module                  Size  Used by
wmi_bmof               16384  0
8821au               2301952  0
```

```
$ modinfo 8821au
filename:       /lib/modules/5.19.6-gentoo/net/wireless/8821au.ko.zst
license:        GPL
description:    Realtek Wireless Lan Driver
author:         Realtek Semiconductor Corp.
version:        v5.12.5.2-0-g70054197b.20210708_COEX20190509-6d6f
vermagic:       5.19.6-gentoo SMP preempt mod_unload 
name:           8821au
retpoline:      Y
depends:
```

morrownr, 2022-10-19:
> I will be able to rebuild your driver for the 6.0.1 kernel in a few hours

This driver seems to be working well with 6.0.1 here with gcc. Right now I am working on upgrades for 6.1... well, actually this driver is already finished for 6.1. I still have other drivers to work on.

I probably could be convinced to install Clang/LLVM but not this week.

King-Mucus, 2022-10-19:
[morrownr.log](https://github.com/morrownr/8821au-20210708/files/9824419/morrownr.log)

This is the build log. Note, however, that I just updated LLVM and Clang to 5.0.2 to 5.0.3 and I still hadn't the chance to test the driver compiled with this new version, nor the commits you made in the last few hours.

Thank you very much for your attention, it is great to make possible issues known. My setup is a bit obscure, based on LLVM/Clang and Musl, so some things might be a bit difficult to replicate outside a niche environment. However, they worked pretty perfectly until very recently.

King-Mucus, 2022-10-19:
Yup. Same issue, even with kernel 6.0.2 and everything compiled with Clang/LLVM 15.0.3.

The following file contains the dmesg output:

[dmesg.txt](https://github.com/morrownr/8821au-20210708/files/9824905/dmesg.txt)

morrownr, 2022-10-21:
I don't have any ideas yet. I'm working on the 8814au driver today but I should be working on this driver at some point next week and I'll keep this in mind. You never know when you will run onto something.

morrownr, 2022-12-21:
Any progress?

closed 2024-04-02 by morrownr