morrownr/8821au-20210708 #57: EW-7611UCB: WiFi doesn't seem to work . Kernel 5.19

dm30, 2022-10-31:
Hi! I did installation as instructed (except enabling Concurrent Mode; Edimax doc. did not mention it).
No errors. Seems to have worked. DKMS status returned:

rtl8821au/5.12.5.2, 5.19.0-23-generic, x86_64: installed

Bluetooth worked out of the box (no driver needed) but WiFi is absent. I moved the dongle to another port in the back but was not successful either.
My system uses Ubuntu 22.10, so kernel 5.19. Is connected through cable too. 
Might have to do with Linux 5.19? Or related to that concurrent mode? 
I really appreciate your help.

morrownr, 2022-11-01:
Hi @dm30 

> rtl8821au/5.12.5.2, 5.19.0-23-generic, x86_64: installed

The driver is installed.

> Bluetooth worked out of the box (no driver needed) but WiFi is absent.

This happens. Been a long day and I don't remember the fix right now. It happens more often with the 8811cu chipsets so going to the following repo and reading the issues will probably help.

https://github.com/morrownr/8821cu-20210118

> Might have to do with Linux 5.19?

No.

> Or related to that concurrent mode?

No.

Let me know what you find out reading in the other repo.

dm30, 2022-11-02:
Thx. You are very kind. 
I checked the issues but could not locate the same issue or solution. Perhaps I missed it.

morrownr, 2022-11-03:
Hmmm...

Post the results of:

$ iw reg get

$ lsusb

We will figure it out.

Nick

dm30, 2022-11-03:
Thank you Nick
iw reg get:
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
lsusb:
Bus 006 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
Bus 005 Device 002: ID 7392:b611 Edimax Technology Co., Ltd Edimax Wi-Fi AC600 Bluetooth4.0 USB Adapter
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1b1c:1c0a Corsair                       
Bus 001 Device 003: ID 0408:7020 Quanta Computer, Inc. RGB-IR Camera
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2022-11-03:
> country 00: DFS-UNSET

This pretty much shuts down 5 GHz.

Run the following with your country code in place of US:

$ sudo iw reg set US

To see if it worked:

$ iw reg get

>  ID 7392:b611

This confirms that this is the correct driver.

Is secure mode on?

What distro, version and kernel are you running?

dm30, 2022-11-04:
This is what I got.
$ sudo iw reg set US
$ iw reg set
Usage:	iw [options] reg set <ISO/IEC 3166-1 alpha2>

Notify the kernel about the current regulatory domain.

Options:
	--debug		enable netlink debugging
Secure mode is secure boot? It is enabled. I dual boot with Windows. Needed for face logging.
The Linux part is Ubuntu 22.10.  Kernel 5.19.0-23-generic

morrownr, 2022-11-04:
> This is what I got.
$ sudo iw reg set US
$ iw reg set

I meant for the second line to be:

$ iw reg get

> Secure mode is secure boot?

Yes

> It is enabled

Okay, so the machine is in secure boot mode. Now let's check to see what Ubuntu thinks:

$ sudo mokutil --sb-state

If Ubuntu is in secure boot mode then hopefully dkms handled the keys for this driver.

You can always turn secure boot mode off temporarily in the bios to see if this is the issue, if it is the issue that this driver will be allowed to spring to life.

What next? We'll worry about that when we get to it.

dm30, 2022-11-04:
Excuse me Nick. I now read more carefully.
$ sudo iw reg set ES // I understand is this for Spain
$ iw reg get
global
country ES: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
sudo mokutil --sb-state
SecureBoot enabled
Are those values correct?
I'll disable secure boot to ckeck if works

morrownr, 2022-11-04:
> country ES: DFS-ETSI

Muy bien.

> SecureBoot enabled

Did the installer tell you it was using the "dkms routines" during the installation? The reason I ask is because the installer is capable of installing the driver with or without dkms... but if dkms is not installed, it has no capability to handle secure boot mode.

If dkms is installed and secure boot mode is enabled, as it is, when you rebooted you should have been given a chance to review and approve the keys? Did that happen?

dm30, 2022-11-04:
I don't remember precisely but I'd swear I installed dkms beforehand so I could check there were no any other drivers installed. I don't remember having to approve anything. I only remember being prompt to open the config. But I change nothing.

morrownr, 2022-11-04:
A quick test with secure boot mode off in bios will tell us if secure boot mode is an issue.

ShunnShine, 2022-11-04:
Sorry to jump in to the middle of this conversation. I installed the driver today and was met with a very similar issue. I disabled secure boot and found that the driver started working.

I'm familiar with that screen on reboot that asks you to review and approve new keys, but it never appeared after I installed the driver.
Here's a copy of the printout from my install:
```Running install-driver.sh version 20221018
6.0.5-200.fc36.x86_64
x86_64
Installing 8821au.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
Sign command: /lib/modules/6.0.5-200.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...
'make' -j12 KVER=6.0.5-200.fc36.x86_64 KSRC=/lib/modules/6.0.5-200.fc36.x86_64/build.......
Signing module /var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko
At main.c:167:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: /var/lib/dkms/mok.key
Cleaning build area...

8821au.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation'
   - Installing to /lib/modules/6.0.5-200.fc36.x86_64/extra/
depmod.....
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] y
Do you want to reboot now? (recommended) [y/N] 
```
I'm not sure what the SSL errors are but they might be involved. I'm interested to hear if secure boot is the problem for @morrownr as well, and I'll be following along to see if we can come up with a solution to turn secure boot back on. Thanks for your help.

morrownr, 2022-11-05:
Hi @ShunnShine 

You might as well join the party.

> Sign command: /lib/modules/6.0.5-200.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

> Signing module /var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko
At main.c:167:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: /var/lib/dkms/mok.key

All of the above have to do with secure boot mode.

> Certificate or key are missing,

That statement is telling us one of both files that dkms is looking for are not where it is looking. That is the main problem. 

Can one or both of you look to see which of the following files is missing?

> Signing key: /var/lib/dkms/mok.key

> Public certificate (MOK): /var/lib/dkms/mok.pub

I do have 2 systems that have secure boot turned on. One has a fresh Xubuntu 22.10 install on it. I will look at doing a test on it later this weekend or early next week. Maybe we can get to the bottom of this. If all else fails, I do have instructions for manual signing here somewhere. I'm actually doing a major upgrade to the install and remove scripts right now in the background and this is one of the top 3 issues I want to address but I haven't started on this issue yet.

Nick

ShunnShine, 2022-11-05:
I’d be happy to help check files. However, I made that comment at the end of my day. I’ll be back at my workstation either tomorrow or the next.

dm30, 2022-11-05:
Hi! Secure mode off and it worked. Seemingly... 
No I see 3  WiFi nets although only 'USB' one lists networks.
Unfortunately I couldn't make it work reliably. Only finds the 5GHz network (there's a twin 2.4GHz) although reception is poor. I tried with phone both at 2.4GHz and 5GHz as well. No matter how, keeps asking for password and eventually connects, but just for few seconds. I don't think receives a single bit from the web. 
Oh, and bluetooth stopped working.  
I'm connected via ethernet with WiFi on in case you want me to check something. 
I really appreciate your support.

[image]

[image]

morrownr, 2022-11-06:
> Secure mode off and it worked.

Okay, so in both cases here, we have issue with secure boot.

@dm30 , let's do away with concurrent mode as a factor:

$ cmode-off.sh
$ sudo ./remove-driver.sh

@dm30 @ShunnShine 

Let's try something. Normally, if dkms is installed, on Ubuntu based systems, secure boot support just works. It seems that is not the case here and I do not know why. What I would like for each of you to do is a manual installation following my instructions below. Maybe this will help us getting started on figuring out what the problem is. Turn secure boot back on in your bios before you start. Post back what you see:

Installation with module signing for Secure Boot

If the driver is not downloaded:

$ mkdir ~/scr

$ cd ~/src

$ git clone git://github.com//morrownr/8821au-20210708.git

$ cd 8821au-20210708

If the driver is already downloaded:

From the driver directory:

$ sudo ./remove-driver.sh

$ cmode-off.sh

$ git pull

$ make

$ sudo make sign

You will be promted for a password, remember it because you will need it.

$ sudo make install

Reboot to activate the newly installed module.

In the MOK managerment screen:

Select "Enroll key" and enroll the key created by above sign step.

When promted, enter the password you entered when created the sign key.

dm30, 2022-11-06:
cmode-off.sh returned 'order not found'. Shoud I continue uninstalling the driver?

morrownr, 2022-11-06:
Just use a text editor to make sure the following line in Makefile is set as shown:

$ nano Makefile

CONFIG_CONCURRENT_MODE = n

And then press on.

dm30, 2022-11-06:
Please excuse that I can't figure out how to type ^O to save it. 

[image]

morrownr, 2022-11-06:
I usually just use `ctrl X` and then `Y` and Enter...

However, the Makefile should be long and all I see is one line. You are editing the `Makefile` in the driver directory? Right.

dm30, 2022-11-07:
Thank you for your patience Nick. I see is not set.

[image]

dm30, 2022-11-07:
This happened. Should I continue?

[image]

dm30, 2022-11-07:
I did all the steps and rebooted Nick. dkms status returns nothing

ShunnShine, 2022-11-07:
Sorry for the delayed response. I made it halfway through the manual key installation and ran into some problems. Although, I think it highlights what the next steps will be.

The last three lines of 
> $ make
```
...
BTF [M] /home/shunn/USB_WIFI_Driver/8821au-20210708/8821au.ko
Skipping BTF generation for /home/shunn/USB_WIFI_Driver/8821au-20210708/8821au.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.0.5-200.fc36.x86_64'
```
Then 

> $ sudo make sign
```
/bin/sh: line 1: openssl: command not found
make: *** [Makefile:2520: sign] Error 127
```

I'm unfamiliar with the BTF generation step, but it's obvious that I need openssl. Do you agree that the next step is to install openssl and retry the driver install?

morrownr, 2022-11-07:
Well, that is not good. I merged a patch into Makefile yesterday that is needed for this to work. You did run `$ git pull`?

morrownr, 2022-11-07:
Oh my... openssl is not installed. No wonder. I guess I need to add a check for that.

Yes, please install openssl and...

```

$ sudo ./remove-driver.sh

$ git pull

$ make

$ sudo make sign

You will be promted for a password, remember it because you will need it.

$ sudo make install

Reboot to activate the newly installed module.

In the MOK managerment screen:

Select "Enroll key" and enroll the key created by above sign step.

When prompted, enter the password you entered when created the sign key.

```

ShunnShine, 2022-11-07:
Okay now I'm at the same place as @dm30:

> $ sudo make sign
```
input password: 
input password again: 
At main.c:298:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: 8821au
make: *** [Makefile:2522: sign] Error 1
```

I made sure to git pull. Is this a failed state or can I continue?

morrownr, 2022-11-07:
It is time to stop and try to figure out what that means.

morrownr, 2022-11-07:
Okay, so you had to install openssl. Did you enroll keys when you installed the distro?

ShunnShine, 2022-11-07:
I'm not sure I'm familiar with what enrolling keys when I installed the distro would entail.
I'm working on a fresh install of Fedora. I installed it with Secure Boot on, but it never asked me to enroll keys. Does that answer?

ShunnShine, 2022-11-07:
Do you think it's a problem with Fedora? I installed this driver on a Linux Mint distro 5 months ago with no problems.

dm30, 2022-11-07:
Dear Nick. I did all the steps above with the same result. OpenSSL was installed indeed (although I read something about being now 'firm')
In case it gives some hint, I was informed that I was going to remove package not included in dkms tree.
Also that the compiler was  not the same employed for the kernel.
"The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  You are using:           gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0"
Also there's a Secure Boot Mode parameter in BIOS with 2 options: Standard -default- for keys being loaded from BIOS and 'Custom' so user can 'configure setting and manually load the secure keys'.
Of course, I don't really know what is that all about. Just in case 8-)
Thank you so much,

morrownr, 2022-11-08:
@dm30 

> Also that the compiler was not the same employed for the kernel.
> "The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-3ubuntu1) 12.2.0
> You are using: gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0"

You can ignore that. I see it all the time. It is just to let you know there could be a problem, not that there is one.

> Also there's a Secure Boot Mode parameter in BIOS with 2 options: Standard -default- for keys being loaded from BIOS and 'Custom' so user can 'configure setting and manually load the secure keys'.

You should probably go with `Standard -default-` unless they have been erased, on which case, you have no keys.

> Of course, I don't really know what is that all about.

Well, it is not like I am a genius or we would have this problem sorted out. Maybe it is time to take what we have seen so far and do some reading. I'm going to be busy for a couple of days anyway. Check in when you have some things figured out... maybe I'll be able you figure something out as well.

morrownr, 2022-11-08:
@ShunnShine 

> I'm not sure I'm familiar with what enrolling keys when I installed the distro would entail.

You may need to check with the Fedora forums to see what you should see during an installation with secure boot on.

On Ubuntu based systems, the installer will detect if it is installing on a system with secure boot on and it will ask you for a secure boot password. After the installation is complete, during reboot, a utulity will come up and ask you if you want to enroll keys. You then either enroll the systems keys or your customs keys and reboot again. All done.

Like I said, I tested a couple of these Realtek drivers on my fresh installation of Xubuntu 22.10 yesterday and both just installed and signed the driver as slick as can be.

Evidently Fedora does something different and we need to know what that is.

ShunnShine, 2022-11-08:
I spent some time reading, and I think the details of secure boot are a few steps above my current level. The latest from ubuntu is looking pretty nice, so I'm thinking I'll migrate over to ubuntu 20.10 and see if that fixes it.

It sounds like you're busy for a few days, but I'll keep the fedora install around in case you were interested in nailing down the source of the issue. Just let me know if you are interested in poking around my setup.

Also, let me know if there's anywhere a novice can contribute to your projects. I'd love to give back where I can.

morrownr, 2022-11-09:
Good day @ShunnShine 

> I think the details of secure boot are a few steps above my current level.

Mine too! We just cannot be experts on everything. Running this site takes a lot of time. What I mean by this site is:

https://github.com/morrownr/USB-WiFi

Many folks never see the Main Menu because search engines take them directly to a repo. You mention contributing. That would be great. Take a look at the Main Menu and see if you find anything you would be interested in working on. If you find something, then I will get my email address to you so we can talk.

For now, I have some priorities so feel free to write over your Fedora installation. We can always come back to it when there is time. If you are interested in checking out Ubuntu, let me throw a little info your way. Ubuntu is the core distro of a family. Ubuntu uses Gnome for the gui desktop. As much as I try to like Gnome, I do not. It depends on your computer and your tastes. I personally prefer Xubuntu as it has the same backend as Ubuntu but the gui desktop is very simply yet configurable and it is ultra stable. 

Nick

sunwire, 2022-11-09:
On Fedora (if you use DKMS) with Secure Boot enabled you have to import the DKMS sign key
`mokutil --import /var/lib/dkms/mok.pub` and reboot to enroll the key

morrownr, 2022-11-10:
@sunwire 

Are you saying to run the above after installation is complete?

Should I add this info to the README?

sunwire, 2022-11-10:
> Are you saying to run the above after installation is complete?

Yes

> Should I add this info to the README?

If you wish

One important thing for Fedora users. The mok.pub  and mok.key keys are created (and module si signed )by dkms, but only if openssl package is installed.

morrownr, 2022-11-14:
@sunwire @dm30 @ShunnShine 

I've been sick for a few days. I'm feeling somewhat better now and would like to work this issue to a conclusion if possible.

@sunwire  Let me repeat what I think you have said. Hopefully it is correct:

```
Fedora users that have Secure Boot turned on should do the following before running the installation script:

Install the `openssl' package.

Import the DKMS sign key:

$ mokutil --import /var/lib/dkms/mok.pub

Reboot to enroll the key.

Follow the driver installation steps.
```

@dm30 @ShunnShine 

If I could get both of you to run the following, I would appreciate it:

```
$ sudo ./remove-driver.sh

Install the `openssl' package.

Import the DKMS sign key:

$ mokutil --import /var/lib/dkms/mok.pub

Reboot to enroll the key.

Follow the driver installation steps.
```

Let me know if this works. Hopefully it does and I can document this for other Fedora users.

Nick

sunwire, 2022-11-14:
> @sunwire Let me repeat what I think you have said. Hopefully it is correct:

Generally yes, but driver build/installation have to be **before** import DKMS key (DKMS will create a pair of keys if they do not exist)

1. install openssl (and dkms if not installed)
2. build/install a driver
3. import the key
4. reboot
5. enroll the key

Maybe this short [film](https://youtu.be/rlCSVyv54fc) will help

morrownr, 2022-11-16:
> Generally yes, but driver build/installation have to be **before** import DKMS key (DKMS will create a pair of keys if they do not exist)
>
> 1. install openssl (and dkms if not installed)
> 2. [...]

@sunwire 

I've been sick the last few and am getting around slowly. I'm trying to work again some this evening.

I've been pondering what you are saying and I am grateful for your help but if you will hang in there with me, I need to sort some things out.

My preference is to have my installation script (install-driver.sh) handle as much as possible. My logic sequence would go something like this:

Check if `mokutil` is installed. If not, error out and tell user to install it.
Check to see if Secure Boot is active. If so, check to see if openssl is installed and if not, error out and tell user to install it.

Here is where I have problems: How do I get the appropriate files in place so that dkms can just run right through the job like it does with Ubuntu?

You are welcome to look through install-driver.sh script if you want. Sure could use some ideas as I really don't want to tell Fedora users that you can't use the script, you have to go do a bunch of command line shit. We surely can do better than that.

Nick

sunwire, 2022-11-16:
@morrownr 
> You are welcome to look through install-driver.sh script if you want.

I've tested  `install-driver.sh` script. It works on Fedora 36 without a problem.
The only additional steps I took were install dkms and openssl before running `install-driver.sh` script, and run `mokutil --import /var/lib/dkms/mok.pub` command after `install-driver.sh` and then reboot.

sunwire, 2022-11-16:
> Should I add `openssl` here in Step 3

In my opinion, yes you should.

I would skip kernel-debug-devel package in  step 3

> Does running `mokutil --import /var/lib/dkms/mok.pub` before running install-driver.sh also work and allow dkms to initiate the signing instead of showing an error?

No, dkms creates /var/lib/dkms/mok.pub key during install (of course if openssl) so there is on key before running `install-driver.sh` script.

morrownr, 2022-11-16:
> > Should I add `openssl` here in Step 3
> 
> In my opinion, yes you should.
> 
> I would skip kernel-debug-devel package in step 3

Okay. Step 3 for Fedora has been changed to the following and is ready to merge:

sudo dnf -y install git dkms kernel-devel openssl

> No, dkms creates /var/lib/dkms/mok.pub key during install (of course if openssl) so there is on key before running `install-driver.sh` script.

Let me throw out how this does down on Ubuntu based distros:

Install an Ubuntu based distro (my favorite is (Xubuntu) on a system with Secure Boot on.
When the system reboots, mokutil immediately pops up allowing you to register the keys.
Reboot again.
When installing one of the Realtek drivers here, you do not have to do anything regarding Secure Boot. Nothing. Nada.

How do we get Fedora to the state, like Ubuntu, where install-driver.sh handles Secure Boot without having to do anything?

There has to be a way.

sunwire, 2022-11-16:
> How do we get Fedora to the state, like Ubuntu, where install-driver.sh handles Secure Boot without having to do anything?

I think currently it is not possible. It looks like Ubuntu creates and imports mok key during system installation but Fedora does not do that.

morrownr, 2022-11-16:
>  It looks like Ubuntu creates and imports mok key during system installation but Fedora does not do that.

That appears to be correct. I am going to back off for a time to study Secure Boot to see if I can come to a better understanding. This is not just an issue with Fedora but with other non-Ubuntu distros as well.

I am going to go ahead and merge the changed Step 3 for Fedora:

sudo dnf -y install git dkms kernel-devel openssl

And see where we go.

If I am understanding this so far. if @dm30 and @ShunnShine were to remove the driver and then reinstall it via the installation instructions, including the new Step 3 for Fedora and then run the following after the installation script is finished, it should work?

mokutil --import /var/lib/dkms/mok.pub

Lastly, I may have time to install Fedora on a test system so if you could give me a link to an amd64 image, I would appreciate it.

Nick

sunwire, 2022-11-17:
> If I am understanding this so far. if @dm30 and @ShunnShine were to remove the driver and then reinstall it via the installation instructions, including the new Step 3 for Fedora and then run the [...]

Correct

> I may have time to install Fedora on a test system so if you could give me a link to an amd64 image, I would appreciate it.

https://getfedora.org/ Workstation is Gnome based, and I think the most popular.

morrownr, 2022-11-17:
@dm30 @ShunnShine 

Are either of you able to test the solution I mentioned 2 messages above?

dm30, 2022-11-17:
Sorry for the delay. Its been very busy week. 
At this point its just running this commands right?

sudo ./remove-driver.sh // reboot
sudo ./install-driver.sh // Dismissing any prompt on rebooting before next step its done 
mokutil --import /var/lib/dkms/mok.pub // reboot and enroll keys

I understand other steps are already done in previous attempts.
I'm just regular guy after a long long work day.
I really appreciate your support.

morrownr, 2022-11-18:
> Sorry for the delay. Its been very busy week. At this point its just running this commands right?

Sort of. Installation Step 3 has changed. It now installs openssl.
 
Install Step 3.
> sudo ./remove-driver.sh // reboot
> sudo ./install-driver.sh // Dismissing any prompt on rebooting before next step its done
> mokutil --import /var/lib/dkms/mok.pub // reboot and enroll keys
 
> I understand other steps are already done in previous attempts.

Sorry that this is messy. You probably have a better appreciation for why I strongly recommend that users buy adapters that use in-kernel drivers at this point.

> I'm just regular guy after a long long work day. I really appreciate your support.

I appreciate your help to test. If this works, that tells me a few things I need to know and I am studying the situation to figure out how to do this without needing to know Calculus II.

Nick

ShunnShine, 2022-11-18:
Success!

On my Fedora machine, I ran the install script with no errors and used mokutil to enroll the key. Although mokutil didn't work without root privileges, so I had to run 
`sudo mokutil --import /var/lib/dkms/mok.pub`

Also on a side note, earlier I upgraded all my packages including my kernel, so I'm running 6.0.8-200.fc36.x86_64. Hopefully, that is unrelated.

Thanks, @morrownr and @sunwire

morrownr, 2022-11-18:
Whew. That is great.

Yes...

$ sudo mokutil --import /var/lib/dkms/mok.pub

Question: Does mokutil come in Fedora by default or is it installed as a package? If package, do you remember the name of the package?

sunwire, 2022-11-18:
> Question: Does mokutil come in Fedora by default or is it installed as a package? If package, do you remember the name of the package?

Mokutil is installed by default in Fedora, and the package name is (what a surprise!) mokutil

nxtwrld, 2022-11-20:
Sorry to barge in into the discussion. I think I have a similar problem. Everything seems to be installed, I just do not get any wifi interface.

dkms status
`rtl8821au/5.12.5.2, 5.19.0-23-generic, x86_64: installed`

lsusb
`Bus 001 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]`

iw reg get
`global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)`

mokutil --sb-state
`SecureBoot disabled`

iwconfig
`lo        no wireless extensions.

eno1      no wireless extensions.`

I have been struggling with it all day, but no luck.

morrownr, 2022-11-20:
> but it is gone after reboot

I just checked this on an Xubuntu 22.10 installation and the country code is indeed gone after reboot. Not sure what is going onl.

Run:

$ ./edit-options.sh

...and post what is on the `options` line?

nxtwrld, 2022-11-20:
i got 

`options 8821au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_dfs_region_domain=0`

morrownr, 2022-11-20:
Try:

options 8821au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_dfs_region_domain=3 rtw_country_code=CZ

nxtwrld, 2022-11-21:
Done, rebooted, but no change.

iw reg get 
`global
country 00: DFS-UNSET`

an

iwconfig
`lo        no wireless extensions.

eno1      no wireless extensions.`

dm30, 2022-11-21:
Hi!
I tried 
sudo apt install -y dkms kernel-devel openssl git
Couldn't find the kernel-devel package.
I carried on, as OpenSSL is supposed to be installed in Ubuntu, but sudo / mokutil --import /var/lib/dkms/mok.pub always returns:
Failed to get file status, /var/lib/dkms/mok.pub
I tried normal step 3 as well, but those packages were already in place.

[1 more reply]