morrownr/8821au-20210708 #58: A lot of 5 GHz Wi-Fi channels are disabled

Danct12, 2022-11-04:
I'm using a TP-Link Archer T2U Nano and I cannot connect to certain Wi-Fi AP. My phone does connect just fine, and when I check the frequency, it was on channel 157 which is disabled in the driver according to iw phy0.

I have tried changing iw reg, but this did nothing to improve the channel situation.

## The following Wi-Fi channels are disabled, according to `iw phy0 info`
```
			* 5075 MHz [15] (disabled)
			* 5080 MHz [16] (disabled)
			* 5085 MHz [17] (disabled)
			* 5090 MHz [18] (disabled)
			* 5100 MHz [20] (disabled)
			* 5120 MHz [24] (disabled)
			* 5140 MHz [28] (disabled)
			* 5160 MHz [32] (disabled)
			* 5180 MHz [36] (13.0 dBm)
			* 5200 MHz [40] (13.0 dBm)
			* 5220 MHz [44] (13.0 dBm)
			* 5240 MHz [48] (13.0 dBm)
			* 5260 MHz [52] (13.0 dBm) (radar detection)
			* 5280 MHz [56] (13.0 dBm) (radar detection)
			* 5300 MHz [60] (13.0 dBm) (radar detection)
			* 5320 MHz [64] (13.0 dBm) (radar detection)
			* 5340 MHz [68] (disabled)
			* 5360 MHz [72] (disabled)
			* 5380 MHz [76] (disabled)
			* 5400 MHz [80] (disabled)
			* 5420 MHz [84] (disabled)
			* 5440 MHz [88] (disabled)
			* 5460 MHz [92] (disabled)
			* 5480 MHz [96] (disabled)
			* 5500 MHz [100] (13.0 dBm) (radar detection)
			* 5520 MHz [104] (13.0 dBm) (radar detection)
			* 5540 MHz [108] (13.0 dBm) (radar detection)
			* 5560 MHA
		ssid XXXXXXXXXXXXz [112] (13.0 dBm) (radar detection)
			* 5580 MHz [116] (13.0 dBm) (radar detection)
			* 5600 MHz [120] (13.0 dBm) (radar detection)
			* 5620 MHz [124] (13.0 dBm) (radar detection)
			* 5640 MHz [128] (13.0 dBm) (radar detection)
			* 5660 MHz [132] (13.0 dBm) (radar detection)
			* 5680 MHz [136] (13.0 dBm) (radar detection)
			* 5700 MHz [140] (13.0 dBm) (radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (disabled)
			* 5765 MHz [153] (disabled)
			* 5785 MHz [157] (disabled)
			* 5805 MHz [161] (disabled)
			* 5825 MHz [165] (disabled)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
```

## Output requested by maintainer
```
6.0.2-1-MANJARO-ARM-ODROID aarch64
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Bus 001 Device 006: ID 046d:c092 Logitech, Inc. G102/G203 LIGHTSYNC Gaming Mouse
Bus 001 Device 005: ID 258a:002a SINO WEALTH Keyboard
Bus 001 Device 004: ID 0a05:7211 Unknown Manufacturer hub
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no
8821au/5.12.5.2.r155.gb46c98d, 6.0.2-1-MANJARO-ARM-ODROID, aarch64: installed
phy#0
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr 12:34:56:78:90:AA
		ssid XXXXXXXXXXXXXXX
		type managed
		channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
		txpower 16.00 dBm
global
country NZ: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 30), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
```

morrownr, 2022-11-04:
Good day @Danct12 

> I cannot connect to certain Wi-Fi AP. My phone does connect just fine, and when I check the frequency, it was on channel 157 which is disabled in the driver according to iw phy0.

WiFi is up there with rocket science as far as complexity is concerned. My short answer is that I have no control over this issue. For better understanding, I'll throw the long answer out. Laws and regulations regarding WiFi vary all over the world. Linux maintains a regulatory database which is located as follows:

https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb

This database is available to all distros and is updated as new information comes in. You can download the file and check if the info for NZ is correct. If it is not correct, then you can send changes to Seth, the maintainer.

If the information is correct in accordance with the information you posted above, then the router that is broadcasting on channel 157 is violating your laws. 

A reasonable solution in your case, if the router supports DFS channels and you do not have any devices that do not support DFS channels, is to move the router to a channel in 52-64 or a channel in 100-140. Those channels are legal in your country. Also legal in your country would be a non-dfs channel in 36-48.

Keep in mind that DFS required the router/AP to wait 60 seconds after boot so as to check for conflicting use of the channel before it can braodcast so if you most to a DFS channel, when you boot your router, you have to wait a while before you can see the signal.

Hope this helps. Marking the issue as info.

Nick

Danct12, 2023-01-19:
I think this is more of a driver issue than a regulatory issue.

On my CF-WU785AC, this is the list of enabled channels:
```
			* 5180 MHz [36] (17.0 dBm)
			* 5200 MHz [40] (17.0 dBm)
			* 5220 MHz [44] (17.0 dBm)
			* 5240 MHz [48] (17.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (radar detection)
			* 5280 MHz [56] (20.0 dBm) (radar detection)
			* 5300 MHz [60] (20.0 dBm) (radar detection)
			* 5320 MHz [64] (20.0 dBm) (radar detection)
			* 5500 MHz [100] (20.0 dBm) (radar detection)
			* 5520 MHz [104] (20.0 dBm) (radar detection)
			* 5540 MHz [108] (20.0 dBm) (radar detection)
			* 5560 MHz [112] (20.0 dBm) (radar detection)
			* 5580 MHz [116] (20.0 dBm) (radar detection)
			* 5600 MHz [120] (20.0 dBm) (radar detection)
			* 5620 MHz [124] (20.0 dBm) (radar detection)
			* 5640 MHz [128] (20.0 dBm) (radar detection)
			* 5660 MHz [132] (20.0 dBm) (radar detection)
			* 5680 MHz [136] (20.0 dBm) (radar detection)
			* 5700 MHz [140] (20.0 dBm) (radar detection)
			* 5720 MHz [144] (20.0 dBm) (radar detection)
			* 5745 MHz [149] (20.0 dBm)
			* 5765 MHz [153] (20.0 dBm)
			* 5785 MHz [157] (20.0 dBm)
			* 5805 MHz [161] (20.0 dBm)
			* 5825 MHz [165] (20.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
```

When compiling this driver with stock configuration, this is the list of enabled channels:
```
			* 5180 MHz [36] (13.0 dBm)
			* 5200 MHz [40] (13.0 dBm)
			* 5220 MHz [44] (13.0 dBm)
			* 5240 MHz [48] (13.0 dBm)
			* 5260 MHz [52] (13.0 dBm) (radar detection)
			* 5280 MHz [56] (13.0 dBm) (radar detection)
			* 5300 MHz [60] (13.0 dBm) (radar detection)
			* 5320 MHz [64] (13.0 dBm) (radar detection)
			* 5500 MHz [100] (13.0 dBm) (radar detection)
			* 5520 MHz [104] (13.0 dBm) (radar detection)
			* 5540 MHz [108] (13.0 dBm) (radar detection)
			* 5560 MHz [112] (13.0 dBm) (radar detection)
			* 5580 MHz [116] (13.0 dBm) (radar detection)
			* 5600 MHz [120] (13.0 dBm) (radar detection)
			* 5620 MHz [124] (13.0 dBm) (radar detection)
			* 5640 MHz [128] (13.0 dBm) (radar detection)
			* 5660 MHz [132] (13.0 dBm) (radar detection)
			* 5680 MHz [136] (13.0 dBm) (radar detection)
			* 5700 MHz [140] (13.0 dBm) (radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (disabled)
			* 5765 MHz [153] (disabled)
			* 5785 MHz [157] (disabled)
			* 5805 MHz [161] (disabled)
			* 5825 MHz [165] (disabled)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
```

I found out that there's a unlisted setting in the source code that will enable all these channel to match the regulatory domain (-DCONFIG_REGD_SRC_FROM_OS) and upon enabling it all channels from 36->165 are enabled and can be used.

robinwesterik, 2023-03-15:
I believe Danct12 is right. Wireless-regdb allows for 5735MHz - 5835MHz in Vietnam, yet these frequencies appear to be disabled in this driver's stock configuration

morrownr, 2023-03-15:
> I believe Danct12 is right. 

I believe he is right also. I have experimented with that setting. There are some downsides to the setting. My testing in January exposed that this setting kills AP mode, at least on some distros, and AP mode is very heavily used. There is also an issue of various distros handling things differently.

Changing the current default would likely cause the issues section to fill up. Keeping the number of problems to a minimum is something that I am always working on because I can't spend all of my time answering questions and fixing problems or I won't have time to actually do work to make things better.

This may be an issue where I need to document the situation and let users make the change if they want to do so.

If you have time, please investigate and provide more info as to why we are seeing the difference in this setting in this situation.

For what it is worth: The regdb and the secure boot issues are at the top of my list to make better. I'm slowly making progress with secure boot but both of these issues are complicated by the lack of standardization as to how distros handle the situations so it is a challenge. I welcome anyone with the time and desire to work on these issues so that we can make things better.

Thanks,

@morrownr

burner1024, 2023-08-06:
So what's the actual issue? The driver hardcodes an old snapshot of regdb?

morrownr, 2023-08-10:
> So what's the actual issue?

That is not clear.

The below earlier statement from me outlined my agenda at the time of writing. There are certainly other issues such as handling situations where users have previously install older versions of a driver. I think I have really good solution to handling secure boot and deleting earlier versions of an installed driver. Those solution should work their way into all of the repos here sometime in the next couple of months. The regdb issue? I got nothing because I have not had time to work on it. Some issues take an incredibly long time to work and test. If anyone figures out this problem, please post.

> The regdb and the secure boot issues are at the top of my list to make better.

@morrownr

burner1024, 2023-08-11:
Sure, I understand the constraints, just want to clarify. If the driver doesn't get the info from OS, then where does it get it from, currently?

dubhater, 2024-04-22:
I have this problem too. With ``-DCONFIG_REGD_SRC_FROM_OS`` channels 144-173 no longer show disabled in iw list, but my phone's hotspot on channel 149 still doesn't appear in the scans. rtw88 can see it...

morrownr, 2024-04-22:
@dubhater 

I'll take a look at this when able. Health issues keeping me out of action.

> rtw88 can see it...

I've been holding off on starting an issue for this repo as users will be looking for something to test and I don't think we are there yet, right?

dubhater, 2024-04-22:
I hope you will get better soon.

I may end up fixing this so I can compare the two drivers in the 5 GHz band.

It is still early. I was just thinking that rtw88 is like pre-alpha right now.

morrownr, 2024-04-23:
> channels 144-173 no longer show disabled in iw list...

I have not been able to duplicate this problem. I dusted off my rtl8811au based adapter, compiled the driver as it is in this repo right now and:

```
			* 5180 MHz [36] (13.0 dBm)
			* 5200 MHz [40] (13.0 dBm)
			* 5220 MHz [44] (13.0 dBm)
			* 5240 MHz [48] (13.0 dBm)
			* 5260 MHz [52] (13.0 dBm) (radar detection)
			* 5280 MHz [56] (13.0 dBm) (radar detection)
			* 5300 MHz [60] (13.0 dBm) (radar detection)
			* 5320 MHz [64] (13.0 dBm) (radar detection)
			* 5500 MHz [100] (13.0 dBm) (radar detection)
			* 5520 MHz [104] (13.0 dBm) (radar detection)
			* 5540 MHz [108] (13.0 dBm) (radar detection)
			* 5560 MHz [112] (13.0 dBm) (radar detection)
			* 5580 MHz [116] (13.0 dBm) (radar detection)
			* 5600 MHz [120] (13.0 dBm) (radar detection)
			* 5620 MHz [124] (13.0 dBm) (radar detection)
			* 5640 MHz [128] (13.0 dBm) (radar detection)
			* 5660 MHz [132] (13.0 dBm) (radar detection)
			* 5680 MHz [136] (13.0 dBm) (radar detection)
			* 5700 MHz [140] (13.0 dBm) (radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (13.0 dBm)
			* 5765 MHz [153] (13.0 dBm)
			* 5785 MHz [157] (13.0 dBm)
			* 5805 MHz [161] (13.0 dBm)
			* 5825 MHz [165] (13.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)

```

This is exactly what I would have expected to see with the exception of channel 144 which should not be disabled. Country is US.

dubhater, 2024-04-23:
Your device probably has a different hardcoded channel plan (offset 0xb8 in the efuse). Mine has the value 0xba. Bit 7 instructs the driver to use this value instead of the user's choice. And indeed, 0xba - bit 7 = 0x3a, which corresponds to a channel plan where channels over 140 are disabled.

To ignore the hardcoded channel plan, use ``-DCONFIG_FORCE_SW_CHANNEL_PLAN``. Maybe together with ``-DCONFIG_REGD_SRC_FROM_OS``.

Now I can use channel 149 with this driver.

Interestingly, rtw88 has no concept of "channel plan". mac80211 tells it what channels to use.

morrownr, 2024-04-23:
FYI: I recompiled again adding only the following to Makefile:

EXTRA_CFLAGS += -DCONFIG_REGD_SRC_FROM_OS

...and I am seeing my 5 GHz list of channels exactly as they should be. So it seems that only the above flag is needed to cause the REGD info to be pulled from the OS. It sure looks to me like this should be a module parameter for those wanting to compile on a very old kernel but what do I know but I can document it.

Thoughts?

dubhater, 2024-04-23:
They had the same thought: the module parameter rtw_regd_src appears when you use ``-DCONFIG_REGD_SRC_FROM_OS``.

> Thoughts?

More testing would be good. I'm only using this driver briefly, to check something or other and compare with rtw88.

morrownr, 2024-04-24:
> Even though they weren't disabled anymore in iw list. 

Hmmm... stuff to test.

> I'm only using this driver briefly, to check something or other and compare with rtw88.

I understand why.

> They had the same thought: the module parameter rtw_regd_src appears when you use -DCONFIG_REGD_SRC_FROM_OS.

Now to document what is going on.

Thanks. Let me know if you find anything else that is interesting.

morrownr, 2024-04-25:
@dubhater 

> ... it wasn't scanning the channels greater than 140 when I used just CONFIG_REGD_SRC_FROM_OS.

I see what you mean. I think I am going to take the time to dig into this. What I would like is for the default to come the os with a module parameter to allow whatever. Sometimes Realtek does not name things in a way that correctly clarifies what is being done so digging around in code is where I am headed... slowly. I'm still going slow.

If you happen to run into any hints, let me know.