morrownr/8821au-20210708 #60: Error on Fedora 37 Beta

Caide21, 2022-11-07:
[Yep@fedora 8821au-20210708]$ sudo ./install-driver.sh 
Running install-driver.sh version 20221018
5.19.7-300.fc37.x86_64
x86_64
Installing 8821au.conf to: /etc/modprobe.d
The non-dkms installation routines are in use.
#make -C /lib/modules/5.19.7-300.fc37.x86_64/build M=/home/caide/Documents/Caide/Software/Externel_wifi_driver/8821au-20210708 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.7-300.fc37.x86_64/build M=/home/caide/Documents/Caide/Software/Externel_wifi_driver/8821au-20210708  modules
make[1]: *** /lib/modules/5.19.7-300.fc37.x86_64/build: No such file or directory.  Stop.
make: *** [Makefile:2517: modules] Error 2
An error occurred. Error = 2
Please report this error.
Please copy all screen output and paste it into the report.
You will need to run the following before reattempting installation.

Caide21, 2022-11-07:
Edit: Just needed to install dkms+

closed 2022-11-07 by Caide21

morrownr, 2022-11-08:
At least you pointed out that I need to look at the non-dkms install routines.

sunwire, 2022-11-08:
> make[1]: *** /lib/modules/5.19.7-300.fc37.x86_64/build: No such file or directory. Stop.

You have to install kernel-devel