morrownr/8821au-20210708 #61: Bit more information on the kernel headers path not found with dkms?

Jibun-no-Kage, 2022-11-14:
Trip over this again,  Odd, this time a simple reboot did not fix the issue.  This was on Pi Zero, but not sure that matters.

Since the install-driver.sh failed, I ran the remove-driver.sh, did not reboot and did a make clean then make, just so see what happened, and this was the result:

root@template:~/8821au-20210708-morrownr# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.78+/build M=/root/8821au-20210708-morrownr  modules
make[1]: *** /lib/modules/5.15.78+/build: No such file or directory.  Stop.
make: *** [Makefile:2517: modules] Error 2
root@template:~/8821au-20210708-morrownr# ls -l /lib/modules/5.15.78+/
total 2436
drwxr-xr-x 11 root root   4096 Nov 13 23:19 kernel
-rw-r--r--  1 root root 581281 Nov 13 23:19 modules.alias
-rw-r--r--  1 root root 610734 Nov 13 23:19 modules.alias.bin
-rw-r--r--  1 root root  13930 Nov 13 23:19 modules.builtin
-rw-r--r--  1 root root      0 Nov 13 23:19 modules.builtin.alias.bin
-rw-r--r--  1 root root  15282 Nov 13 23:19 modules.builtin.bin
-rw-r--r--  1 root root  77245 Nov 13 23:19 modules.builtin.modinfo
-rw-r--r--  1 root root 220535 Nov 13 23:19 modules.dep
-rw-r--r--  1 root root 297158 Nov 13 23:19 modules.dep.bin
-rw-r--r--  1 root root    324 Nov 13 23:19 modules.devname
-rw-r--r--  1 root root  66705 Nov 13 23:19 modules.order
-rw-r--r--  1 root root    913 Nov 13 23:19 modules.softdep
-rw-r--r--  1 root root 262188 Nov 13 23:19 modules.symbols
-rw-r--r--  1 root root 318677 Nov 13 23:19 modules.symbols.bin

Pi headers for the given kernel is there...
\# uname -a
Linux template 5.15.78+ #1599 Fri Nov 11 12:20:14 GMT 2022 armv6l GNU/Linux

\# apt install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20221104-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

\# find /usr/lib | grep 15.78+ | grep build
[no output]

Really weird, since the correct headers are there.

I am going to recreate the image, and see if I can recreate it again.

Jibun-no-Kage, 2022-11-14:
Disregard the above, I figured out the latest rpi-update is missing something, build directory, and out of sync with raspberrypi-kernel-headers package deployment again.

closed 2022-11-14 by Jibun-no-Kage

morrownr, 2022-11-14:
> I figured out the latest rpi-update is missing something, build directory, and out of sync with raspberrypi-kernel-headers package deployment again.

In looking over reports in the other Realtek driver repos here, I see other reports like this. This is a problem the RasPi folks need to fix. Have you looked to see if this has been reported?

Jibun-no-Kage, 2022-11-14:
Yup, I just updated the applicable issue, with the Pi foundation folks on github.  They also need to update rpi-source script, it is broken or still broken being based on Python2.  Ouch.  This issue is just, mostly an FYI, for others that might trip over the issue.  I learning to appreciate exactly what you noted in the past, that realtek driver support is a mine field at times, for many diverse reasons.  :)

morrownr, 2022-11-14:
> I learning to appreciate exactly what you noted in the past, that realtek driver support is a mine field at times, for many diverse reasons. :)

This is very true. I started this site ( https://github.com/morrownr/USB-WiFi ) as a way to help with a problem area in Linux. My opinion is that the best way to help is making accurate information available and that is my primary priority. Sure, I work on code and other things but getting good information out there is priority. There are many good adapters for Linux but people were unaware of the many adapters that use in-kernel drivers because the `Best USB Adapters for Linux` lists you see online are filled with bad information. Search the net for `Best USB Adapters for Kali Linux` and you see a bunch of Realtek adapters listed and the much better Mediatek adapters missing in many cases. Do these people actually test before writing? In Main Menu item 2, I, also with others, have compiled an astounding list that I think is better than any other list on the internet as far as good adapters for Linux is concerned.

Jibun-no-Kage, 2022-11-14:
I was reading about Mediatek based dongles, although I have mostly realtek right now.  And I even had a bit of trouble finding information on Mediatek initially, but I know about your site, so figured I would just explore a bit before going to the 'source' and it is insane how much (bad) marketing there is about wifi in general never mind usb dongles.  I have experience with C and driver compiling, so doing it now and then is no big deal for me, but if you are new to it?  It would drive you nuts.  As was asked about 20 years ago, yes 20 years ago, why Linux is not driving deeper into desktop space, and I said then what is still true to day, lack of agreed to standards.  When the average windows user can just plug and play, and it works (not always great but works), Linux just can't compete.  Servers are a different issue, because the expectation is that the hardware is standardized for the software is standardized to higher degree.  Unlike desktop/laptop, etc.  Even the Raspberry Pi devices, explicit standardization, the non-core drivers are  a PITA.

morrownr, 2022-11-14:
>  I have experience with C and driver compiling, so doing it now and then is no big deal for me, but if you are new to it? It would drive you nuts.

Agree. Users that are not programmers would best be served by using adapters that are plug and play. And there are a lot of Linux users out there. Sure Linux has nearly a 100% market share in Super Computers, a 90% market share with Smart Phones and a large market share with servers but the desltop market has been slow to come around. The lack of standardization is a large part of the reason as that creates difficulties in schools as it is hard to even decide what to teach.

Something to be aware of is that things do change. A lot of what you see on the internet now regarding usb wifi adapters has either changed or is not applicable anymore. Linux is in a very different situation today regarding usb wifi adapters than it was even 6 years ago. Realtek recently released another WiFi 6 chipset, the rtl8832bu. It appears to be WiFi 6 in that I am not seeing 6 GHz capability (WiFi 6e). Mediatek earlier this year released the mt7921au chipset which is a full blown WiFi 6e chipset. In fact, Mediatek had the driver for it, mt7921u, in the Linux kernel before products even hit the shelves. It used to be that you only saw thing from Intel, AMD and some of the other big players. I am seeing a lot of things that tell me we are making a lot of progress toward making many things on Linux easier to use. Hopefully, at some point, things gel and Linux can achieve more market share.

morrownr, 2022-11-14:
FYI: I just counted the hits that this site (morrownr) has had over the last two weeks: 21,528.

There are hits from all over the world from business as well as home users. I never expected that when I started and it continues to grow. Keep in mind this is not the type of site where users come back on a regular basis as most grab a driver or check some information and then may not be back for a year. That number is actually quite astounding given what this site does and it leads me thing think there are more Linux users out there on desktops, laptops and Pi's than most estimates allow for. Super computer and servers generally do not use usb wifi adapters so this is all coming from folks with desktops, laptops and things like RasPi's.

Jibun-no-Kage, 2022-11-14:
My experience as well, with Linux started with servers, and then VMware based "ESX" no ESXi (not the "i") on the bones of the anaconda code/startup to load their custom kernel, which was the 1st generation of their hypervisor emulation product, later they removed the emulation and built a true native hypervisor, and but still used anaconda.  The funny thing was, much of the ESX tool set was Linux compliant, even the driver set was Linux compliant, so I already knowing some Linux, as a Windows based engineer, got the opportunity to grow with VMware, the internal Linux team refused to touch virtualization... very bad strategic decision, and the Windows team just wanted to do Windows, long story short, the next 18 or more years, I had a career as an enterprise virtualization architect.  Loved it.  I still only use Linux desktop to a limited extent, maybe 60% Windows and 40% Linux, now.  Frankly I like being cross platform, even with microcontrollers, Pi, ESP8266, etc.   Have to keep the old brain challenged. :)

As for wireless, I have WiFi 6 on my newest micro PC, (Beelive GT-R 7) but all of my routers (mesh) are older and only do WiFi 2/3, which old, true, but for what I need, mostly IoT, just fine.  Even among my friends, that know nothing about WiFi hate changing things so they survive on routers that are 4+ years old and never change until they are forced to replace... not upgrade.  They will upgrade their phone twice a year, but the poor home wifi, is abused until dies.

I am sure the actual user base for Linux desktop under reported, agree on that, but out of my as 100 computer active friends and family, there are maybe 2 that even know that their phone is a type of Linux, when 50 are iPhone and 50 are Android.  They just don't care about knowing anything.  The 2 that want to know any thing about Linux, only want to know enough to avoid paying Microsoft $$$.  LOL.

Jibun-no-Kage, 2022-11-15:
Just FYI...

Once I got git clone of rpi-source working (issues including still needs python2), and on a fresh install, install-driver.sh worked as expected.  Seriously thinking of doing a PR for rpi-source to get it to python3.

morrownr, 2022-11-15:
> Seriously thinking of doing a PR for rpi-source to get it to python3.

Go for it and explain the big headache we are seeing. Maybe that will get their attention. I looked and I see at least 3 issues in other repos that I have worked over the last week that are this problem. In those, I found a thread that somehow works and pointed folks to it. Now I don't remember where it is but you know the problem so it doesn't matter.

> I am testing 8811au driver as well.

This repo is that driver. He he. BTW, what adapters do you have and would you mind if I add you to a list folks that I can call on if I need testing? I am working behind the scenes to reduce the number of reported issues and I have some target areas but I will need some folks to test as I go along.

> on a fresh install, install-driver.sh worked as expected.

That is good and I did read your other replies. I have been sick for the last few days and am still not fully up to speed so things are going slow.

Nick

Jibun-no-Kage, 2022-11-15:
Sure, add me the list.  I am already on it, technically, right?  I have done some limited testing using one of your other releases.  :)

As for adapters...

RTL8188EUS (few, generic)
RTL8188CUS (few, generic)
RTL8811AU (couple, generic)
RTL8821CU (couple, generic, one is BrosTrend, big antenna)
(I think I have a RTL8821AU, looking for it)

Plan to add a few more as I upgrade home environment, have not done much 5GHz, hoped 2.4GHz and 5GHz would be superseded by something better, but guess that is a pipe dream.

I usually use WiFi on raspberry pi devices, but I have the ability to test in various virtualization environments, and on ADM/Intel hardware.

I tend of like to use DKMS when possible, if supported by the GitHub release.  But will use manual compile if needed or often in testing.

I have older routers EA3500s and EA2500s, tend to run OpenWRT usually, and FreshTomato at times.  Am familiar with JTAG, serial firmware flashing for the routers I have.  I think I even have a working EA2000 and EA1200 in my graveyard.

Once I get a chance, I will send you my cheat notes on installation, I am planning to create SaltStack state scripts for installed dongles.  If I get a round to it may do it for ansible as well, just don''t have ansible environment setup at the moment.  Am getting tired of creating micro SD base images by hand a couple of times a year.  Keeping ARMv61, ARMv7 and ARMv8/aarch64 differences in text notes, time to use configuration management correctly.  Have you thought about adding SaltStack or Ansible templates to your distributions?  Just a thought, that came to mine.

All good... But I am long winded, so don't let me get started.  LOL.  Get well,

morrownr, 2022-11-16:
> Sure, add me the list. I am already on it, technically, right?

Well, I guess so because of the 8821cu driver. However, I am starting an actually document to keep track of assistants and what they have. This has gotten to be so much work that I have got to get this part of it more organized. Speaking of you already being on this list. I think I have the new 8821cu driver ready to go to a repo and then you can have access, It has everything but the extra monitor mode stuff but I can add that as time allows. I'll try to get it up and let you know this week. I think it has the potential to be a better driver.

> Plan to add a few more as I upgrade home environment, have not done much 5GHz, hoped 2.4GHz and 5GHz would be superseded by something better, but guess that is a pipe dream.

I can't sat if 6 GHz is going to be better because I only have one adapter that does 6 GHz so far. You are aware of the new mt7921au chipset? It already has an in-kernel driver and is WiFi 6 and WiFi 6e capable. I have a little cheap adapter that has the chipset and I did a speed test and posted the results... go to the Main Menu:

https://github.com/morrownr/USB-WiFi

Go to Issues and look at issue 139. Keep in mind the test is using WiFi 5 (AC) 5 Ghz. 

Alfa should be out with new adapters with the mt7921au chipset soon. One is said to be a high power version. Hmmm...

> All good... But I am long winded, so don't let me get started.

No problem. I enjoy talking about this stuff. I run OpenWRT on my main router.

Cheers

Jibun-no-Kage, 2022-11-16:
Do you have a specific or defined test plan you want to be followed?

Just a side question, have you considered at driver for rtl8188eu?  I have a few, and I use now, the ivanovborislav variant, rather than the native support.  The native support fails to gain an IPv4 address consistently, or such has been my experience.

I am looking for an mt7921 based adapter, there are a couple on ebay that look interesting, that use a PCI-e frame to host the Mediatek.

Right now, I use two Edge 4 Routers, one live the other configured but power off, in the event of a failure I can turn off (if needed) the primary, and via IoT power swtch, via my phone, turn on the backup.  I used to use a Cisco hardware VPN 800 series but when I retired, I lost access to the licensing, so looked for something that was a bit more robust, and since I did not want WiFi on my gateway router, it seemed a good option.  It has had a couple of stability bugs, but the vendor has provided decent support, even though the Edge 4 is a bit older, its firmware was updated just a few months ago, again.  My OMG fail back is a EA3500 with OpenWRT.  I also have an IpFire Firewall, but it really does not have to do much, just something fun to have around to keep up on firewall tends.  The Edge 4 supports robust managed VLANs, so at some point I will implement such to isolate the IoT stuff.  Some where I thought I read that the old 800 series Cisco devices can run open firmware so that is on my long term list.  :)

morrownr, 2022-11-17:
> Do you have a specific or defined test plan you want to be followed?

I do but let me get it up first. I'm going slow these days. I've been sick lately and am not getting much done..

> Just a side question, have you considered at driver for rtl8188eu?

Not really. I'm already maintaining 5 drivers and may add another before the end of the year but I am more or less maxed out. This is a lot of work. You are welcome to upload and maintain a driver here if you want and I can give a pointer or two as things progress.

> I am looking for an mt7921 based adapter, there are a couple on ebay that look interesting, that use a PCI-e frame to host the Mediatek. key-a or is it a key-e module (don't call from memory). I will wait for the release of the newer adapters as well.

I've seen a couple of other folks that have used a usb frame to host a little mt7921k card. Interesting. I might have to check that out.

Need to run for now. Not feeling good.

Nick

Jibun-no-Kage, 2022-11-17:
No worries, no rush, will be here when you are ready... seriously, get well.  REALLY... get well.

I will consider maintaining, I think with some a bit of help I might get the hang of it.  Really my Linux side C experience is initial issue, just need to spend some time with it.  And it has been a long time since I did any serious C coding, but it should be like riding a bike right?  LOL.

morrownr, 2022-11-17:
Feeling a little better this morning. I brought up the new private site:

https://github.com/morrownr/8821cu-20210916

You have access but since it is private, you have to log in. I will likely post a msg in the current repo (the public one) to invite others but I may wait until after the holiday (next week). I am planning to be out of town and won't have much time to address questions or issues until the week after the holiday). What do you think? Now or wait?

> Really my Linux side C experience is initial issue, just need to spend some time with it. And it has been a long time since I did any serious C coding, but it should be like riding a bike right? LOL.

My C coding is rusty but working on these drivers doesn't help with that as it is a very minor part of the job. I just take my time and look at a C reference book if I need to refresh my memory on C. What actually takes most of the time is dealing with users that have a messed up dev environments on their systems. Seriously! I'm working behind the scenes with a couple of others in an attempt to identify core causes of issues to see what we can do to reduce reports by >50%. I think we can do it. I have a good relationship with one retailer that is currently building a sorted list of their support tickets so we can make a plan. I've already done two things that can help many more are needed. These Realtek drivers are a support nightmare which is why I keep links and recommendations all over the place users to look at and use adapters with in-kernel drivers.

Nick

morrownr, 2022-11-17:
Here is my private email: morrownr@gmail.com

I will delete this message online after sending it to you. Please send me an email so that I know you got it.

morrownr, 2022-11-17:
>  "Dude, It is the holiday, man!" Or should have that been more of a Sean Penn voice?

I was thinking John Wayne or Clint Eastwood.

The repo is there for you so whenever you want to log in and play around, go for it:

https://github.com/morrownr/8821cu-20210916

We can talk more as we get a chance but I'm considering using this repo for some interesting testing as part of my "let's get rid of some of these reported issues" project. It is stuff that WILL break things badly at times as I figure out what works and what does not. It would be good to know what CPU arch's (x86, x86_64, arm and arm64) and distros/versions you have available for testing.

Cheers

Jibun-no-Kage, 2022-11-17:
Really?  I would never believe JW or CE would say "Dude" LOL

morrownr, 2022-11-17:
Well, no, I guess not, Maybe more of a "Listen here MF, it is a holiday!"

And then there is Genghis Khan "Kill 'em, he doesn't need a holiday."

Jibun-no-Kage, 2022-11-17:
LMAO... Or North... "It is likes a... some, er, Holiday... thing, don't you know... eh"  South... "Hey, Ese, We like only do... that day... when we are with family in the U S of A amigo.  Of course, my Dachshund, just say "BARK" and waits for food to fall from the sky.

Jibun-no-Kage, 2022-11-18:
Ok, first pass, both 32 and 64 bit Pi OS, got dkms error 10.  But... let me confirm, I want to create some completely clean images and test again.  I had run rpi-update and rpi-source on the images, so that might have broken something some how.