morrownr/8821au-20210708 #63: Ubuntu installed succesfully, but not detected

nxtwrld, 2022-11-26:
Opening new thread for Ubuntu issues. The USB devices is detected and driver is installed, but the system does not recognize any WiFi device and does not offer any networks. Any help would be appreciated.

dkms status
`rtl8821au/5.12.5.2, 5.19.0-23-generic, x86_64: installed`

lsusb
`Bus 001 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]`

iw reg get
`global country 00: DFS-UNSET (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN (2402 - 2472 @ 40), (N/A, 20), (N/A) (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN (57240 - 63720 @ 2160), (N/A, 0), (N/A)`

mokutil --sb-state
`SecureBoot disabled`

iwconfig
```
lo no wireless extensions.

eno1 no wireless extensions.
```

I have been struggling with it all day, but no luck.

I have tried to set the proper iw

```
sudo iw reg set CZ
and I get
global country CZ: DFS-ETSI

```
but it is gone after reboot

System is
```
Kubuntu 22.10
Kernel 5.19.0-23-generic
```

morrownr, 2022-11-26:
Hi @nxtwrld 

Given that secure boot is not an issue as it is not active and dkms shows the adapter to be installed, this issue goes beyond the normal check this and check that. Let me ask some questions:

- Does this adapter have bluetooth capability?

- If so, is the bluetooth capability working? Can you test it if you don't know?

Can I get you to run the following and post the results?

```
grep [[:alnum:]] /sys/module/8821au/parameters/*
```

FYI: Issues very similar to this come up with the 8821cu chipset so depending on your answers to the above, we may need to add one or more module parameters and we'll probably have to experiment to see what works.

nxtwrld, 2022-11-28:
I am getting 
`grep: /sys/module/8821au/parameters/*: No such file or directory`

morrownr, 2022-11-28:
You did not answer the first 2 questions that i asked.

nxtwrld, 2022-11-28:
It is the TP-LINK Archer T2U Plus without bluetooth.

morrownr, 2022-11-28:
Hmmm...

Run and post the results of:

```
rfkill list all
```

Also, have you tried a different usb port?

morrownr, 2022-11-28:
> The rfkill list all gives me empty response

That is not right. It should show something like this:

```
$ rfkill list all
0: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: hci1: Bluetooth
	Soft blocked: no
	Hard blocked: no
```

Is rfkill installed?

nxtwrld, 2022-11-28:
it looks installed
` rfkill --version
rfkill from util-linux 2.38`

nxtwrld, 2022-11-28:
but if I plugin a bluetooth dongle, it gives me
`0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
`

morrownr, 2022-11-28:
Okay, brain surgery time.

Since secure boot shows to not be active and the device id shows this to be the correct driver, it is time to dig...

Please post the results of:

sudo dkms status

We are looking for a conflicting driver. Have you previously installed another driver without fully removing it?

nxtwrld, 2022-11-28:
dkms status give me just one 
`rtl8821au/5.12.5.2, 5.19.0-23-generic, x86_64: installed`

but I did try to install other drivers previously, but I removed them using dkms.

I believe I tried https://github.com/cilynx/rtl88x2bu and https://github.com/lixz789/mt7610u_wifi_sta_v3002_dpo_20130916

morrownr, 2022-11-28:
> dkms status give me just one

Good. However, there could be a manually installed driver.

Recommend you do the following to make sure:

$ sudo ./remove-driver.sh
$ sudo ./install-driver.sh

The removal script will remove manually installed drivers as well as those installed with dkms. That should leave you clean and then you can install fresh.

> and https://github.com/lixz789/mt7610u_wifi_sta_v3002_dpo_20130916

If you have an adapter that uses a mt7610u chipset, installing a driver is unnecessary at that chipset is support with an in-kernel driver... in other words, it would be plug and play.

nxtwrld, 2022-11-28:
I removed the drivers (dkms was empty), rebooted, installed the driver and rebooted (dkms reports rtl88821au), but rfkill gives me still just the bluetooth dongle....

morrownr, 2022-11-29:
Time to backup a bit.

Which version of Ubuntu are you using? Server or desktop? Version 22.04?

What hardware is it running on?

nxtwrld, 2022-11-29:
Operating System: Kubuntu 22.10
KDE Plasma Version: 5.25.5
KDE Frameworks Version: 5.98.0
Qt Version: 5.15.6
Kernel Version: 5.19.0-23-generic (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Xeon® CPU E3-1245 v5 @ 3.50GHz
Memory: 31,3 GiB of RAM
Graphics Processor: Quadro M620/PCIe/SSE2
Product Name: HP Z2 Mini G3 Workstation

morrownr, 2022-11-29:
Nice system. 

I just reread this thread in an effort to see if there is something that i am missing. Nothing jumps out at me. What I can do is test here. One of my test systems has Xubuntu 22.10 installed so as I have time, hopefully today, I will try to duplicate this issue and let you know what I find.

Nick

rop12770, 2022-12-15:
I can't also use this on the latest kernel version on Kubuntu.
I need to start my system with kernel 5.15.0-52.

morrownr, 2022-12-15:
@nxtwrld @rop12770

Sorry for the delay. I will work on this later this morning. My test system will be Xubuntu 22.10 with kernel 5.19.

morrownr, 2022-12-15:
@nxtwrld @rop12770

I just installed this driver on a test system (x86_64) running Xubuntu 22.10 with secure boot on. It works. Kernel 5.19 is in use. I have nothing to go on as I was not able to duplicate the problem.

The question is what is different between my system and your systems?

You can crosscheck to make sure this is the right driver. Read file supported-device-IDs.

You can post the `options` line in 8821au.conf by running:

$ sudo ./edit-options.sh

You can run `sudo ./remove-driver.sh` as it  will remove any conflicting non-dkms installations ... and then install again. Post the screen output from the installation script.

You can post the results of:

$ lsmod

If is not clear why rfkill was not producing output.

You can go to the driver directory and run the following as a check:

$ make clean
$ make

Post the screen output.

rop12770, 2022-12-16:
> @nxtwrld @rop12770
> 
> Sorry for the delay. I will work on this later this morning. My test system will be Xubuntu 22.10 with kernel 5.19.

Thank You

rop12770, 2022-12-18:
Hello @morrownr 

I re-did the install.sh and now it's working.
Maybe some packages were missing on the previous attempt.
Thank You for your help.

nxtwrld, 2022-12-21:
Sorry, I was on vacation... I pulled the latest driver, tried the `sudo  ./remove-driver.sh`, rebooted and then did all the
`make clean
make
sudo make install`

rebooted, but  no lock, even though everything went nice and smooth.

running the `sudo ./edit-options.sh` I just get and empty nano editor.

Here is the `lsmod`:
`Module                  Size  Used by
rfcomm                 86016  16
snd_seq_dummy          16384  0
snd_hrtimer            16384  1
nvidia_uvm           1359872  0
cmac                   16384  3
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 32768  6 algif_hash,algif_skcipher
bnep                   28672  2
binfmt_misc            24576  1
intel_rapl_msr         20480  0
intel_rapl_common      40960  1 intel_rapl_msr
intel_tcc_cooling      16384  0
snd_ctl_led            24576  0
nvidia_drm             73728  7
snd_hda_codec_realtek   163840  1
snd_hda_codec_generic   102400  1 snd_hda_codec_realtek
ledtrig_audio          16384  2 snd_ctl_led,snd_hda_codec_generic
snd_hda_codec_hdmi     81920  1
btusb                  61440  0
snd_hda_intel          53248  4
nls_iso8859_1          16384  1
x86_pkg_temp_thermal    20480  0
btrtl                  24576  1 btusb
intel_powerclamp       24576  0
snd_intel_dspcfg       36864  1 snd_hda_intel
btbcm                  24576  1 btusb
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
btintel                45056  1 btusb
snd_hda_codec         172032  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
btmtk                  16384  1 btusb
nvidia_modeset       1146880  17 nvidia_drm
snd_seq_midi           20480  0
coretemp               24576  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_hda_core          118784  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
joydev                 32768  0
rapl                   20480  0
mei_pxp                20480  0
snd_hwdep              20480  1 snd_hda_codec
bluetooth             827392  46 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
snd_rawmidi            45056  1 snd_seq_midi
hp_wmi                 20480  0
sparse_keymap          16384  1 hp_wmi
ecdh_generic           16384  2 bluetooth
nvidia              40857600  1048 nvidia_uvm,nvidia_modeset
mei_hdcp               24576  0
intel_cstate           20480  0
platform_profile       16384  1 hp_wmi
wmi_bmof               16384  0
snd_pcm               155648  5 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
input_leds             16384  0
ee1004                 20480  0
ecc                    40960  1 ecdh_generic
snd_seq                77824  9 snd_seq_midi,snd_seq_midi_event,snd_seq_dummy
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
snd_timer              40960  3 snd_seq,snd_hrtimer,snd_pcm
drm_kms_helper        200704  1 nvidia_drm
fb_sys_fops            16384  1 drm_kms_helper
[... 201 lines cut ...]
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/halrf/halrf_psd.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/rtl8821a/halhwimg8821a_mac.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/rtl8821a/halhwimg8821a_bb.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/rtl8821a/halhwimg8821a_rf.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/halrf/rtl8812a/halrf_8812a_ce.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/halrf/rtl8821a/halrf_8821a_ce.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/halrf/rtl8821a/halrf_iqk_8821a_ce.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/phydm/txbf/haltxbfjaguar.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/btc/halbtc8821a1ant.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/hal/btc/halbtc8821a2ant.o
  CC [M]  /home/om/Downloads/8821au-20210708-main/platform/platform_ops.o
  LD [M]  /home/om/Downloads/8821au-20210708-main/8821au.o
  MODPOST /home/om/Downloads/8821au-20210708-main/Module.symvers
  CC [M]  /home/om/Downloads/8821au-20210708-main/8821au.mod.o
  LD [M]  /home/om/Downloads/8821au-20210708-main/8821au.ko
  BTF [M] /home/om/Downloads/8821au-20210708-main/8821au.ko
Skipping BTF generation for /home/om/Downloads/8821au-20210708-main/8821au.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-26-generic'
`

Sorry, for being such a pain in the.....

morrownr, 2023-01-04:
@nxtwrld 

Somehow I missed the above message. What is your status?

fumin, 2023-09-29:
I am hitting the same problem, and I see that in my case, there's an error message in `dmesg` when I plug in my tp-link AC600 Archer T2U Nano.

Error message:
```
[  416.148338] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000c1308944, val ffffffffc453db7c
```

Relevant info:
```
$ dmesg
[  382.531310] usb 1-5: USB disconnect, device number 2
[  415.811973] usb 1-5: new high-speed USB device number 6 using xhci_hcd
[  416.054168] usb 1-5: New USB device found, idVendor=2357, idProduct=011e, bcdDevice= 2.00
[  416.054173] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  416.054175] usb 1-5: Product: 802.11ac WLAN Adapter 
[  416.054177] usb 1-5: Manufacturer: Realtek 
[  416.054179] usb 1-5: SerialNumber: 00e04c000001
[  416.148338] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000c1308944, val ffffffffc453db7c
```

```
# rtl8821au, 5.12.5.2 is installed
$ dkms status
librealsense2-dkms, 1.3.19, 5.15.0-83-generic, x86_64: installed
librealsense2-dkms, 1.3.19, 5.15.0-84-generic, x86_64: installed
librealsense2-dkms, 1.3.19, 5.4.0-163-generic, x86_64: installed
nvidia, 535.104.05, 5.15.0-83-generic, x86_64: installed
nvidia, 535.104.05, 5.15.0-84-generic, x86_64: installed
nvidia, 535.104.05, 5.4.0-163-generic, x86_64: installed
rtl8821au, 5.12.5.2, 5.15.0-84-generic, x86_64: installed
rtl8821CU, 5.4.1, 5.15.0-84-generic, x86_64: installed
rtl8821CU, 5.4.1, 5.4.0-163-generic, x86_64: installed
```

```
# The TP-Link 802.11ac WLAN Adapter below is the adapter that is not working.
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0000:3825   USB OPTICAL MOUSE
Bus 001 Device 004: ID 2a7a:939f  CASUE USB KB
Bus 001 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 006: ID 2357:011e TP-Link 802.11ac WLAN Adapter 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

morrownr, 2023-09-29:
Hi @fumin 

Ouch. I have a couple of ideas but need more info:

Is this adapter capable of bluetooth?
What distro are you using and what version was originally installed?

fumin, 2023-09-29:
Thanks @morrownr for reaching out.

My adapter is a [tp-link Archer T2U Nano AC600](https://www.tp-link.com/us/home-networking/usb-adapter/archer-t2u-nano/) .
I think it supports only 2.4G and 5G wifi, without bluetooth.

My operating system is Ubuntu 20.04.6 LTS.
Since I did my installations today, my 8821au-20210708 driver version is the main branch at commit 19660ee3f67125d2ef31b626f9fff946bd3e8d01 .

Happy to provide more info, and run whatever commands that will bring out more info.

morrownr, 2023-09-29:
> Could this frantic installing and uninstalling cause issues?

Yes. In a couple of place on this site I post a saying: sudo is a weapon of mass destruction on a Linux system. Realtek out-of-kernel drivers are really bad. They are not designed for regular desktop and server installations. They are made for embedded use. I do my best to get the info into the hands of Linux users. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

You can read menu item 1 for good information and then look at menu item 2 for adapters that work without having to put up with this madness.

Multiple installations and removals is going to make this complicated and your system being based on a distro that is rather old also complicates things.

Do you have an adapter supported by the following driver?

rtl8821CU, 5.4.1, 5.15.0-84-generic, x86_64: installed
rtl8821CU, 5.4.1, 5.4.0-163-generic, x86_64: installed

If not, that really needs to be cleaned out as it is a very very old driver.

We can painfully start going over many things. This could take days. If it were me, I would consider saving the important files, then do a format so as to do a clean installation of a more modern disro.

Think about it and let know how you want to go forward.

Regards

fumin, 2023-09-30:
Thanks @morrownr for your info.

Regarding the question:

```
Do you have an adapter supported by the following driver?
rtl8821CU, 5.4.1, 5.15.0-84-generic, x86_64: installed
rtl8821CU, 5.4.1, 5.4.0-163-generic, x86_64: installed
If not, that really needs to be cleaned out as it is a very very old driver.
```

Yes, I do, it is a [Mercusys AC650 MU6H](https://www.mercusys.com/sg/product/details/mu6h) .
I wonder do you mean there may be conflicts between that rtl8821CU and this 8821au driver?

Regarding our ubuntu distro 20.04.6, unfortunately it's mandated by our company, and I guess it's unlikely I can convince my company to change in the short term.

I am aware this is going to be painful, but if you have further ideas, I am always grateful to try them out.

Thanks

morrownr, 2023-10-04:
Sorry for the delay. Health problems.

> I wonder do you mean there may be conflicts between that rtl8821CU and this 8821au driver?

No. I just like to see a tidy setup.

> Regarding our ubuntu distro 20.04.6, unfortunately it's mandated by our company, and I guess it's unlikely I can convince my company to change in the short term.

Understand. Is it possible for you to save your important files and then do a clean (format C: as we used to say) installation of 20.04.6? We basically have a system in an unknown configuration and I think it would help to change that.

> Thanks a lot for your list of drivers that plug-and-play in linux ! This is awesome! I am going to read through and pick one!

That list gets over 10,000 hit per week. What you could do is get an adapter that uses either the mt7612u or mt7610u chipset and there would be no more compiling. Plug it in and use it. There are sections of both chipsets in the Plug and Play list.

closed 2024-04-02 by morrownr