morrownr/8821au-20210708 #64: driver appears installed but iwconfig shows adapter not connecting to the wifi network

DC072012, 2022-11-28:
I searched all of the issues and don't see how to fix this.  The USB dongle is wlan1 in the following.  Any help will be greatly appreciated.

Background:
Linux raspberrypi 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l

lsusb
Bus 001 Device 004: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
Bus 001 Device 005: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
--------------
iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:"djnet-1"  
          Mode:Managed  Frequency:2.452 GHz  Access Point: 04:D4:C4:07:EF:F8   
          Bit Rate=39 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=37/70  Signal level=-73 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan1     unassociated  Nickname:"WIFI@RTL8821AU"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
--------------------------
ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:9c:78:e0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 634  bytes 224038 (218.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 634  bytes 224038 (218.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.57.175  netmask 255.255.255.0  broadcast 192.168.57.255
        inet6 fdfd:21f0:e140:495d:e99a:8d16:53e1:89db  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::99ef:3632:cec7:b88c  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:c9:2d:b5  txqueuelen 1000  (Ethernet)
        RX packets 994  bytes 262013 (255.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 287  bytes 38127 (37.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:c0:ca:a7:88:7c  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 2 overruns 0  carrier 0  collisions 0
---------------------

dkms status
rtl8821au, 5.12.5.2, 5.15.76-v7+, armv7l: installed

-----------------------------------
grep [[:alnum:]] /sys/module/8821au/parameters/*
/sys/module/8821au/parameters/rtw_ampdu_enable:1
/sys/module/8821au/parameters/rtw_beamform_cap:10
/sys/module/8821au/parameters/rtw_channel:1
/sys/module/8821au/parameters/rtw_drv_ant_band_switch:1
/sys/module/8821au/parameters/rtw_dynamic_agg_enable:1
/sys/module/8821au/parameters/rtw_en_dyn_rrsr:1
/sys/module/8821au/parameters/rtw_en_gro:1
/sys/module/8821au/parameters/rtw_en_napi:1
/sys/module/8821au/parameters/rtw_hiq_filter:1
/sys/module/8821au/parameters/rtw_ht_enable:1
/sys/module/8821au/parameters/rtw_hw_wps_pbc:1
/sys/module/8821au/parameters/rtw_ips_mode:1
/sys/module/8821au/parameters/rtw_led_ctrl:1
/sys/module/8821au/parameters/rtw_lps_chk_by_tp:1
/sys/module/8821au/parameters/rtw_pll_ref_clk_sel:15
/sys/module/8821au/parameters/rtw_power_mgnt:1
/sys/module/8821au/parameters/rtw_rrsr_value:-1
/sys/module/8821au/parameters/rtw_rx_stbc:1
/sys/module/8821au/parameters/rtw_scan_interval_thr:16000
/sys/module/8821au/parameters/rtw_stbc_cap:19
/sys/module/8821au/parameters/rtw_vcs_type:1
/sys/module/8821au/parameters/rtw_vht_24g_enable:1
/sys/module/8821au/parameters/rtw_vht_enable:1
/sys/module/8821au/parameters/rtw_wmm_enable:1

-----------------------------
 rfkill list all
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no

-------------------------------------
iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, NO-IR
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)

morrownr, 2022-11-28:
Hi @DC072012 

I really need you to read and follow the pinned issue called Problem Reporting Guidelines.

Also, `ifconfig` has been depreciated and is no longer supported. Please use `iw` and `ip` instead.

Nick

DC072012, 2022-11-28:
Sorry - I didn't see the pinned post.  I am now following pinned post to the best of my ability - please advise if I am still doing something wrong.  I appended my original post to the end of this.
Thank you.

Here is info in order requested, some of which is redundant with my first post I think.

Wifi dongle recognized but will not connect to wifi network.
----------------------
Linux raspberrypi 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l
------------------------
Raspberry pi 3B+
--------------

NOTE:  I tried ipconfig and got this:
ipconfig
-bash: ipconfig: command not found
_It does not work on my Pi_

Here is the output from the pinned post:
sudo uname -mr
5.15.76-v7+ armv7l
---------------
mokutil --sb-state
-bash: mokutil: command not found
---------
lsusb
Bus 001 Device 004: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
Bus 001 Device 005: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
------------
rfkill list all
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no

---------------
dkms status
rtl8821au, 5.12.5.2, 5.15.76-v7+, armv7l: installed
------------
iw dev
phy#1
	Interface wlan1
		ifindex 4
		wdev 0x100000001
		addr 00:c0:ca:a7:88:7c
		type managed
		txpower -100.00 dBm
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr ba:27:eb:c9:2d:b5
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:c9:2d:b5
		ssid djnet-1
		type managed
		channel 9 (2452 MHz), width: 20 MHz, center1: 2452 MHz
		txpower 31.00 dBm
-----------
iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
[... 120 lines cut ...]
Hard blocked: no
1: phy1: Wireless LAN
Soft blocked: no
Hard blocked: no

iw reg get
global
country US: DFS-FCC
(902 - 904 @ 2), (N/A, 30), (N/A)
(904 - 920 @ 16), (N/A, 30), (N/A)
(920 - 928 @ 8), (N/A, 30), (N/A)
(2400 - 2472 @ 40), (N/A, 30), (N/A)
(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, NO-IR
(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
(2402 - 2482 @ 40), (6, 20), (N/A)
(2474 - 2494 @ 20), (6, 20), (N/A)
(5140 - 5360 @ 160), (6, 20), (N/A)
(5460 - 5860 @ 160), (6, 20), (N/A)

morrownr, 2022-11-28:
> NOTE: I tried ipconfig and got this:
ipconfig
-bash: ipconfig: command not found
It does not work on my Pi

There is no `ipconfig` as far as I know. `ip` and `iw` are what I am talking about.

Example:

$ ip link
$ iw dev

Those replace `iwconfig`.

I'm seeing a lot of problems with the RasPiOS lately. It is not obvious what is going on here. I'll look at doing some testing here lately this week. Tell me exactly which iso of the RasPiOS you are using and tell me which RasPi is booting it?

You could try:

$ sudo apt update
$ sudo apt upgrade
$ sudo reboot
in the driver directory:
$ sudo ./remove-driver.sh
$ sudo reboot
sudo ./install-driver.sh

DC072012, 2022-11-28:
Thanks again.  The new info you requested is below:  I will try the sequence of steps you included and report back.  

-------------------------

ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:9c:78:e0 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether b8:27:eb:c9:2d:b5 brd ff:ff:ff:ff:ff:ff
4: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT mode DORMANT group default qlen 1000
    link/ether 00:c0:ca:a7:88:7c brd ff:ff:ff:ff:ff:ff

-----------
iw dev
phy#1
	Interface wlan1
		ifindex 4
		wdev 0x100000001
		addr 00:c0:ca:a7:88:7c
		type managed
		txpower -100.00 dBm
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr ba:27:eb:c9:2d:b5
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:c9:2d:b5
		ssid djnet-1
		type managed
		channel 9 (2452 MHz), width: 20 MHz, center1: 2452 MHz
		txpower 31.00 dBm

---------------
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
----------------------
cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000cf9c78e0
Model		: Raspberry Pi 3 Model B Plus Rev 1.3

morrownr, 2022-11-28:
> iw dev
phy#1
Interface wlan1
ifindex 4
wdev 0x100000001
addr 00:c0:ca:a7:88:7c
type managed
txpower -100.00 dBm

Okay, there we go. That is the wifi interface for your usb wifi adapter. All looks well. 

I also see that your internal wifi is connected to an AP on channel 9. Your wifi router/AP should really only be set to 1, 6, or 11 for the 2.4 GH band. You may want to disconnect that connection. In the README I have instructions for how to get rid of wifi connections on a RasPi. Once that connection is gone, try connecting with to a 5 GHz channel as that would have to be from the adapter.

FYI: I have a RasPi3B as well. I use the following version of RasPiOS with good success:

https://www.raspberrypi.com/software/operating-systems/

Raspberry Pi OS (64-bit)
Raspberry Pi OS with desktop

Also, FYI: The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

It has a lot of info.

DC072012, 2022-11-28:
Thanks again.  I did follow the remove / reinstall process.
I also changed the wifi channel on my router to channel 11.

USB adapter is seen but still is not connecting (see new ip and iw below)

I'll look at your website also and see if there is anything I can pick up there.

Updating to a new operating system is problematic unless it can be done remotely in headless operation.  The Pi is installed in a remote location and has a bunch of wires attached to the device it is monitoring.

If I try to connect with the 5 ghz channel and it doesn't work then I'm in a big problem and would have to drive to the remote site and hard reboot the Pi.

Further suggestions are welcome.
-----------
ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:9c:78:e0 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether b8:27:eb:c9:2d:b5 brd ff:ff:ff:ff:ff:ff
4: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT mode DORMANT group default qlen 1000
    link/ether 00:c0:ca:a7:88:7c brd ff:ff:ff:ff:ff:ff

------------
iw dev
phy#1
	Interface wlan1
		ifindex 4
		wdev 0x100000001
		addr 00:c0:ca:a7:88:7c
		type managed
		txpower -100.00 dBm
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr ba:27:eb:c9:2d:b5
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:c9:2d:b5
		ssid djnet-1
		type managed
		channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
		txpower 31.00 dBm

morrownr, 2022-11-28:
> If I try to connect with the 5 ghz channel and it doesn't work then I'm in a big problem and would have to drive to the remote site and hard reboot the Pi.

I understand. So that I can more fully understand, tell what your goal is? Is it to make wifi more reliable or to increase range?

> Doesn't that Pi also support 5GHZ without the dongle?

Let me ask my RasPi3B... it said, nope, I can only do 2.4 GHz band. I think dual band started with the RasPi3B+ but I would have to check on that.

The status as I see it right now is that the driver here has done its job. You have a wifi interface for the usb adapter. If it were me, I would probably leave things along until I am going to make a trip to that location anyway. Then I would put the correct line in the config.txt to shut down the internal wifi and then I would reboot and as you are using wpa_supplicant to connect, the new adapter should use your connection information and connect.

DC072012, 2022-11-28:
Thank you again.  I do have the RasPi3B+.

My goal is to increase the signal strength to get it more reliable.  Using the onboard wifi the signal is around -70db.  I'm expecting the USB would come in around -40 which would be less sensitive to other interference.  

Next time I'm going to be near the remote site I will test your idea of turning off the internal wifi and see if it will cause the USB to start working.

In the meantime if you do any further testing or have more suggestions please let me know.

DC072012, 2022-11-30:
I was able to get to the remote site.  I connected to the onboard wifi and added the "dtoverlay=disable-wifi" to the  /boot/config.txt file.  Power cycled the Pi, and the USB would not connect to wifi.  Tried this repeatedly.  

I also unplugged the wifi adapter, tried a different USB port, and power cycled the Pi.  Still the USB wifi would not connect.  

Then watching my router monitor the USB connected briefly, then was gone again and never reconnected.

I couldn't connect to the Pi without going through a bunch of hoops since the onboard wifi was turned off.  It was 30 degrees and raining so I swapped out the SD card with a backup so the Pi is working again on the onboard wifi.  I have a cron job to restart the onboard wifi when it goes down and that has run 3 times today.  So I still need to solve this.

If you have an opportunity to do some testing with my setup please let me know what you find. 

Thanks for your help.

closed 2024-04-02 by morrownr