morrownr/8821au-20210708 #66: The current master is full current right?

Jibun-no-Kage, 2023-01-04:
The current master is full current right?  Revised scripts, etc.?

morrownr, 2023-01-04:
Let me double check... be right back.

morrownr, 2023-01-04:
The current Main is up to date and is in sync with 8821cu-test. Go for it.

Jibun-no-Kage, 2023-01-04:
Cool, installed on Pi OS 11 64 and 32 Bit... smooth... for some reason of all the older WiFi Realtek Adapters I have the 8821AU based seem to be the best at scanning and connecting to APs.  Not that the onboard Pi WiFi is bad, but of the USB dongles the 8821AUs just seem the best at connecting holding throughput.

closed 2023-01-04 by Jibun-no-Kage

morrownr, 2023-01-04:
> Cool, installed on Pi OS 11 64 and 32 Bit... smooth... for some reason of all the older WiFi Realtek Adapters I have the 8821AU based seem to be the best at scanning and connecting to APs. Not that [...]

Concur. The 8811au and 8812au are very solid adapters and the drivers here are solid as well. I think the changes to the scripts we have been making will help inexperienced users get the drivers installed at a higher rate as well.

This reminds me that I think I need to finish syncing the 8812au repo...

Jibun-no-Kage, 2023-01-04:
Hey I even got a few other driver authors on github to step up to use DKMS as well, like for my 8818EU driver.  Maybe some of them will look at your architecture discovery methodology and adopt similar?  It is such a PITA to edit the Makefile or even run separate scripts that tweak it, before doing a manual make or using DKMS.

morrownr, 2023-01-04:
Hey, I see a future where you submit a PR. I don't have an 8818eu based adapter but I could look things over for you.

morrownr, 2023-01-04:
For what it is worth: I have designed the scripts to work across 5 different Realtek drivers with the need to only change the module name and version. That probably translates to other Realtek drivers as well but would have to check and test.

Jibun-no-Kage, 2023-01-04:
I don't want to steal it, or do I?  I bet I can get the current author of the driver I use to use a variant of your scripting, if you are cool with that?

morrownr, 2023-01-04:
I am cool with that. We might get a little standardization going... imagine that.