morrownr/8821au-20210708 #67: Debian 11 driver doesn't pick up usb wifi adapter

hman901, 2023-01-07:
uname -mr
5.10.0-20-amd64 x86_64
dkms status
rtl8821au, 5.12.5.2, 5.10.0-20-amd64, x86_64: installed
mokutil --sb-state
SecureBoot enabled
lsusb
Bus 001 Device 003: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
iw dev
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr 74:d8:3e:d0:42:f6
		type P2P-device
		txpower 0.00 dBm
	Interface wlp58s0
		ifindex 3
		wdev 0x1
		addr 74:d8:3e:d0:42:f6
		ssid Q_&_F.co
		type managed
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 22.00 dBm

lsmod
Module                  Size  Used by
ctr                    16384  1
ccm                    20480  3
rfcomm                 94208  0
cmac                   16384  3
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 32768  6 algif_hash,algif_skcipher
bnep                   28672  2
squashfs               69632  0
btusb                  65536  0
btrtl                  28672  1 btusb
btbcm                  20480  1 btusb
btintel                32768  1 btusb
snd_soc_skl_hda_dsp    28672  5
intel_pmc_core_pltdrv    16384  0
snd_soc_hdac_hdmi      45056  1 snd_soc_skl_hda_dsp
intel_pmc_core         45056  0
bluetooth             753664  33 btrtl,btintel,btbcm,bnep,btusb,rfcomm
snd_hda_codec_hdmi     73728  1
snd_hda_codec_realtek   163840  1
jitterentropy_rng      16384  1
snd_hda_codec_generic    98304  1 snd_hda_codec_realtek
msr                    16384  0
x86_pkg_temp_thermal    20480  0
snd_soc_dmic           16384  1
parport_pc             40960  0
intel_powerclamp       20480  0
drbg                   40960  1
snd_sof_pci            24576  0
snd_sof_intel_byt      24576  1 snd_sof_pci
coretemp               20480  0
ppdev                  24576  0
snd_sof_intel_ipc      20480  1 snd_sof_intel_byt
snd_sof_intel_hda_common   102400  1 snd_sof_pci
snd_sof_xtensa_dsp     16384  2 snd_sof_intel_hda_common,snd_sof_intel_byt
snd_sof               139264  4 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_byt,snd_sof_intel_ipc
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
kvm_intel             331776  0
ledtrig_audio          16384  2 snd_hda_codec_generic,snd_sof
snd_soc_skl           180224  0
mei_wdt                16384  0
lp                     20480  0
mei_hdcp               24576  0
parport                73728  3 parport_pc,lp,ppdev
[... 143 lines cut ...]
drm                   626688  7 drm_kms_helper,i915
crc32c_intel           24576  3
t10_pi                 16384  1 nvme_core
crc_t10dif             20480  1 t10_pi
crct10dif_generic      16384  0
ptp                    32768  1 e1000e
crct10dif_pclmul       16384  1
psmouse               184320  0
thunderbolt           299008  0
i2c_i801               32768  0
i2c_hid                32768  0
intel_lpss_pci         24576  0
intel_ish_ipc          28672  0
intel_lpss             16384  1 intel_lpss_pci
pps_core               24576  1 ptp
intel_ishtp            61440  2 intel_ishtp_hid,intel_ish_ipc
i2c_smbus              20480  1 i2c_i801
crct10dif_common       16384  3 crct10dif_generic,crc_t10dif,crct10dif_pclmul
hid                   151552  6 i2c_hid,usbhid,hid_multitouch,hid_sensor_hub,intel_ishtp_hid,hid_generic
idma64                 20480  0
usb_common             16384  3 xhci_hcd,usbcore,uvcvideo
wmi                    36864  3 hp_wmi,intel_wmi_thunderbolt,wmi_bmof
battery                24576  0
video                  61440  1 i915
button                 24576  0

morrownr, 2023-01-07:
Hi @hman901 

As I look over the info you posted, the thing that got my attention is:

SecureBoot enabled

This driver is certainly capable of working with SecureBoot but the installation scripts only set things up automatically for the distros that it knows how to support which includes all Ubuntu based distros.

The README.md file, the main screen, has info about SecureBoot in 3 places and the info in the FAQ gives a link that is about Debian. You can try the following first but if it does not work, please read the FAQ:

SecureBoot requires that the `openssl` and `mokutil` packages are installed so install them if they are not installed and run:

$ sudo mokutil --import /var/lib/dkms/mok.pub

After a reboot, see if the driver is used by the kernel.

Let me know. If you get it working, I'd like to add the info to the docs so that future Debian users have the info.

Nick

hman901, 2023-01-08:
Thanks it work with SecureBoot disabled

sudo mokutil --import /var/lib/dkms/mok.pub
Failed to get file status, /var/lib/dkms/mok.pub

closed 2023-01-08 by hman901