morrownr/8821au-20210708 #68: Doesn't work with Ourlink 1200MBPS USB Adapter in Ubuntu 22.04

own3mall, 2023-01-26:
lsusb shows:

`Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter`

However, using these drivers, everything compiles and installs, but Ubuntu 22.04 can't seem to use the adapter.  There is no wireless adapter according to network manager / settings.  

I'm using an Ourlink 1200MBPS USB AC Adapter.  Original drivers from the CD are here:  

https://mega.nz/file/Z1dmkCrQ#_a5_BCQa70ptYUP-N4gZ9tojTWwbUT7s-rkwgiMtPwQ

Any help?  Running kernel `5.15.0-58-generic`

morrownr, 2023-01-26:
Hi @own3mall 

> RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

This driver does not work with your adapter.

The 4th item in the README is:

```
Note: Please read "supported-device-IDs" for information about how to confirm that this is the correct driver for your adapter.
```

The site Main Menu can help you get to the correct driver:

https://github.com/morrownr/USB-WiFi

Click on menu item 5 for Realtek driver info.

Good luck.

closed 2023-01-26 by own3mall

morrownr, 2023-01-26:
Good. I'm trying to make it less complicated but that is how it is with Realtek based adapter. For less complicated, go to the Main Menu I posted above and select menu item 2. The Mediatek based adapters are less complicated.