morrownr/8821au-20210708 #70: Injection not working on archer t2u plus

RootLUG, 2023-02-06:
Hello,
I have recently bought [Archer T2U plus wifi adapter](https://www.tp-link.com/en/home-networking/high-gain-adapter/archer-t2u-plus/) and tried the driver included in this repository. The installation went without the errors and upon reboot the adapter is recognized by the system, however the list of capabilities in iw list says the frame injection is not supported and I verified it also running aircrack-ng suite and hcxdumptools to verify. The monitor mode however works fine. The documentation here and from searching github issues appears that the frame injection should be supported so I assume it is some bug in the driver?

The host platform is raspberry pi 4B with latest official kali for rpi4 (`Linux kali-raspberry-pi 5.15.44-Re4son-v7l+ #1 SMP Debian kali-pi (2022-07-03) armv7l GNU/Linux`). I tried also both usb2 and usb3 ports to see if there is any difference (there wasn't)

Are there some hw logs and/or commands I can try to debug this issue or provide more data on this issue?

morrownr, 2023-02-06:
Can I get you to provide the commands you were using to do the tests so that I can try to duplicate the results?

morrownr, 2023-02-06:
Test results:

```
$ sudo aireplay-ng --test wlan0mon
08:47:33  Trying broadcast probe requests...
08:47:33  Injection is working!
08:47:34  Found 50 APs

08:47:34  Trying directed probe requests...
08:47:34  90:58:51:09:54:33 - channel: 1 - 'SETUP-542F'
08:47:38  Ping (min/avg/max): 2.388ms/6.314ms/14.983ms Power: -73.73
08:47:38  11/30:  36%

08:47:38  7C:9A:54:84:5B:68 - channel: 1 - ''
08:47:40  Ping (min/avg/max): 2.876ms/14.674ms/30.017ms Power: -68.82
08:47:40  22/30:  73%

08:47:40  90:58:51:09:54:36 - channel: 1 - ''
08:47:43  Ping (min/avg/max): 5.911ms/12.971ms/21.734ms Power: -74.20
08:47:43  20/30:  66%

08:47:43  3C:B7:4B:05:0D:AC - channel: 1 - ''
08:47:43  Ping (min/avg/max): 5.625ms/11.386ms/19.104ms Power: -71.83
08:47:43  30/30: 100%

08:47:43  3C:B7:4B:05:0D:B0 - channel: 1 - ''
08:47:43  Ping (min/avg/max): 4.212ms/10.600ms/20.404ms Power: -72.07
08:47:43  30/30: 100%

08:47:43  F4:C1:14:02:39:76 - channel: 1 - ''
08:47:43  Ping (min/avg/max): 5.312ms/8.587ms/13.862ms Power: -57.13
08:47:43  30/30: 100%

...

```

> the list of capabilities in iw list says the frame injection is not supported

I need you to post a copy of this. I can't say that I recall "iw list" making a statement like this.

I have a repo that provides some help for monitor mode users. See menu item 10 at the Main Menu:

https://github.com/morrownr/USB-WiFi

If you will provide me the commands you used use hcxdumptool, I will test it also.

Regards,

Nick

morrownr, 2023-02-07:
@RootLUG 

While looking at this issue, I did find a bug that could contribute to you seeing what you are seeing. More info as I am able to post it.

gemesa, 2023-02-07:
> Hello, I have recently bought [Archer T2U plus wifi adapter](https://www.tp-link.com/en/home-networking/high-gain-adapter/archer-t2u-plus/) and tried the driver included in this repository.

I just bought the same HW and I am using `aircrack-ng` but I am running Fedora with kernel 6.1. I can support solving this issue if necessary. Here is my `aireplay-ng` output:
```
$ sudo ./aireplay-ng --test wlan0mon
08:49:51  Trying broadcast probe requests...
08:49:51  Injection is working!
08:49:53  Found 26 APs

08:49:53  Trying directed probe requests...
08:49:53  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:49:54  Ping (min/avg/max): 2.411ms/42.655ms/74.007ms Power: -83.33
08:49:54  30/30: 100%

08:49:54  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:49:56  Ping (min/avg/max): 7.715ms/40.539ms/135.651ms Power: -48.87
08:49:56  30/30: 100%

08:49:56  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:49:57  Ping (min/avg/max): 4.884ms/40.152ms/111.183ms Power: -50.33
08:49:57  30/30: 100%

08:49:57  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:49:58  Ping (min/avg/max): 18.170ms/34.448ms/61.587ms Power: -48.87
08:49:58  30/30: 100%

08:49:58  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:49:59  Ping (min/avg/max): 5.019ms/40.210ms/84.487ms Power: -49.13
08:49:59  30/30: 100%

08:49:59  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:00  Ping (min/avg/max): 16.031ms/38.863ms/58.867ms Power: -49.07
08:50:00  30/30: 100%

08:50:00  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:01  Ping (min/avg/max): 7.728ms/32.361ms/65.951ms Power: -80.33
08:50:01  30/30: 100%

08:50:01  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:02  Ping (min/avg/max): 7.320ms/29.457ms/58.634ms Power: -76.20
08:50:02  30/30: 100%

08:50:02  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:03  Ping (min/avg/max): 7.604ms/25.916ms/55.116ms Power: -71.73
08:50:03  30/30: 100%

08:50:03  XX:XX:XX:XX:XX:XX - channel: 1 - ''
08:50:06  Ping (min/avg/max): 38.557ms/100.748ms/185.907ms Power: -72.47
08:50:06  30/30: 100%

08:50:06  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:07  Ping (min/avg/max): 5.022ms/14.560ms/34.918ms Power: -74.00
08:50:07  30/30: 100%

08:50:07  XX:XX:XX:XX:XX:XX - channel: 1 - ''
08:50:10  Ping (min/avg/max): 27.904ms/99.595ms/189.479ms Power: -73.90
08:50:10  29/30:  96%

08:50:10  XX:XX:XX:XX:XX:XX - channel: 1 - ''
08:50:13  Ping (min/avg/max): 10.480ms/101.330ms/190.495ms Power: -74.26
08:50:13  27/30:  90%

08:50:13  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:14  Ping (min/avg/max): 2.310ms/28.751ms/51.475ms Power: -76.60
08:50:14  30/30: 100%

08:50:14  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:15  Ping (min/avg/max): 7.822ms/37.401ms/63.393ms Power: -51.47
08:50:15  30/30: 100%

08:50:15  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:16  Ping (min/avg/max): 8.465ms/23.883ms/50.975ms Power: -82.27
08:50:16  30/30: 100%

08:50:16  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:17  Ping (min/avg/max): 6.609ms/28.858ms/50.418ms Power: -82.93
08:50:17  30/30: 100%

08:50:17  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:22  Ping (min/avg/max): 11.013ms/11.013ms/11.013ms Power: -85.00
08:50:22   1/30:   3%

08:50:22  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:24  Ping (min/avg/max): 11.489ms/39.320ms/86.183ms Power: -71.73
08:50:24  30/30: 100%

08:50:24  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:25  Ping (min/avg/max): 5.339ms/48.326ms/122.663ms Power: -77.67
08:50:25  30/30: 100%

08:50:25  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:28  Ping (min/avg/max): 6.328ms/35.186ms/70.802ms Power: -83.62
08:50:28  16/30:  53%

08:50:28  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:30  Ping (min/avg/max): 6.679ms/48.460ms/175.920ms Power: -78.73
08:50:30  30/30: 100%

08:50:30  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:31  Ping (min/avg/max): 6.456ms/30.881ms/63.136ms Power: -76.40
08:50:31  30/30: 100%

08:50:31  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:32  Ping (min/avg/max): 6.550ms/33.722ms/120.859ms Power: -80.53
08:50:32  30/30: 100%

08:50:32  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:33  Ping (min/avg/max): 2.252ms/32.946ms/59.953ms Power: -73.40
08:50:33  30/30: 100%

08:50:33  XX:XX:XX:XX:XX:XX - channel: 1 - '<SSID>'
08:50:34  Ping (min/avg/max): 8.342ms/28.375ms/51.528ms Power: -80.47
08:50:34  30/30: 100%
```

RootLUG, 2023-02-07:
Hello,
I did som testing, and unfortunately I discovered it may have been using the wrong dkms driver as previously I tried a different one from different repo and seems that it wasn't remove properly when uninstalling so both drivers/modules were loaded
That' my bad and I should had veriefied it before posting the issue, I'm sorry for that.

Anyhow I had properly removed the previous driver and re-run the tests. The tests now reports that injection works although the numbers looks a little off when I compare it to another hardware (from wifi pineapple MK7), as if there is some filtering . I also noticed when changing the interface to monitor mode that it just changes from `link/ether` to `link/ieee802.11/radiotap` (when viewing via `ip addr`)  and there is no wlanXmon, is that expected?

```
hcxdumptool -i wlan1 --check_injection
initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 5934/186 proberesponse 3033
packet injection is working on 2.4GHz!
packet injection is working on 5GHz!
injection ratio: 25% (BEACON: 12078 PROBERESPONSE: 3033)
your injection ratio is average, but there is still room for improvement
antenna ratio: 61% (NETWORK: 31 PROBERESPONSE: 19)
your antenna ratio is good

terminating...
```

output of iw list, phy1/wlan1 is the usb t2u plus, I originally thought that the `Supported interface modes` section is supposed to report "frame_injection" but apparently it doesn't and I'm dumb):

[iw_list.txt](https://github.com/morrownr/8821au-20210708/files/10673291/iw_list.txt)

```
aireplay-ng -9 -i wlan0mon wlan1

09:10:08  Trying broadcast probe requests...
09:10:10  No Answer...
09:10:10  Found 0 APs

09:10:10  Trying card-to-card injection...
09:10:10  Attack -0:           OK
09:10:10  Attack -1 (open):    OK
09:10:10  Attack -1 (psk):     OK
09:10:11  Attack -2/-3/-4/-6:  OK
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
09:10:11  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xba7e).
... this repeats forever
```

```
# aireplay-ng -9 wlan1
15:39:13  Trying broadcast probe requests...
15:39:13  Injection is working!
15:39:15  Found 2 APs

15:39:15  Trying directed probe requests...
15:39:15  XXX - channel: 11 - 'wifi name1'
15:39:15  Ping (min/avg/max): 2.082ms/6.555ms/18.582ms Power: -59.60
15:39:15  30/30: 100%

15:39:15  XXX - channel: 12 - 'wifi name2'
15:39:16  Ping (min/avg/max): 4.391ms/19.748ms/47.842ms Power: -66.53
15:39:16  30/30: 100%
```

morrownr, 2023-02-07:
@RootLUG @gemesa @Jibun-no-Kage

There was an issue where the channel was not being changed properly at times in monitor mode. It appears that I have now fixed that. I also worked on some other parts of monitor mode and I plan on doing more work on monitor mode tomorrow. I should be doing a very large merge later this week with significant improvements. I could use the 3 of you to do some testing if you have time. I'll post here when things are ready to be tested.

Jibun,

I know you have a 8811au based adapter and I hope you have the time. I am also planning tomorrow to merge the remaining updates from the recent 8821cu project. As you know, this would likely have identified @RootLUG 's old driver and removed it for him during the installation process. So, if you have some time later this week or this weekend, I can use some help to make sure things are working well.

Nick

RootLUG, 2023-02-07:
@morrownr
I would be glad to help with testing just let me then know what tests to run

morrownr, 2023-02-07:
> I would be glad to help with testing just let me then know what tests to run

> I can also support with testing.

That is great. Give me a chance to do some locally testing before merging. This is going to be a big patch. I should break it up into several patches but I have too many other things I need to be working on while you guys test.

The main areas to test are the scripts and monitor mode. I also have a monitor mode script that is in a separate repo that I am going to update today as well and I'd like for you to test that script with the driver. I'll let you know when both repo are good to go.

Nick

morrownr, 2023-02-07:
Update: I think things are going well with the driver but I'm not sure if I will get the patch merged today or tomorrow. I'm testing right now.

Earlier I updated the below repo. This repo is something I have thrown together over time to help folks get into monitor mode. Before @Jibun-no-Kage says something... yes, I know it needs functions and a menu.

I'd like for you to test monitor mode with `start-mon.sh` plus anything else you want.

https://github.com/morrownr/Monitor_Mode

Jibun-no-Kage, 2023-02-08:
Sure... I believe I can do some testing just let me know when I can pull no problem.

morrownr, 2023-02-08:
Status: I am breaking the work into 2 parts. I just finished merging the work on the scripts and documentation. I'll post when the work on the code is merged.

morrownr, 2023-02-08:
@Jibun-no-Kage @RootLUG @gemesa 

Status: I just merged a lot of code changes and additions. It mostly had to do with monitor mode. You are cleared to start testing. Take your time. Please report results back here.

Please use the start-mon.sh script and guide to help guide things so that we can get the basic monitor mode stuff checked. If any of you use Wireshark and can do capture testing, that would be great. You are welcome to test any other functions and modes of the driver.

Of note: with the new install-driver.sh script, if you update `git pull` you no longer longer have to uninstall the previously installed driver because install-driver.sj will now detect installed drivers and remove them before installing. You can also test on ARM without having to run a script as the platform is now detected.

Also, I will be merged a few additional things over the next few days but it is just cosmetic stuff that should not affect your testing.

Thanks,

Nick

gemesa, 2023-02-09:
To me it is looking very good. Also I think `airodump-ng` is detecting access points much faster than before. So after pulling the new changes I ran the following:
```
$ sudo ./remove-driver.sh
$ reboot
$ sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230126
: x86_64 (architecture)
: 8/8 (in-use/total processing units)
: 16140456 (total system memory)
: 6.1.9-200.fc37.x86_64 (kernel version)
: gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
: dkms-3.0.10
: SecureBoot disabled
: ---------------------------
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
Sign command: /lib/modules/6.1.9-200.fc37.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...
Key file /var/lib/dkms/mok.key not found and can't be generated, modules won't be signed

Building module:
Cleaning build area...
./dkms-make.sh..............
Cleaning build area...
Compile time: 218.37 seconds
The driver was built by dkms successfully.

8821au.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.9-200.fc37.x86_64/extra/
depmod......
The driver was installed by dkms successfully.
Do you want to edit the driver options file now? [y/N]
Do you want to apply the new options by rebooting now? (recommended) [y/N]
$ reboot
$ sudo airmon-ng

PHY	Interface	Driver		Chipset

null	wlp0s20f0u1

$ sudo ./start-mon.sh wlp0s20f0u1
 --------------------------------
    start-mon.sh 20230206
 --------------------------------
    WiFi Interface:
             wlp0s20f0u1
 --------------------------------
    name  -  wlan0mon
    type  -  monitor
    state -  DORMANT
    addr  -  XX:XX:XX:XX:XX:XX
    chan  -  1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
    txpw  -  16.00 dBm
 --------------------------------

 Note: DORMANT = interface is up
       but inactive.

 The Interface is now ready for Monitor Mode use.

[... 47 lines cut ...]
 XX:XX:XX:XX:XX:XX  -81       23        0    0   9  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -56      107        0    0   6  195   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -70       50       13    0   6  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -77       70        6    0   5  195   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -76       73       11    0   9  195   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -77       62        0    0   9  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -74       62        0    0   6  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -82       32        0    0   1  270   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -82       31        1    0   1  540   WPA3 CCMP   SAE  <ESSID>
 XX:XX:XX:XX:XX:XX  -82        9        0    0   1  540   WPA2 CCMP   MGT  <ESSID>
 XX:XX:XX:XX:XX:XX  -81       27        0    0   1  270   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -88        6        0    0   1  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -95        4        0    0   1  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -31       91        0    0   1  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -87       22        0    0   1  130   WPA2 CCMP   PSK  <ESSID>
 XX:XX:XX:XX:XX:XX  -75       36        0    0   1  195   WPA2 CCMP   PSK  <length:  0>
```
 then after exiting `start-mon.sh` and returning the adapter to original settings:
```
$ sudo airmon-ng
 
 PHY	Interface	Driver		Chipset
 
 null	wlp0s20f0u1
```

morrownr, 2023-02-09:
@gemesa 

Thanks for the report.

> To me it is looking very good.

That is good news. There is always the possibility of breaking things when merging a LOT of new code. 

> Also I think airodump-ng is detecting access points much faster than before. 

Yes, this is possible. 

Nick

gemesa, 2023-02-09:
Should I test anything else?

morrownr, 2023-02-09:
@gemesa 

> Should I test anything else?

Well, yes. Can you go in depth on testing the core monitor capabilities such as:

- changing interface name
- entering and existing monitor mode
- changing channel (there was a bug here and I hope it is fixed now)
- changing mac address
- changing txpw

What you can do is keep 2 terminals open and while using start-min.sh, you can execute `iw dev` in the other terminal to make sure the changes are made correctly. Use channels from both bands. Keep in mind that txpw is locked on some adapters and I can't so anything about that. On the other hand, be careful because  some adapters will let you increase power to the point that you will burn the adapter.

Of note for and @RootLUG , There are two reasons that I wrote start-mon.sh:

- self-defense. I was getting too many issues from users that were using guides that were written for adapters that use in-kernel drivers. In-kernel drivers can do things these Realtek drivers cannot. This script actually works on Realtek drivers and in-kernel drives and it saves me a lot of reports.

- Going into monitor mode using Aircrack-ng is not that dependable in this day in time. It used to work well but as time has passed and cleaver coders have added code to spawn new processes very quickly, killing processes does not work that well. start-mon.sh does not kill interfering processes, it stops (pauses) them. This avoid the spawning of new processes and it lets you restart them without a reboot. Please use start-mon.sh for testing as it will lead to more reliable results.

Yes, I know start-mon.sh needs functions to reduce code duplication and it certainly would benefit from having an interactive menuing system but my to-do list is overflowing as usual. I tried to document the source and keep it organized so if anyone wants to make into something really good, they can understand where they are starting.

Nick

morrownr, 2023-02-09:
@RootLUG 

Good report.

> not sure what those radiotap errors are

Hmmm... I'll work on that with the time I have available today.

> these are without using that start-mon.sh script

As I noted in the previous message, be skeptical of using `airmon-ng start`. You may think it worked but the results you get can be less than accurate. Network Manager is particularly bad for rapidly spawning new processes in this day in time when you try to kill it but it is not the only guilty app.

RootLUG, 2023-02-09:
I will try to re-run the test tomorrow as I will have physical access to the device so I can then also disable the NetworkManager and other processes that may interfere

gemesa, 2023-02-09:
Working:
- changing interface name
- entering and exiting monitor mode
- changing mac address
- changing txpw

Not working:
- changing channel

It stays on channel 5 no matter what. I have tried multiple channels like 1, 6, 7, 10, 32, 34, 36.

```
$ sudo ./start-mon.sh wlp0s20f0u1
 --------------------------------
    start-mon.sh 20230206
 --------------------------------
    WiFi Interface:
             wlp0s20f0u1
 --------------------------------
    name  -  wlan0mon
    type  -  monitor
    state -  DORMANT
    addr  -  12:34:56:78:90:ab
    chan  -  5 (2432 MHz), width: 20 MHz (no HT), center1: 2432 MHz
    txpw  -  16.00 dBm
 --------------------------------

 Note: DORMANT = interface is up
       but inactive.

 The Interface is now ready for Monitor Mode use.

 You can place this terminal in
 the background while you run any
 applications you wish to run.

 Press any key to continue...
```
other terminal:
```
$ iw dev
phy#7
	Interface wlan0mon
		ifindex 10
		wdev 0x700000001
		addr 12:34:56:78:90:ab
		type monitor
		channel 5 (2432 MHz), width: 20 MHz (no HT), center1: 2432 MHz
		txpower 16.00 dBm
```
after exiting `start-mon.sh` and restoring settings:
```
$ iw dev
phy#7
	Interface wlp0s20f0u1
		ifindex 10
		wdev 0x700000001
		addr XX:XX:XX:XX:XX:XX
		type managed
		txpower -100.00 dBm

```

Btw I did some injection testing also and did not encounter any errors. After changing to monitor mode with `start-mon.sh` I ran:

```
$ sudo ./hcxdumptool -i wlan0mon --check_injection                  

This is a penetration testing tool!
It is made to detect vulnerabilities in your NETWORK mercilessly!
Don't report bugs if this tool does exactly what it was coded to do!

initialization of hcxdumptool 6.2.7 (depending on the capabilities of the device, this may take some time)...
warning possible interfere: NetworkManager is running with pid 1768

warning possible interfere: wpa_supplicant is running with pid 1829

BPF is unset. Make sure hcxdumptool is running in a 100% controlled environment!
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 5885/177 proberesponse 193   
packet injection is working on 2.4GHz!
packet injection is working on 5GHz!
injection ratio: 12% (BEACON: 1528 PROBERESPONSE: 193)
your injection ratio is poor - improve your equipment and/or get closer to the target
antenna ratio: 44% (NETWORK: 75 PROBERESPONSE: 33)
your antenna ratio is average, but there is still room for improvement

terminating...
$ sudo ./aireplay-ng --test wlan0mon
19:27:52  Trying broadcast probe requests...
19:27:52  Injection is working!
19:27:54  Found 11 APs

19:27:54  Trying directed probe requests...
19:27:54  XX:XX:XX:XX:XX:XX - channel: 1 - ''
19:27:54  Ping (min/avg/max): 2.660ms/19.735ms/39.047ms Power: -70.00
19:27:54  30/30: 100%

19:27:54  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:27:55  Ping (min/avg/max): 4.863ms/8.506ms/20.829ms Power: -32.60
19:27:55  30/30: 100%

19:27:55  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:27:56  Ping (min/avg/max): 4.426ms/20.655ms/37.842ms Power: -87.36
19:27:56  28/30:  93%

19:27:56  XX:XX:XX:XX:XX:XX - channel: 1 - ''
19:27:57  Ping (min/avg/max): 2.437ms/27.768ms/54.159ms Power: -82.86
19:27:57  29/30:  96%

19:27:57  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:27:57  Ping (min/avg/max): 3.136ms/17.321ms/33.731ms Power: -83.69
19:27:57  29/30:  96%

19:27:57  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:28:03   0/30:   0%

19:28:03  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:28:04  Ping (min/avg/max): 4.195ms/12.343ms/24.844ms Power: -85.90
19:28:04  29/30:  96%

19:28:04  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:28:10  Ping (min/avg/max): 1.829ms/1.829ms/1.829ms Power: -83.00
19:28:10   1/30:   3%

19:28:10  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:28:13  Ping (min/avg/max): 5.048ms/27.937ms/66.620ms Power: -95.11
19:28:13  19/30:  63%

19:28:13  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:28:19   0/30:   0%

19:28:19  XX:XX:XX:XX:XX:XX - channel: 1 - '<ESSID>'
19:28:19  Ping (min/avg/max): 3.182ms/9.238ms/15.870ms Power: -85.73
19:28:19  30/30: 100%
```

morrownr, 2023-02-09:
@gemesa 

> Not working:

> changing channel

> It stays on channel 5 no matter what. I have tried multiple channels like 1, 6, 7, 10, 32, 34, 36.

I am unable to duplicate this. Can I get you to refresh both repos and try again?

FYI: I have been updating the driver repo a lot. Before doing any testing:

$ git pull
$ sudo ./install-driver.sh

It would not hurt to do a `git pull` with Monitor Mode as well.

There were 3 large merges in the last 24 hours to the driver directory. One of which was this morning. There have been two merges to Monitor Mode today. Unless you update right before testing, you may not have current code.

Side note: I just realized that some distros are using IWD instead of wpa_supplicant these days so I need to work on start-mon.sh. What distro are you using? Has it swwitched to IWD? The reason I ask is this could account for what you are seeing.

Nick

morrownr, 2023-02-09:
> sudo hcxdumptool -i wlx0011c2ac47c7 --check_injection

> 1 radiotap error encountered during the test

@RootLUG @gemesa 

Could one of you gents that is more familiar with `hcxdumptool` than I am please tell me where the log is. That error is far from helpful.

morrownr, 2023-02-09:
We may have to send @Jibun-no-Kage out for pizza... this is turning into real work.

Jibun-no-Kage, 2023-02-10:
Great... If someone wants those little fishes on their pizza.... Good luck I don't do fish on pizza.... ah.... humm..... GO FISH.

morrownr, 2023-02-10:
Okay. The pizza was good. No little fish. The pineapple was good.

Now, I found a mistake that I just merged a fix for and it had to do with injection...update your repos and test test test...

Nick

gemesa, 2023-02-10:
> > Not working:
> >
> > changing channel
> 
> I am unable to duplicate this. Can I get you to refresh both repos and try again?

You are right, I refreshed it and now it is working.

gemesa, 2023-02-10:
I tested injection once again with `hcxdumptool` and `aireplay-ng` and got the same result as previously (no errors), poor injection rate for some targets which are relatively far but that is not a problem of the driver.

gemesa, 2023-02-10:
> Side note: I just realized that some distros are using IWD instead of wpa_supplicant these days so I need to work on start-mon.sh. What distro are you using? Has it swwitched to IWD? The reason I ask is this could account for what you are seeing.

I am using fedora 37 btw [which is using wpa_supplicant](https://nullr0ute.com/2023/01/using-iwd-for-wifi-in-fedora/).

morrownr, 2023-02-10:
@gemesa @

> I tested injection once again with hcxdumptool and aireplay-ng and got the same result as previously (no errors), poor injection rate for some targets which are relatively far but that is not a problem of the driver.

I have been testing with hcxdumptool as well as Aircrack-ng. I am new to hcxdumptool but since you were using it, I needed to become at least a little familiar with it. I have been concerned about how it operates and the results it is providing. My recommendation for now is that you consider discontinuing the use of hcxdumptool. I am considering listing it as non-supported in the README of this and the other Realtek drivers maintained here. Why?

From the hcxdumptool README:

```No support for a third party driver which is not part of the official kernel (https://www.kernel.org/)```

There is also more information which gets into some degree of detail but at the end of the day, the author of hcxdumptool spells out what is supported and this driver does not fall into that catagory. I tested this morning with Aircrack-ng and Wireshark and saw very good results with this driver. For what it is worth: Realtek's out-of-kernel drivers do not meet Linux Wireless Standards so non-support of these Realtek drivers is to be expected to one degree or another. Regarding what the author said above: He is saying that he only supports adapters that use in-kernel drivers. You may not have noticed but I encourage Linux users to buy adapters that are supported with in-kernel drivers because I am only too aware of the Realtek driver incompatibilities. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu item 2 gives a list of adapters that use in-kernel drivers. Menu item 5 covers the Realtek drivers.

If you are serious about security analysis/pen testing and need an adapter right now, I recommend the Alfa ACHM. If you are not in a hurry, Alfa is releasing the AXML later this month and it appears that their plan is to release a long range version by mid-year. The AXML and its long range cousin use the mt7921au chipset which is supported with an excellent in-kernel driver. It is also a tri-band WiFi 6 chipset.

Now, back to our regularly scheduled programming and ordering pizza with little fish on top.

So, let's continue testing as we need the best driver it can be.

Nick

Jibun-no-Kage, 2023-02-10:
Pi OS has not shifted to IWD as yet.  I was thinking of testing it at some point, but not a priority.  I searched the Debian forum and posted a question, since I could not find any specific strategic direction on Debian 12 moving to IWD.  As I recall IWD would retire DHCPCD as well, of course Network Manager which is not a documented option for Pi OS 11, does not leverage wpa_supplicant.

But all this is a bit like using ntp server vs. chrony vs. systemd-timesync, the Linux world seems to take a very long time to come to any consensus. seems most of the time, consolidation of methods only comes when active support for some option declines.  Not a great strategic methodology IMHO.

Not the best option survives, but the option that seems to be the most fun to work on?  Too harsh?

Jibun-no-Kage, 2023-02-10:
[Laughing so hard... crying.... The fish comment.]

gemesa, 2023-02-10:
> My recommendation for now is that you consider discontinuing the use of hcxdumptool.

Sure, I can stop using it. I have been using `aircrack-ng` only previously, and for some reason I thought you specifically asked us to test with `hcxdumptool` as well but now scrolling up back I see no such message. Maybe it was in my dreams lol.
Anyway, thanks for the explanation and adapter recommendations. I have been looking at Alfa ACHM for a long time but I think I will order an AXML instead.

RootLUG, 2023-02-10:
I've re-run the tests, this time I disabled all problematic programs such as NetworkManager and dhclient as to not infere with the results:
```
┌──(root㉿kali-raspberry-pi)-[~]
└─# hcxdumptool -i wlan0mon --check_injection
initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 5934/186 proberesponse 2894
packet injection is working on 2.4GHz!
packet injection is working on 5GHz!
injection ratio: 23% (BEACON: 12456 PROBERESPONSE: 2894)
your injection ratio is poor - improve your equipment and/or get closer to the target
antenna ratio: 65% (NETWORK: 35 PROBERESPONSE: 23)
your antenna ratio is good

terminating...

┌──(root㉿kali-raspberry-pi)-[~/8821au-20210708]
└─# aireplay-ng -9 -i wlan2 wlan0mon

16:55:55  Trying broadcast probe requests...
16:55:57  No Answer...
16:55:57  Found 2 APs

16:55:57  Trying directed probe requests...
16:55:57  XXXX - channel: 11 - 'essid'
16:55:57  Ping (min/avg/max): 1.436ms/4.632ms/14.696ms Power: -59.13
16:55:57  30/30: 100%

16:55:57  Injection is working!

16:55:57  XXXX- channel: 12 - 'essid'
16:55:57  Ping (min/avg/max): 1.691ms/9.779ms/33.723ms Power: -70.67
16:55:57  30/30: 100%

16:55:57  Trying card-to-card injection...
16:55:57  Attack -0:           OK
16:55:57  Attack -1 (open):    OK
16:55:57  Attack -1 (psk):     OK
16:55:57  Attack -2/-3/-4/-6:  OK
16:55:57  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x1ac5).
16:55:57  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x1ac5).

... error repeats

┌──(root㉿kali-raspberry-pi)-[~/8821au-20210708]
└─# aireplay-ng -9 wlan0mon
16:58:39  Trying broadcast probe requests...
16:58:39  Injection is working!
16:58:41  Found 2 APs

16:58:41  Trying directed probe requests...
16:58:41  XXXX - channel: 11 - 'essi'
16:58:41  Ping (min/avg/max): 1.958ms/7.687ms/24.294ms Power: -63.20
16:58:41  30/30: 100%

16:58:41  XXXX - channel: 12 - 'essid'
16:58:41  Ping (min/avg/max): 2.156ms/10.151ms/37.680ms Power: -63.80
16:58:41  30/30: 100%
```

RootLUG, 2023-02-13:
Is kismet for you guys working fine with the latest version?
I was now preparing the system for another wardriving run and noticed that kismet is no longer able to capture data from the t2u plus. upon enabling the source, all the wlans on system will become unavailable in a sense they are no longer listed as data sources in kismet. This happens even if the interface is actived in monitor mode using that start-monitor.sh script, which I noticed also gives the following errors (running as root, same as before when doing aireplay tests so it's not a permission error):

```
 --------------------------------
    start-mon.sh 20230206
 --------------------------------
    WiFi Interface:
             wlan1
 --------------------------------
    name  -  wlan1
    type  -  managed
    state -  DOWN
    addr  -  XXXX
 --------------------------------

 Do you want to set a new addr? [y/N]

command failed: Operation not permitted (-1)
RTNETLINK answers: Operation not permitted
```

the script also creates the wlan0mon inteface but it is not in a monitor mode (radiotap) but a classic eth/wlan device. When running airodump on the interface I get the following error:
```
��# airodump-ng wlan0mon
Warning: Detected you are using a non-UNICODE terminal character encoding.
ioctl(SIOCSIWMODE) failed: Operation not permitted
ioctl(SIOCSIWMODE) failed: Operation not permitted
Error setting monitor mode on wlan0mon
Failed initializing wireless card(s): wlan0mon
```
Seems like the card has now problem switching into the monitor mode. I tried reinstalling the dkms driver as well as upgrading via apt to latest kali but the issue persist. I will try on a freshly installed system on another raspberry to rule out the issue of system being messed up by my doing later today. In dmesg I see the following errors:
```
[  310.778659] Hardware name: BCM2711
[  310.778667] Backtrace:
[  310.778676] [<c0bfe690>] (dump_backtrace) from [<c0bfe8dc>] (show_stack+0x20/0x24)
[  310.778699]  r7:00000178 r6:00000000 r5:c10353f8 r4:60080013
[  310.778708] [<c0bfe8bc>] (show_stack) from [<c0c0324c>] (dump_stack_lvl+0x70/0x94)
[  310.778726] [<c0c031dc>] (dump_stack_lvl) from [<c0c03288>] (dump_stack+0x18/0x1c)
[  310.778745]  r7:00000178 r6:00000009 r5:c050f3f8 r4:c1050ec8
[  310.778753] [<c0c03270>] (dump_stack) from [<c02264b4>] (__warn+0xfc/0x158)
[  310.778770] [<c02263b8>] (__warn) from [<c0bff034>] (warn_slowpath_fmt+0xa4/0xe4)
[  310.778789]  r7:00000009 r6:00000178 r5:c1050ec8 r4:c1050f50
[  310.778797] [<c0bfef94>] (warn_slowpath_fmt) from [<c050f3f8>] (proc_register+0x138/0x1bc)
[  310.778815]  r8:c280c2d4 r7:c280c280 r6:c5d2c780 r5:00000008 r4:c42fdc58
[  310.778823] [<c050f2c0>] (proc_register) from [<c050f800>] (proc_mkdir_data+0x74/0xa8)
[  310.778840]  r10:c280c280 r9:c4fc0000 r8:c4fc0000 r7:bf64c6e4 r6:c280c280 r5:c52b2700
[  310.778850]  r4:c4fc0000
[  310.778857] [<c050f78c>] (proc_mkdir_data) from [<bf5b235c>] (rtw_adapter_proc_init+0x54/0x174 [8821au])
[  310.779318]  r4:f084d000
[  310.779324] [<bf5b2308>] (rtw_adapter_proc_init [8821au]) from [<bf5b25a4>] (rtw_adapter_proc_replace+0x88/0xac [8821au])
[  310.780177]  r9:c4fc0000 r8:f0850000 r7:bf64c6e4 r6:0000029e r5:c52b2700 r4:0000029e
[  310.780188] [<bf5b251c>] (rtw_adapter_proc_replace [8821au]) from [<bf583104>] (rtw_ndev_notifier_call+0xb8/0xc0 [8821au])
[  310.780978]  r10:c4fc0340 r9:c1527994 r8:c15ed8c8 r7:c57898ec r6:0000000b r5:0000000b
[  310.780989]  r4:c4fc0000 r3:00000001
[  310.780997] [<bf58304c>] (rtw_ndev_notifier_call [8821au]) from [<c024f3a4>] (raw_notifier_call_chain+0x44/0x78)
[  310.781410]  r5:ffffffed r4:bf062974
[  310.781418] [<c024f360>] (raw_notifier_call_chain) from [<c0a754bc>] (call_netdevice_notifiers_info+0x5c/0xa8)
[  310.781439]  r7:c15ecd40 r6:c15ecd40 r5:0000000b r4:c57898ec
[  310.781447] [<c0a75460>] (call_netdevice_notifiers_info) from [<c0a84c4c>] (dev_change_name+0x19c/0x2d4)
[  310.781467]  r6:00000000 r5:c5789bc0 r4:c4fc0000
[  310.781475] [<c0a84ab0>] (dev_change_name) from [<c0a955ec>] (do_setlink+0x874/0xac0)
[  310.781495]  r10:00000000 r9:c5789bc0 r8:c3d6aa80 r7:00000000 r6:c5789cb8 r5:c5789ad8
[  310.781506]  r4:c4fc0000
[  310.781512] [<c0a94d78>] (do_setlink) from [<c0a95fa4>] (__rtnl_newlink+0x5d4/0x910)
[  310.781532]  r10:00000000 r9:c5789cb8 r8:c6139000 r7:c4fc0000 r6:00000000 r5:c3d6aa80
[  310.781543]  r4:00000000
[  310.781549] [<c0a959d0>] (__rtnl_newlink) from [<c0a96334>] (rtnl_newlink+0x54/0x74)
[  310.781569]  r10:c1a92540 r9:c3d6aa80 r8:c15ed910 r7:c5789cb8 r6:c6139000 r5:c3d6aa80
[  310.781579]  r4:c372ea00
[... 68 lines cut ...]
[  310.785515]  r4:c3d6aa80
[  310.785522] [<c0ae9f50>] (netlink_rcv_skb) from [<c0a923c4>] (rtnetlink_rcv+0x20/0x24)
[  310.785540]  r7:c1406cc8 r6:c3d6aa80 r5:c1a92440 r4:c18efc00
[  310.785549] [<c0a923a4>] (rtnetlink_rcv) from [<c0ae9640>] (netlink_unicast+0x220/0x300)
[  310.785566] [<c0ae9420>] (netlink_unicast) from [<c0ae9930>] (netlink_sendmsg+0x210/0x4a8)
[  310.785585]  r10:00000000 r9:00000000 r8:c3c69c00 r7:00000030 r6:c3d6aa80 r5:c5789f48
[  310.785596]  r4:00000030
[  310.785603] [<c0ae9720>] (netlink_sendmsg) from [<c0a53fe4>] (sock_sendmsg+0x44/0x54)
[  310.785621]  r10:c5789dcc r9:00000000 r8:00000000 r7:c236e080 r6:00000000 r5:c236e080
[  310.785631]  r4:c5789f48
[  310.785638] [<c0a53fa0>] (sock_sendmsg) from [<c0a54bfc>] (____sys_sendmsg+0x1e8/0x21c)
[  310.785653]  r5:00000000 r4:c5789f48
[... cut ...]
[  310.785711]  r8:c0200244 r7:00000128 r6:be8e1a28 r5:00000000 r4:c236e080
[  310.785719] [<c0a5709c>] (sys_sendmsg) from [<c0200040>] (ret_fast_syscall+0x0/0x1c)
[  310.785735] Exception stack(0xc5789fa8 to 0xc5789ff0)
[  310.785746] 9fa0:                   00000000 be8e1aa4 00000003 be8e1a28 00000000 00000001
[  310.785758] 9fc0: 00000000 be8e1aa4 00000004 00000128 63ea2f81 00000000 004dfc4c 00000000
[  310.785770] 9fe0: 00000128 be8e19d0 b6d45c2b b6cae616
[  310.785779]  r6:00000004 r5:be8e1aa4 r4:00000000
[  310.785788] ---[ end trace 42823aa49a845154 ]---
```

RootLUG, 2023-02-13:
I can confirm it is also present on a freshly installed system, I flashed kali on rpi4 from stock image, installed driver from git, started kismet and/or airodump and got the same error. interestingly on some reboots, kismet was able to read data when it was first put into monitor mode manually but when kismet was restarted it got the same issue. I haven't found a reliable way to reproduce or find out what is causing it yet. I also tried rolling back to commit 174a1cbd901c23a21dceebec88673b16a7fe3b30 but that didn't helped either. It is quite interesting as the same commit has worked before, is there perhaps some persistence where something is set on adapter side that could cary on when plugged into the freshly installed system?

morrownr, 2023-02-14:
> I haven't found a reliable way to reproduce or find out what is causing it yet.

Me either but I am trying.

> It is quite interesting as the same commit has worked before, is there perhaps some persistence where something is set on adapter side that could cary on when plugged into the freshly installed system?

Same here. In fact, I have not touched the code in the older version of this driver except for documentation and scripts in a long time and it shows the same problems now. In fact, the other 3 Realtek drivers that we have been working on lately all show the same problem. I've tested on kernels 5.19 and 6.1. Same problem.

What I know is that things go downhill at the point that the script changes the interface name. I checked without using the script and it happens manually as well.

I am going to continue searching for the problem but this is a mystery right now. Nothing makes sense. It is like stepping in a pile Dog do. You have no idea what dog left it and now you can't get it off of your shoe.

Nick

morrownr, 2023-02-14:
The sad saga continues...

I decided to see if the problem exists in a Realtek driver from another author:

```
rtl88x2bu 3-3:1.0 wlan0mon: renamed from wlxbcec434157ce
[  206.929790] RTW: rtw_ndev_notifier_call(wlan0mon) state:11
[  206.929836] ------------[ cut here ]------------
[  206.929837] name 'wlan0'
[  206.929843] WARNING: CPU: 3 PID: 2796 at fs/proc/generic.c:712 remove_proc_entry+0x149/0x1c0
[  206.929848] Modules linked in: intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp mt76x0u coretemp mt76x0_common snd_hda_codec_realtek kvm_intel snd_hda_codec_generic mt76x02_usb ledtrig_audio nls_iso8859_1 snd_hda_codec_hdmi uvcvideo mt76_usb snd_hda_intel snd_intel_dspcfg kvm snd_intel_sdw_acpi videobuf2_vmalloc snd_hda_codec snd_usb_audio dcdbas snd_hda_core mt76x02_lib videobuf2_memops videobuf2_v4l2 rapl mt76 snd_usbmidi_lib mei_pxp mei_hdcp dell_smm_hwmon snd_seq_midi videobuf2_common intel_cstate snd_hwdep snd_seq_midi_event 88x2bu(OE) snd_pcm joydev snd_rawmidi mac80211 videodev at24 input_leds mc snd_seq snd_seq_device cfg80211 libarc4 snd_timer snd mei_me soundcore mei mac_hid sch_fq_codel msr parport_pc ppdev lp parport ramoops reed_solomon efi_pstore pstore_blk pstore_zone ip_tables x_tables autofs4 btrfs blake2b_generic xor raid6_pq zstd_compress libcrc32c dm_mirror dm_region_hash dm_log hid_logitech_hidpp hid_logitech_dj amdgpu iommu_v2 gpu_sched
[  206.929900]  drm_buddy uas usb_storage hid_generic usbhid hid radeon i2c_algo_bit drm_ttm_helper ttm drm_display_helper crct10dif_pclmul crc32_pclmul cec polyval_clmulni rc_core polyval_generic ghash_clmulni_intel sha512_ssse3 drm_kms_helper aesni_intel syscopyarea sysfillrect sysimgblt fb_sys_fops crypto_simd ahci cryptd i2c_i801 drm xhci_pci i2c_smbus xhci_pci_renesas r8169 libahci lpc_ich realtek video wmi
[  206.929923] CPU: 3 PID: 2796 Comm: ip Tainted: G           OE      6.1.9-060109-generic #202302010835
[  206.929925] Hardware name: Dell Inc. Vostro 470/0YJPT1, BIOS A14 06/25/2018
[  206.929927] RIP: 0010:remove_proc_entry+0x149/0x1c0
[  206.929929] Code: 00 48 c7 c7 a8 cb bd 9c e8 22 51 99 00 0f 0b 48 c7 c7 a8 dc d7 9d e8 a6 32 a6 00 4c 89 e6 48 c7 c7 ee ca bd 9c e8 05 51 99 00 <0f> 0b eb 9d 48 c7 c7 a8 dc d7 9d e8 87 32 a6 00 eb 8f 41 83 6d 60
[  206.929931] RSP: 0018:ffffa2bb02cc36e0 EFLAGS: 00010246
[  206.929933] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[  206.929934] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[  206.929935] RBP: ffffa2bb02cc3710 R08: 0000000000000000 R09: 0000000000000000
[  206.929936] R10: 0000000000000000 R11: 0000000000000000 R12: ffffa2bb003ea2d8
[  206.929937] R13: ffff8f95177bbcc0 R14: ffffa2bb003e5000 R15: ffff8f95177bbcc0
[  206.929938] FS:  00007ff5d1aecb80(0000) GS:ffff8f9712d80000(0000) knlGS:0000000000000000
[  206.929940] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  206.929941] CR2: 00007ff49910fe24 CR3: 000000011a048001 CR4: 00000000001706e0
[  206.929943] Call Trace:
[  206.929944]  <TASK>
[  206.929948]  rtw_adapter_proc_replace+0x77/0xb4 [88x2bu]
[  206.930028]  rtw_ndev_notifier_call+0x80/0x9a [88x2bu]
[  206.930094]  raw_notifier_call_chain+0x49/0x70
[  206.930097]  call_netdevice_notifiers_info+0x52/0xa0
[  206.930100]  dev_change_name+0x223/0x350
[  206.930104]  do_setlink+0xc26/0xe20
[  206.930108]  ? __nla_validate_parse+0x138/0x1c0
[  206.930113]  __rtnl_newlink+0x6b4/0x710
[  206.930116]  rtnl_newlink+0x48/0x80
[  206.930119]  rtnetlink_rcv_msg+0x170/0x430
[  206.930121]  ? __kmem_cache_alloc_node+0x1cc/0x2f0
[  206.930125]  ? __alloc_skb+0x97/0x1d0
[  206.930128]  ? rtnl_calcit.isra.0+0x150/0x150
[  206.930130]  netlink_rcv_skb+0x57/0x110
[  206.930133]  rtnetlink_rcv+0x15/0x30
[  206.930135]  netlink_unicast+0x247/0x390
[  206.930139]  netlink_sendmsg+0x25e/0x4e0
[  206.930142]  sock_sendmsg+0x6d/0x70
[  206.930145]  ____sys_sendmsg+0x288/0x320
[  206.930148]  ___sys_sendmsg+0x9a/0xf0
[  206.930152]  __sys_sendmsg+0x89/0xf0
[  206.930154]  __x64_sys_sendmsg+0x1d/0x30
[  206.930156]  do_syscall_64+0x5b/0x90
[  206.930158]  ? irqentry_exit+0x43/0x50
[  206.930161]  ? exc_page_fault+0x91/0x1b0
[  206.930163]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[  206.930167] RIP: 0033:0x7ff5d1927b17
[  206.930169] Code: 0f 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b9 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 89 54 24 1c 48 89 74 24 10
[  206.930170] RSP: 002b:00007fff0621b148 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[  206.930172] RAX: ffffffffffffffda RBX: 0000000063eb9f1e RCX: 00007ff5d1927b17
[  206.930173] RDX: 0000000000000000 RSI: 00007fff0621b1b0 RDI: 0000000000000003
[  206.930174] RBP: 0000000000000000 R08: 0000000000000001 R09: 000055fb19f162b0
[  206.930175] R10: 0000000000000078 R11: 0000000000000246 R12: 0000000000000001
[  206.930176] R13: 00007fff0621b260 R14: 0000000000000000 R15: 000055fb18672040
[  206.930178]  </TASK>
[  206.930179] ---[ end trace 0000000000000000 ]---
```

Same thing. I don't see anything in the log until attempting to change the interface name. Since the problem is showing up on different kernels, I am beginning to wonder if something has flowed into our systems that is causing this breakage.

What distro and version are you using? I am using an Ubuntu variant based on Ubuntu 22.04.

morrownr, 2023-02-14:
Can I get you to use another method of going into monitor mode besides start-mon.sh so we can rule it out?

morrownr, 2023-02-14:
I am worn out. I have stepped through the code. I have tested two external Realtek drivers from devs I trust and they show the same problem. This problem is not cause by a recent merge and everything was working fine until a couple of days ago. It is obviously aliens causing this... hey, I have no other explanation at this point.

I have now tested start-mon.sh along with packet injection with my Alfa ACHM that uses the in-kernel mt7610u driver and it works fine. In fact, the following is all I see in the log when the interface name is changed:

[ 3255.475339] mt76x0u 1-1.1:1.0 wlan0mon: renamed from wlx11c0caad4bcc

For now, I am going to have to sit on my hands and wait for more information. I am like a detective that has no leads.

Let me know if you discover anything else.

morrownr, 2023-02-14:
@alkisg

You may want to read this thread.

The only other thing I can think of right now is to test on non-Ubuntu/Debian distros.

morrownr, 2023-02-14:
@ivanovborislav

You may want to read this thread also. I downloaded and tested your rtl88x2bu driver and I see the same problem.

Please let know if you have anything to add.

RootLUG, 2023-02-14:
> The only other thing I can think of right now is to test on non-Ubuntu/Debian distros.

I can confirm it is also an issue on other distros, I just tested it on a freshly installed Manjaro on RPI4 (&updated via pacman) and it's doing the same thing, both airmon and kismet also hangs/crash when trying to switch into monitor mode, this time there wasn't any output in dmesg on manjaro. Another difference I noticed is that on debain/ubuntu based system I can't even shutdown the system properly as it just "hangs" and stops responding while Manjaro is at least able to shut down cleanly.

I am thinking what could even be the difference now that it is crashing, before when it was working vs now I flashed the system from the same image onto the sdcard so from that perspective the system itself must be identical, I also avoided upgrading any packages to rule it it's not some package/update that breaks it. So when testing the drivers the only steps were logging into the fresh system via ssh, git clone, install driver, run kismet/airmon/hcxdumptool etc.. for testing. the only thing that I can think of that is different between the runs is the system time/date but that doesn't make sense how it could influence this unless the driver is using it somewhere?

Jibun-no-Kage, 2023-02-14:
Am I reading the trace right, the fault is triggered at "do_syscall_64+0x5b/0x90" ?  Is that sycall not code in kernel or otherwise external to our driver?

alkisg, 2023-02-14:
> @alkisg You may want to read this thread.

Personally I haven't even tried packet injection yet, so I can't comment on this one! :)

morrownr, 2023-02-17:
I'm still searching for the problem. I have it narrowed down based on the following log entry:

```
[41895.608865] rtl8821cu 1-1.1:1.0 wlan0mon: renamed from wlxe84e0689293f
[41895.635143] ------------[ cut here ]------------
[41895.635146] WARNING: CPU: 1 PID: 16271 at /var/lib/dkms/rtl8821cu/5.12.0.4/build/os_dep/linux/rtw_proc.c:6271 rtw_adapter_proc_replace+0x9f/0xb4 [8821cu]

```

Why it just showed up is a mystery.

If you all want to take a look at the code, have at it.

Nick

morrownr, 2023-02-17:
Here is the function where we are seeing the problem show up:

os_dep/linux/rtw_proc.c

```
void rtw_adapter_proc_replace(struct net_device *dev)
{
	struct proc_dir_entry *drv_proc = get_rtw_drv_proc();
	struct proc_dir_entry *dir_dev = NULL;
	_adapter *adapter = rtw_netdev_priv(dev);
	int i;

	dir_dev = adapter->dir_dev;

	if (dir_dev == NULL) {
		rtw_warn_on(1);
		return;
	}

	for (i = 0; i < adapter_proc_hdls_num; i++)
		remove_proc_entry(adapter_proc_hdls[i].name, dir_dev);

	rtw_odm_proc_deinit(adapter);

#ifdef CONFIG_MCC_MODE
	rtw_mcc_proc_deinit(adapter);
#endif /* CONIG_MCC_MODE */

	remove_proc_entry(adapter->old_ifname, drv_proc);

	adapter->dir_dev = NULL;

	rtw_adapter_proc_init(dev);

}
```

RootLUG, 2023-03-04:
I'm not sure if the interface renaming is the only problem. I just tried the latest start-mon.sh which should not change the interface name but when trying it I see the errors:
```
 The following processes have been stopped:

  PID Name
  345 dhclient
  489 NetworkManager

 Note: The above processes can be returned
 to a normal state at the end of this script.

 Press any key to continue...

 --------------------------------
    start-mon.sh 20230220
 --------------------------------
    WiFi Interface:
             wlan1
 --------------------------------
    name  -  wlan1
    type  -  managed
    state -  DOWN
    addr  -  XXXX
 --------------------------------

 Do you want to set a new addr? [y/N]

command failed: Operation not permitted (-1)
RTNETLINK answers: Operation not permitted
```
the inteface afterwards is NOT put into monitor mode (e.g. no radiotap listed in `ip addr`; after finishing the rest of the prompts from start-mon) and the system goes into the same broken state as with the previous bug. So if the latest version does not rename the interface then it should not produce that error if that would be the only issue. I don't see any error output in dmesg in this case.
Is there anything else I can try to test?

morrownr, 2023-03-04:
Let me see if I can duplicate this. What distro and kernel are you running?

RootLUG, 2023-03-07:
It's official kali image for rapsberry pi 4B, updated to the latest version prior to running the script above:
`Linux kali-raspberry-pi 5.15.44-Re4son-v7l+ #1 SMP Debian kali-pi (2022-07-03) armv7l GNU/Linux`

morrownr, 2023-03-07:
I'll add a note to my to-do list to take a look at this distro when able. I am currently in the process of moving so my little lab is packed. This may take some time.

I'm not seeing the last issue you posted above on the distros I regularly test and I have been looking for it.

The original issue regarding the interface renaming is something I have been able to duplicate on some but not all systems. One distro where I could not duplicate it on was the RasPiOS. In looking at the differences between the RasPiOS and the systems where I could duplicate the problem was the use of `persistent interface naming`. RasPiOS does not have this turned on by default. I tried turning it off on two of the systems that showed the problem and the problem went away. I also tested a couple of adapters that use in-kernel drivers and this problem is not an issue.

I am aware of the exact line of code in this and 3 other drivers here that triggers the ugliness in dmesg. I have chased the issue backwards with no success. I have downloaded and tested a couple of other 88x2bu drivers from other authors and they show the same problem.

All indications are that `start-mon.sh` is not the problem. The problem is only happening with the 88x2bu, 8821cu, 8821au and 8812au drivers on some distros. I do not know what is causing the problem. This is not easy to solve.

@morrownr

closed 2024-04-02 by morrownr