morrownr/8821au-20210708 #76: errors encountered during script installation on Armbian

tabshadel, 2023-03-12:
root@armbian:~/src/8821au-20210708# sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: aarch64 (architecture)
: 2/4 (in-use/total processing units)
: 943800 (total system memory)
: 5.15.99-flippy-82+o (kernel version)
: gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
: dkms-2.8.7
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.99-flippy-82+o (aarch64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

morrownr, 2023-03-12:
Hi @tabshadel 

Can I get you to run the following and post the results?

make clean
make

tabshadel, 2023-03-12:
Hi NickThanks for your reply and help. Here is the output of make clean and make:My system: Armbian 23.02.0-trunk Bullseye with Linux ***@***.***:/src/8821au-20210708# make clean#make -C /lib/modules/6.1.18-flippy-82+/build M=/src/8821au-20210708 cleancd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.kocd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.kocd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.kocd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.kocd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.kocd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.kocd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.kocd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.kocd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.korm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.orderrm -fr *.mod.c *.mod *.o .*.cmd *.ko *~rm -fr .***@***.***:/src/8821au-20210708# makemake ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.18-flippy-82+/build M=/src/8821au-20210708  modulesmake[1]: Entering directory '/usr/src/linux-headers-6.1.18-flippy-82+'   ERROR: Kernel configuration is invalid.         include/generated/autoconf.h or include/config/auto.conf are missing.         Run 'make oldconfig && make prepare' on kernel src to fix it. make[1]: *** [Makefile:813: include/config/auto.conf] Error 1make[1]: Leaving directory '/usr/src/linux-headers-6.1.18-flippy-82+'make: *** [Makefile:2501: modules] Error ***@***.***:/src/8821au-20210708# Sent from Mail for Windows From: NickSent: Saturday, March 11, 2023 8:51 PMTo: morrownr/8821au-20210708Cc: tabshadel; MentionSubject: Re: [morrownr/8821au-20210708] errors encountered during script installation on Armbian (Issue #76) Hi @tabshadelCan I get you to run the following and post the results?make cleanmake

tabshadel, 2023-03-12:
PSI also tried the ***@***.***:/src/8821au-20210708# make oldconfig && make preparemake: *** No rule to make target 'oldconfig'.  ***@***.***:/src/8821au-20210708# Sent from Mail for Windows From: NickSent: Saturday, March 11, 2023 8:51 PMTo: morrownr/8821au-20210708Cc: tabshadel; MentionSubject: Re: [morrownr/8821au-20210708] errors encountered during script installation on Armbian (Issue #76) Hi @tabshadelCan I get you to run the following and post the results?make cleanmake

morrownr, 2023-03-12:
>  5.15.99-flippy-82+o (kernel version)

That is from your original post that showed the output of the install-driver.sh script. It says you are running kernel 5.15.

> Leaving directory '/usr/src/linux-headers-6.1.18-flippy-82

Later on I see the above which says the header files are for kernel 6.1.

> ERROR: Kernel configuration is invalid.  

That statement sums it up. Your development environment is messed up. The header files have to be the same version as the kernel you are running. I would help if I could but I have no idea how this mess was created. You may want to post in the appropriate Armbian forum as this is a problem with Armbian.

@morrownr

morrownr, 2023-04-06:
Status?