morrownr/8821au-20210708 #81: Works on TP-Link AC600 and Arch Linux kernel 6.2

robinwesterik, 2023-03-15:
I've just installed this via the AUR and I can confirm that it works flawlessly with my TP-Link AC600 Archer T2U Nano on Arch Linux kernel 6.2

monuyadav016, 2023-03-16:
@robinwesterik Does it support VIF (virtual interface). Were you able to perform Evil Twin Attack using it? I am confused what is different about this fork than the one maintained by [aircrack-ng](https://github.com/aircrack-ng/rtl8812au).

monuyadav016, 2023-03-17:
I was confused whether the drivers did not support VIF or the hardware itself did not provide that capability. Your answer explains it. Thank you @morrownr .

morrownr, 2023-04-06:
> I was confused whether the drivers did not support VIF or the hardware itself did not provide that capability. Your answer explains it. Thank you @morrownr .

You are welcome.

The short answer to the VIF question is that it is a driver issue. If VIF support is important to your project, getting an adapter that works with in-kernel drivers is likely your best answer. We use Issues in USB-WiFi for users to ask specific questions about adapters so feel free to ask if you are looking for something specific.

I'll go ahead and close this issue.

@morrownr

closed 2023-04-06 by morrownr