morrownr/8821au-20210708 #82: Rokland 8812au driver

paultreseler, 2023-03-19:
Got link to install driver from 
https://store.rokland.com/pages/alfa-awus036acs-kalipi-driver-install-instructions

when attempt first line of sudo receive

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package dkms

ran your 
sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
and receive
'mokutil' cannot be found, but can be installed
Why would I want to Delete my Keys?

am running 
Kali 2023.1
Raspberry Pi 4B
kernel 5.15.44 -Rea4on-v8l+ aarch64

morrownr, 2023-03-19:
Hi @paultreseler 

> when attempt first line of sudo receive

Not sure that I understand this. I am not aware of a `sudo receive` for installing. Can you clarify?

Did you run `sudo ./install-driver.sh`? If so, copy the information from the screen and post it here.

Also, can you provide me with a link to the Kali distro you are using so that I can download it?

Regards,

@morrownr

morrownr, 2023-03-26:
@paultreseler 

Fixed?

closed 2023-04-06 by morrownr