morrownr/8821au-20210708 #85: Unable to build the package. (Missing file in linux headers).

Supercip971, 2023-03-25:
I am not sure if it is an issue with this project or my Linux install, but when I try to install the package with:
```
sudo bash ./dkms-make.sh
```
It generates those errors:
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.8-arch1-1/build M=/home/cyp/app/8821au-20210708  modules
  CC [M]  /home/cyp/app/8821au-20210708/core/rtw_cmd.o
  CC [M]  /home/cyp/app/8821au-20210708/core/rtw_security.o
  CC [M]  /home/cyp/app/8821au-20210708/core/rtw_debug.o
  CC [M]  /home/cyp/app/8821au-20210708/core/rtw_io.o
  CC [M]  /home/cyp/app/8821au-20210708/core/rtw_ioctl_query.o
  CC [M]  /home/cyp/app/8821au-20210708/core/rtw_ioctl_set.o
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:252: /home/cyp/app/8821au-20210708/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
  CC [M]  /home/cyp/app/8821au-20210708/core/rtw_ieee80211.o
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:252: /home/cyp/app/8821au-20210708/core/rtw_security.o] Error 1
make[2]: *** [scripts/Makefile.build:252: /home/cyp/app/8821au-20210708/core/rtw_debug.o] Error 1
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:252: /home/cyp/app/8821au-20210708/core/rtw_io.o] Error 1
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:252: /home/cyp/app/8821au-20210708/core/rtw_ioctl_query.o] Error 1
make[2]: *** [scripts/Makefile.build:252: /home/cyp/app/8821au-20210708/core/rtw_ioctl_set.o] Error 1
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:252: /home/cyp/app/8821au-20210708/core/rtw_ieee80211.o] Error 1
make[1]: *** [Makefile:2021: /home/cyp/app/8821au-20210708] Error 2
````
For more information:
- I'm using archlinux (`6.2.8-arch1-1 x86_64`)
- I have installed `linux-headers` (`6.2.8.arch1-1`)

Thanks in advance ! <3

morrownr, 2023-03-25:
@Supercip971 

> sudo bash ./dkms-make.sh

There is nothing even close to the above in the documentation about installing this driver. Please read the README.md and following the installation instructions.

Supercip971, 2023-03-25:
> @Supercip971
> 
> > sudo bash ./dkms-make.sh
> 
> There is nothing even close to the above in the documentation about installing this driver. Please read the README.md and follow the installation instructions.

When I follow the installation instruction, the execution fails when executing this script:
```
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 12/12 (in-use/total processing units)
: 16287180 (total system memory)
: 6.2.8-arch1-1 (kernel version)
: gcc (GCC) 12.2.1 20230201
: dkms-3.0.10
: ---------------------------

Checking for previously installed drivers.
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/v4l2loopback/0.12.5/source/dkms.conf does not exist.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/lib/modules/6.2.8-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.8-arch1-1 (x86_64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```
(the output of `/var/lib/dkms/rtl8821au/5.12.5.2/build/make.log` is the same as the original message).

morrownr, 2023-03-25:
Hi @Supercip971 

> Error! Could not locate dkms.conf file.
> File: /var/lib/dkms/v4l2loopback/0.12.5/source/dkms.conf does not exist.

Something is wrong with your `dkms` installation.

> The dkms installation routines are in use.

But `dkms` is installed and the installation script is trying to use it.

`install-driver.sh` has multiple routines for installing the driver. It checks to see if `dkms` is installed and it if is, it uses `dkms`. If `dkms` is not installed, then it will use an automated manual installation. 

What you can do is use your package manager to remove/uninstall `dkms` and then run the installation script again.

We can see what happens.

Supercip971, 2023-03-26:
I reinstalled `dkms`, `linux-headers` and `linux` and it fixed the issue. 
Thank you very much for your help !

closed 2023-03-26 by morrownr