morrownr/8821au-20210708 #86: Can not install driver for Archer T2U Plus to OpenSuse

YouriyPaluch, 2023-03-29:
Hello.
I can not install driver. After install i got the error. Please help me. 
boba@localhost:~/src/8821au-20210708> sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 6/6 (in-use/total processing units)
: 16239600 (total system memory)
: 6.2.8-1-default (kernel version)
: gcc (SUSE Linux) 13.0.1 20230314 (experimental) [revision 42630fadbe248717859d61c0244c821c32b4e52c]
: dkms-2.8.7
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------

Building module:
cleaning build area...
./dkms-make.sh...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.8-1-default (x86_64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 0.78 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

morrownr, 2023-03-30:
Hi @YouriyPaluch 

> Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.

Please post the contents of the above mentioned make.log file.

YouriyPaluch, 2023-03-30:
Please:

DKMS make.log for rtl8821au-5.12.5.2 for kernel 6.2.8-1-default (x86_64)
четвер, 30 березня 2023 20:43:43 +0300
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.8-1-default/build M=/var/lib/dkms/rtl8821au/5.12.5.2/build  modules
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ioctl_set.o

In the file that is included in the <command-line>:
/usr/src/linux-6.2.8-1/include/linux/kconfig.h:5:10: critical error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~

In the file that is included in the <command-line>:
/usr/src/linux-6.2.8-1/include/linux/kconfig.h:5:10: critical error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation aborted.

In the file that is included in the <command-line>:
/usr/src/linux-6.2.8-1/include/linux/kconfig.h:5:10: critical error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation aborted.
compilation aborted.
make[2]: *** [/usr/src/linux-6.2.8-1/scripts/Makefile.build:253: /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_cmd.o] Error 1
make[2]: *** Очікування завершення завдань...
make[2]: *** [/usr/src/linux-6.2.8-1/scripts/Makefile.build:253: /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_security.o] Error 1
make[2]: *** [/usr/src/linux-6.2.8-1/scripts/Makefile.build:253: /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_debug.o] Error 1

In the file that is included in the <command-line>:
/usr/src/linux-6.2.8-1/include/linux/kconfig.h:5:10: critical error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation aborted.

In the file that is included in the <command-line>:
/usr/src/linux-6.2.8-1/include/linux/kconfig.h:5:10: critical error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation aborted.

In the file that is included in the <command-line>:
/usr/src/linux-6.2.8-1/include/linux/kconfig.h:5:10: critical error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation aborted.
make[2]: *** [/usr/src/linux-6.2.8-1/scripts/Makefile.build:253: /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ioctl_query.o] Error 1
make[2]: *** [/usr/src/linux-6.2.8-1/scripts/Makefile.build:253: /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_io.o] Error 1
make[2]: *** [/usr/src/linux-6.2.8-1/scripts/Makefile.build:253: /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ioctl_set.o] Error 1
make[1]: *** [/usr/src/linux-6.2.8-1/Makefile:2036: /var/lib/dkms/rtl8821au/5.12.5.2/build] Error 2
make: *** [Makefile:2501: modules] Error 2

morrownr, 2023-03-30:
>  gcc (SUSE Linux) 13.0.1 20230314 (experimental) 

I don't test on gcc 13 yet. Any possibility of getting you to back gcc down to v12?

YouriyPaluch, 2023-03-30:
Maybe yes. I will think about it. Do you have plan to test and if necessary to finalize driver?

sunwire, 2023-03-30:
@YouriyPaluch did you install kernel-default-devel?

morrownr, 2023-03-30:
> Maybe yes. I will think about it. Do you have plan to test and if necessary to finalize driver?

The driver is finalized. For today anyway. This is not Windows, it is Linux. The development model of Linux is one of continuous development. Since we are about half way through a kernel dev cycle, I may need to finalize again in about a month... or maybe before. 

While I do look ahead and test on development level kernels, I do not with gcc. When gcc 13 is released, I will test and make adjustments as necessary. Recommend you use gcc 12 for now.

YouriyPaluch, 2023-03-30:
I roll back to linux kernel 6.2.2-1. Module working. But speed very slowly. Less 1MB. And uplouding faster three times then download. I try remove driver. Download new package and install again

morrownr, 2023-03-30:
Please paste the results of the following commands:

lsmod
dkms status

YouriyPaluch, 2023-03-30:
I got it after installing. As you see gcc is 13. I do not know where is problem in kernel 6.2.8.-1

: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 6/6 (in-use/total processing units)
: 16251672 (total system memory)
: 6.2.2-1-default (kernel version)
: gcc (SUSE Linux) 13.0.1 20230314 (experimental) [revision 42630fadbe248717859d61c0244c821c32b4e52c]
: dkms-2.8.7
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------

Building module:
cleaning build area...
./dkms-make.sh...........
cleaning build area...
Compile time: 116.63 seconds
The driver was built by dkms successfully.
: ---------------------------

8821au.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.2.2-1-default/updates/
Adding any weak-modules
getopt: unrecognized option '--no-initramfs'
Usage: ${0##*/} [options] {--add-modules|--remove-modules}
       ${0##*/} [options] {--add-kernel|--remove-kernel} {kernel-release}

--add-modules
        Add a list of modules read from standard input. Create
        symlinks in compatible kernel's weak-updates/ directory.
        The list of modules is read from standard input.

--remove-modules
        Remove compatibility symlinks from weak-updates/ directories
        for a list of modules.  The list of modules is read from
        standard input.

--add-kernel
        Add compatibility symlinks for all compatible modules to the
        specified or running kernel.

--remove-kernel
        Remove all compatibility symlinks for the specified or current
        kernel.

--verbose
        Print the commands executed.

-dry-run
        Do not create/remove any files.
depmod....
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.
: ---------------------------

Do you want to edit the driver options file now? (recommended) [Y/n]

YouriyPaluch, 2023-03-30:
lsmod:
Module                  Size  Used by
snd_seq_dummy          16384  0
snd_hrtimer            16384  1
snd_seq               110592  7 snd_seq_dummy
xt_conntrack           16384  7
xt_MASQUERADE          20480  7
nf_conntrack_netlink    65536  0
xfrm_user              65536  1
xfrm_algo              20480  1 xfrm_user
xt_addrtype            16384  2
af_packet              65536  10
br_netfilter           32768  0
bridge                442368  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
nf_conntrack_netbios_ns    16384  1
nf_conntrack_broadcast    16384  1 nf_conntrack_netbios_ns
nft_fib_inet           16384  1
nft_fib_ipv4           16384  1 nft_fib_inet
nft_fib_ipv6           16384  1 nft_fib_inet
nft_fib                16384  3 nft_fib_ipv6,nft_fib_ipv4,nft_fib_inet
nft_reject_inet        16384  8
nf_reject_ipv4         16384  1 nft_reject_inet
nf_reject_ipv6         20480  1 nft_reject_inet
nft_reject             16384  1 nft_reject_inet
nft_ct                 24576  8
nft_chain_nat          16384  3
nf_tables             344064  494 nft_ct,nft_reject_inet,nft_fib_ipv6,nft_fib_ipv4,nft_chain_nat,nft_reject,nft_fib,nft_fib_inet
ip6table_nat           16384  0
ip6table_mangle        16384  0
ip6table_raw           16384  0
ip6table_security      16384  0
iptable_nat            16384  1
nf_nat                 61440  4 ip6table_nat,nft_chain_nat,iptable_nat,xt_MASQUERADE
nf_conntrack          212992  7 xt_conntrack,nf_nat,nft_ct,nf_conntrack_netbios_ns,nf_conntrack_broadcast,nf_conntrack_netlink,xt_MASQUERADE
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
iptable_mangle         16384  0
iptable_raw            16384  0
iptable_security       16384  0
nfnetlink              24576  4 nf_conntrack_netlink,nf_tables
ip6table_filter        16384  0
ip6_tables             40960  5 ip6table_filter,ip6table_raw,ip6table_nat,ip6table_mangle,ip6table_security
iptable_filter         16384  1
bpfilter               16384  0
qrtr                   57344  4
binfmt_misc            28672  1
squashfs               94208  12
loop                   36864  24
nls_iso8859_1          16384  1
nls_cp437              20480  1
vfat                   24576  1
fat                   110592  1 vfat
snd_soc_avs           217088  0
snd_hda_codec_hdmi     94208  1
snd_soc_hda_codec      28672  1 snd_soc_avs
snd_hda_ext_core       40960  2 snd_soc_avs,snd_soc_hda_codec
joydev                 28672  0
snd_soc_core          438272  2 snd_soc_avs,snd_soc_hda_codec
snd_hda_codec_realtek   196608  1
snd_compress           32768  2 snd_soc_avs,snd_soc_core
hid_chicony            16384  0
snd_hda_codec_generic   110592  1 snd_hda_codec_realtek
intel_rapl_msr         20480  0
snd_pcm_dmaengine      20480  1 snd_soc_core
uvcvideo              176128  0
snd_hda_intel          65536  1
snd_intel_dspcfg       36864  2 snd_soc_avs,snd_hda_intel
mei_pxp                20480  0
[... 76 lines cut ...]
drm_buddy              20480  1 i915
drm_display_helper    217088  1 i915
aesni_intel           401408  0
crypto_simd            20480  1 aesni_intel
cryptd                 32768  2 crypto_simd,ghash_clmulni_intel
cec                    90112  2 drm_display_helper,i915
usbcore               434176  7 xhci_hcd,snd_usb_audio,usbhid,snd_usbmidi_lib,uvcvideo,8821au,xhci_pci
rc_core                77824  1 cec
ttm                   102400  1 i915
video                  69632  2 asus_wmi,i915
wmi                    45056  4 video,asus_wmi,wmi_bmof,mxm_wmi
btrfs                2076672  1
blake2b_generic        20480  0
xor                    24576  1 btrfs
raid6_pq              126976  1 btrfs
libcrc32c              16384  4 nf_conntrack,nf_nat,btrfs,nf_tables
crc32c_intel           24576  2
sg                     49152  0
dm_multipath           45056  0
dm_mod                233472  1 dm_multipath
scsi_dh_rdac           16384  0
scsi_dh_emc            16384  0
scsi_dh_alua           28672  0
msr                    16384  0
efivarfs               28672  1

YouriyPaluch, 2023-03-30:
dkms status:
rtl8821au/5.12.5.2, 6.2.2-1-default, x86_64: installed

YouriyPaluch, 2023-04-04:
After reboot system om my PC and reboot router my usb wi-fi module working very good on kernel 6.2.2-1. I close this issue. Thanks a lot for your time!

closed 2023-04-04 by YouriyPaluch

morrownr, 2023-04-04:
That is good to hear. Enjoy.