morrownr/8821au-20210708 #87: mok.pub not found

basushaunak, 2023-04-06:
Hi,
I am using Fedora 37 Workstation, when I am trying to install the driver, I am getting the following error:
**Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------
Sign command: /lib/modules/6.2.9-200.fc37.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...
Key file /var/lib/dkms/mok.key not found and can't be generated, modules won't be signed**

My device: Dell Optiplex 3060 (i3 8th Gen)
Output of your set of commands:
6.2.9-200.fc37.x86_64 x86_64
SecureBoot enabled
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1a2c:2124 China Resource Semico Co., Ltd Keyboard
Bus 001 Device 004: ID 1c4f:0034 SiGma Micro XM102K Optical Wheel Mouse
Bus 001 Device 003: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 001 Device 002: ID 22b8:2e24 Motorola PCS motorola edge 30
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8821au/5.12.5.2, 6.2.9-200.fc37.x86_64, x86_64: installed
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

sunwire, 2023-04-06:
Just install openssl

basushaunak, 2023-04-06:
Thanks a lot, issue resolved :-)

closed 2023-04-06 by basushaunak

morrownr, 2023-04-06:
Hi @basushaunak 

In the README:

Installation Step 3:

Mandatory if Secure Boot is active: openssl mokutil

You need to install openssl and mokutil if they are not installed.

Installation Step 8:

Note: Fedora users that have secure boot turned on should run the following to enroll the key:

sudo mokutil --import /var/lib/dkms/mok.pub

There is additional infomation in FAQ.md.

After looking at Step 8, I should probably have the note to Fedora users above the `sudo sh install-driver.sh` command.

@morrownr

morrownr, 2023-04-06:
> Just install openssl

Hey @sunwire 

Does Fedora have an instruction about secure boot on its support website? I could add it to the FAQ.

sunwire, 2023-04-06:
> Does Fedora have an instruction about secure boot on its support website? I could add it to the FAQ.

I'm afraid not, but there is [documentation](https://docs.fedoraproject.org/en-US/fedora/latest/system-administrators-guide/kernel-module-driver-configuration/Working_with_Kernel_Modules/#sect-signing-kernel-modules-for-secure-boot) that can be helpful.

BTW: There is a rpm package on [COPR](https://copr.fedorainfracloud.org/) (it is like PPA for Ubuntu or AUR for Arch) and it is maintained be me https://copr.fedorainfracloud.org/coprs/sunwire/dkms-8821au/