morrownr/8821au-20210708 #90: ish alpine linux i686 error / wrong kernel header

punkplayer1, 2023-04-26:
Hello all

i hope u can help me. I really new to this and i m trying and trying but not i need help. Actually i am very excited if my issue could be solved

intention:
Installing rtl8811au driver for running alfa AWUS036ACS on iphone xr (i686) via iSH (Alpine linux)

i followed your guide (thanks for creating btw) and this is what i got:

Smartphone:~/8821au-20210708# ./install-driver.sh
Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.
Once the header files are properly installed, please run "sudo ./install-driver.sh"

Anybody facing this issue?

i ran also following commands before:

Smartphone:~# apk add linux-lts-dev make gcc

Smartphone:~/rtl8812au# apk add alpine-sdk linux-lts-
dev

looking forward for response

morrownr, 2023-04-26:
There is definitely a header files problem. I'm not familiar with Alpine beyond what is in the README and that didn't work so you may need to contact someone familiar with Alpine. You might also do some search engine work.

There are Alpine users that stop by but when that will happen is unknown.

punkplayer1, 2023-04-26:
Thanks Bro
I will post a solution here if i get one. Could be interesting for others

morrownr, 2023-06-12:
Gents,

Right now the Installation Steps in the README show the following for Alpine:

sudo apk add linux-lts-dev make gcc

If it needs to be updated, let me know.

closed 2024-04-02 by morrownr