morrownr/8821au-20210708 #91: Raspberry 2b compile error

ruathudo, 2023-05-25:
I have tp-link archer t2u nano model. Running raspbian kernel 6.1.21-v7. I increased the swap to 1024Mb. But could not compile the driver. 

The lsusb show:
```
Bus 001 Device 004: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
 ```

When try to compile as the instruction from readme, I got this error:
```
make ARCH=arm CROSS_COMPILE= -C /lib/modules/6.1.21-v7+/build M=/home/pi/8821au-20210708  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v7+'
  CC [M]  /home/pi/8821au-20210708/core/rtw_cmd.o
gcc: internal compiler error: Segmentation fault signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
make[2]: *** [scripts/Makefile.build:250: /home/pi/8821au-20210708/core/rtw_cmd.o] Error 4
make[1]: *** [Makefile:2012: /home/pi/8821au-20210708] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v7+'
make: *** [Makefile:2501: modules] Error 2
```

Could you please help! Thanks.

ruathudo, 2023-05-25:
I changed to the new sd card, reinstalled lasted Raspberry OS and the issue is gone. I am able to compile and everything is working fine. Seem something wrong with the old microSD card. Close the issue.

closed 2023-05-25 by ruathudo

morrownr, 2023-05-26:
Hi @ruathudo 

Sorry for the delay. Busy.

The 32 bit version of RasPiOS from earlier this year was messed up. There was no problem in the 64 bit version. If I recall correctly, it was in the header files. That may have been your problem and a new download/burn with a newer version may have fixed the problem.