morrownr/8821au-20210708 #93: cannot compile on linux 6.3.x

kjonca, 2023-06-24:
CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/nlrtw.o
/var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.c:468:17: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
  468 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8821au/5.12.5.2/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl8821au/5.12.5.2/build/include/osdep_service.h:57,
                 from /var/lib/dkms/rtl8821au/5.12.5.2/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.c:17:
/usr/src/linux-headers-6.3.0-1-common/include/net/cfg80211.h:8392:6: note: declared here
 8392 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.c:483:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  483 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.3.0-1-common/include/net/cfg80211.h:8375:6: note: declared here
 8375 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/hal/hal_intf.o
make[2]: *** [/usr/src/linux-headers-6.3.0-1-common/scripts/Makefile.build:257: /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.3.0-1-common/Makefile:2050: /var/lib/dkms/rtl8821au/5.12.5.2/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.3.0-1-amd64'

sunwire, 2023-06-24:
You have to update to the latest version https://github.com/morrownr/8821au-20210708/commit/78187cda81ce567c97cd8a4e11ed89def2d95644

morrownr, 2023-06-24:
$ git pull
$ sudo sh install-driver.sh

kjonca, 2023-06-24:
Well I was pretty sure that I did "git pull". After making fresh "git clone" driver seems to be installed.

closed 2023-06-24 by kjonca