morrownr/8821au-20210708 #95: rock64 with TP-Link AC600

epragalakis, 2023-07-09:
I'm using a rock64 single board computer with arch linux (6.2.10-1-aarch64-ARCH) 
and the AC600 nano Wi-Fi / Bluetooth 4.2 USB Adapter (Archer T2UB Nano).

I followed the instructions on this repo without any errors but I'm not able to get any new results back with the `ip link` command.

Are these drivers meant to work with this hardware & software? 
How can I verify that the installation was successful or check for any error logs in case there are any?

morrownr, 2023-07-09:
Hi @epragalakis 

> Am I using the correct driver?

Let's find out. Band and model information are of no help when it comes to T-Link adapters as T-Link changes chipsets and keeps the model numbers the same very often. Run the following and post the results:

$ lsusb

> Are these drivers meant to work with my hardware & software?

I maintain several drivers here and I do my best to make them work on as much hardware and Linux software as possible. There are practical limits. I do not have your hardware so will need help from you. Can you post a link to the Rocky64? Something that gives me an idea of the processor and other details.

> How can I verify that the installation was successful or check for any error logs in case there are any?

If you used the installation script. install-driver.sh, it will tell you if you were successful. Did you follow the installation steps in the README?

While we are troubleshooting, since most users come direct instead of starting from the site Main Menu, let me give you a link. The first 2 menu items make for good reading:

https://github.com/morrownr/USB-WiFi

epragalakis, 2023-07-10:
1) lsusb
```
Bus 001 Device 002: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC
```

2) Here is some more info on the rock64 single board computer: https://wiki.pine64.org/wiki/ROCK64

3) I followed the installation instructions sucessffully (and I hace done this a couple of times after removing them completely just to double check that nothing went wrong with the process)

4) Reading the link above I might go with `Linksys AE6000` since its small enough for my needs.

morrownr, 2023-07-10:
> Bus 001 Device 002: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC

I see the problem. You adapter uses the rtl8811cu chipset. Remove this driver with the remove-driver.sh script and then delete the folder. Go to the follow repo and start over:

https://github.com/morrownr/8821cu-20210916

There is an added bonus with this chipset in that it now has an in-kernel driver. Once you are on kernel 6.3 or higher, you may be able to remove the out-of-kernel driver and use the in-kernel driver.

> Reading the link above I might go with Linksys AE6000 since its small enough for my needs.

I have one of those. I can't complain. Good little adapter.

epragalakis, 2023-07-12:
thanks for your time :)

closed 2023-07-12 by epragalakis

morrownr, 2023-07-12:
You are welcome.