morrownr/8821au-20210708 #98: broken update fedora 37 via copr

JonnoN, 2023-07-12:
while running dnf up:

Running scriptlet: kernel-core-6.3.12-100.fc37.x86_64                                                                                                                                                                                 65/65   
dkms: running auto installation service for kernel 6.3.12-100.fc37.x86_64Error! Could not locate dkms.conf file.
File: /var/lib/dkms/8821au/5.12.5.2-2.git20230312.fc37/source/dkms.conf does not exist.

afterwards:
\# dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/8821au/5.12.5.2-2.git20230312.fc37/source/dkms.conf does not exist.

broken symlink: /var/lib/dkms/8821au/5.12.5.2-2.git20230312.fc37/source -> /usr/src/8821au-5.12.5.2-2.git20230312.fc37

from the upgrade:
Package dkms-8821au.noarch 5.12.5.2-2.git20230312.fc37 will be upgraded
Package dkms-8821au.noarch 5.12.5.2-2.git20230710.fc37 will be an upgrade

So I guess this is a problem with the copr package being updated in the same run as the kernel? Sorry if this is the wrong place to report that. I'm not very familiar with dkms packages. Thank you for looking

sunwire, 2023-07-13:
I just updated the package on my virtual test machine  (f37) and everything works fine.
Maybe try reinstalling the package (sudo dnf reinstall dkms-8821au), or if that doesn't help, remove it and install it again.

On f38 everything works too.

JonnoN, 2023-07-13:
Thanks for checking. I found some other complaints of people having trouble with other drivers when both the driver and kernel are updated at the same time. Seems like a dkms bug. 

I fixed it with:
`dnf remove dkms-8821au`
`rm -rf /var/lib/dkms/8821au/*`
`dnf install dkms-8821au`

this fixed it for the running kernel, but not the new one, so then I ran
`dkms autoinstall -k 6.3.12-100.fc37.x86_64`

closed 2023-07-13 by JonnoN

morrownr, 2023-07-15:
There you Fedora guys go breaking...your own distro.

Remember that it is a good idea to update this driver before a new kernel installs but it appears your guys have your own packaging so I am out of the loop.

@morrownr